SuperHouse/esp-open-rtos #2: Open source startup code

projectgus, 2015-06-02:
At the moment the startup code is provided in binary libmain. The reset vector is in ROM which performs low level initialisation then maps the SPI flash and calls into the entry point at `call_user_start`.

Startup code includes "standard" libc-style setup, calling into libpp for radio calibration, other functions?

It should be possible to write open source routines that give more fine-grained control to the program author.

Some more details (ROM & IoT SDK based):
https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process

projectgus, 2015-06-04:
First stage is enabling SPI flash cashing so functions from flash can be called (obviously).

@pvvx has a decompiled reconstruction of that sequence here (in Russian):
http://esp8266.ru/forum/threads/revers-sdk-espressif-i-rom-bios-dlja-sozdanija-otkrytogo-sdk.292/#post-5190

pfalcon, 2015-06-07:
> setup of the FreeRTOS task which handles wifi functions.

Can we make FreeRTOS be optional at all? Both because it's extra bloat and because of its questionable licensing.

pfalcon, 2015-06-07:
Another effort: http://richard.burtons.org/2015/05/17/decompiling-the-esp8266-boot-loader-v1-3b3/

projectgus, 2015-06-07:
> Can we make FreeRTOS be optional at all? Both because it's extra bloat and because of its questionable licensing.

I think this is something that might become possible. there are quite a few FreeRTOS-specific calls in the binary libraries, but most of them are in libmain. Some shims some other symbols (mostly mutexing) and it might be possible.

I've thought that Contiki would be interesting to try to port at some point in the future, so having an OS-agnostic base some day would be great from that point of view too.
 
At the very least, the current set of goals would form a solid base for that kind of porting.

Re: OTA bootloader support & rBoot, yes this is an excellent reference thanks. I'll add another issue to track OTA support.

pfalcon, 2015-06-08:
> because of its questionable licensing.

To clarify, by "questionable licensing" of FreeRTOS I of course mean not GPL, but exception 2 to it, which, arguably, make the RTOS license not an open-source license. Summary of my concerns are expressed here: https://github.com/micropython/micropython/issues/1284#issuecomment-105486744

And it's ironical that you consider "Espressif MIT license" a problem, even that additional clause just states status quo: Espressif MAC layer can be run only on Espressif hardware (i.e. that clause covers Espressif ass, doesn't limit users in any realistic way). Whereas FreeRTOS license essentially denies users of basic right of optimizing their software (because if you benchmark software built on FreeRTOS, it's never possible to say precisely how much of your app and how much of FreeRTOS you benchmark, so conservative treatment is that you can't do that at all; and oftentimes, different people do benchmarking vs optimization based on known problem spots, so publishing of benchmarking results is essential, and that's again prohibited by FreeRTOS license).

All in all Open Software/Hardware community should be more selective about what tool they use and avoid double standards in treating hardware vs software vendors.

projectgus, 2015-06-08:
Yes, I fundamentally agree. And I appreciate that the FreeRTOS restriction is a sticking point to a fully open framework.

The reason I'd like to avoid the Espressif MIT License is primarily that it's currently possible to avoid that restriction, without a large drop in functionality.

Avoiding the FreeRTOS licensing restrictions is currently not possible without disabling nearly all functionality. At least the scope of that license is restricted to a small set of esp8266-specific FreeRTOS functions and some binary linked symbols. In time, the current goals of this project will hopefully bring those binary linked symbols down to a set which can be "shimmed out" and replaced with something else. If that turns out to be impossible, then the very least I can say is that those license restrictions will only ever cover the part of the code which relates to FreeRTOS itself.

I 100% agree with you that becoming RTOS-agnostic would be a great thing. It's the kind of benefit that a FOSS licensed framework can uniquely bring. I definitely don't plan to do any work that goes against the possibilities you're hoping to see.

pfalcon, 2015-06-08:
Ok, thanks for fundamentally agreeing.

> currently not possible without disabling nearly all functionality.

Well, nothing in the system calls for the need of RTOS or specifically FreeRTOS: MAC layer delivers packets by an interrupt, the lwIP needs only few timers to run (can be done on interrupts too). The need for "RTOS" comes only at the level of BSD sockets API. And then I'm not 100% sure if it's unavoidable either (let's just assume that lwIP sample code is implemented that way), but then it's still not FreeRTOS-specific. Summing up: starting with plain SDK, it would be possible to arrive at the same functionality, but without dependency on particular RTOS (or on RTOS at all).

All the above can be just considered an independent feedback, and from my side, is an analysis and justification why I may want to go with integrating open builds on lwIP and axTLS into esp-open-sdk (there will still be enough area of collaboration, but no matter what, it's duplication of effort which might have been avoided).

projectgus, 2015-06-08:
There are a large number of FreeRTOS symbols both implemented in and linked from the binary libraries, particularly libmain.a and libpp.a.

pfalcon, 2015-06-08:
Yes, and libmain.a and libpp.a of normal SDK has 0 number of FreeRTOS symbols, which means you won't need to work them around with shims and stuff. Taking into account that all this work is already a big workaround for lack of open SDK, limiting number of sub-workarounds may seem helpful.

I point this not for the sake of argument or to make you suddenly drop FreeRTOS approach, but to properly document implications of both choices and see whether there would be good grounds for somebody to follow another approach than selected by you. (Well, of course if you'd assess that you haven't gone too far with FreeRTOS and still could switch, that would be even cooler ;-) ).

projectgus, 2015-06-08:
You make a good point, I missed that by "plain SDK" you meant "IoT SDK".

To be honest I haven't looked too closely at the IoT SDK. When I started this project it was under the "Espressif GPL", and the RTOS SDK was plain MIT, and that more or less made the decision for me. Maybe if I was starting today, now everything has been relicensed, I would think differently.

There are still some unusual symbols in the IoT SDK. IoT SDK libmain.a actually contains an object file "mem_manager.o", linked from the other libraries, whose functions all match FreeRTOS names ie vPortFree, vPortMalloc, xPortGetFreeHeapSize, etc. Looks like it was probably derived from one of the FreeRTOS heap_xx.c memory management implementations.

Also AFAIK noone has figured out the signatures of IoT SDK ROM functions like ets_task and ets_post, which seem to be used internally for message passing. Maybe they are wrappers around XTOS primitives? You might be able to ignore these though, if you just want to run on "bare metal" with a black box wifi stack, and ignore the RTOS in the background.

I'm going to stick with FreeRTOS for now as I have a much better handle on how its parts fit together. There are also a couple of other minor reasons -
* Not wanting to rewrite axTLS for lwIP raw sockets (I started down this path back in April but it's quite a substantial set of changes).
* Wanting to avoid the "helpful" nature of some of the IoT SDK functions that remember SSIDs, etc.

I will keep an eye out for developments on people deciphering more of the IoT SDK internals, though. There are definitely upsides, like the amount of other code people are writing against it, and it seems like it's getting a lot more attention from Espressif.

pfalcon, 2015-06-08:
> IoT SDK libmain.a actually contains an object file "mem_manager.o", linked from the other libraries, whose functions all match FreeRTOS names ie vPortFree, vPortMalloc, xPortGetFreeHeapSize, etc. [...]

Yes, I noticed that too, and had the same suspicion either - we know that SDK implementation is a mix of bunch of open-source projects, so maybe even "plain SDK" has some bit taken from FreeRTOS.

> Also AFAIK noone has figured out the signatures of IoT SDK ROM functions like ets_task and ets_post, which seem to be used internally for message passing.

Meet system_os_post() and friends in SDK documentation which are wrappers around them.

> You might be able to ignore these though, if you just want to run on "bare metal" with a black box wifi stack, and ignore the RTOS in the background.

So, let's just pronounce what the fuzz is all about - it's about getting rid of proprietary callback-style espconn API, and being able to use standard BSD socket API, and standard sequential-style programs. To achieve that, one would need to deal with ets_task and ets_post (and more).

> Not wanting to rewrite axTLS for lwIP raw sockets (I started down this path back in April but it's quite a substantial set of changes).

Was it? To say it explicitly, I'm interested in having a compliant MicroPython implementation on esp8266, so there's bit of headroom between providing Python-level socket API and the way it may be implemented in C, but still: Let's look how lwIP's tcp_write() works - it buffers data, and then a callback will be called when it's actually sent. Pardon me, but that's just how send() works - on Linux, it will happily slurp megabytes and return immediately. If there was an error, you will learn that much later. If buffer is full, it will block before space is available. That's all doable with tcp_write + tcp_sent callback (callback sets flags for next call to send() impl to check).

Now recv() part: Well, that's done with tcp_recv callback. No problem, let it put packets in queue. Then recv() will either take next from queue, acknowledge it with tcp_recved(), or block if queue is empty (block here and above means busy-wait, sure).

So, /me fails to see how lwIP API is "substantially" different from BSD API. The only caveat with all the above is to have the data pumping. As long as main loop belongs to SDK code, that's not doable. So, need to apply "inversion of inversion of control" pattern ;-).

> and it seems like it's getting a lot more attention from Espressif.

Yes, exactly. For a lot of time it looked like RTOS SDK was  a once-off code drop.

> I'm going to stick with FreeRTOS for now

Well, sounds good. There's certainly a difference between knowing how something can be done and actually doing and have something working, if you're well on the road with FreeRTOS SDK on that, makes sense.

foogod, 2015-07-01:
Hiho..  I wanted to let folks here know about an effort I've been working on over the past week or two to set up a public site/wiki/maillist specifically for reverse engineering efforts around the ESP8266:

http://esp8266-re.foogod.com/

The site is intended to be fully open to anybody who wants to post whatever info they have gleaned from pretty much any area of reverse engineering these devices.  In particular, something that you may be interested in is I have now posted most of the notes I have from my own efforts at reverse-engineering the startup code (and related data structures, etc) of the Espressif RTOS SDK (About 70-80% thus far, and I hope to have a bit more stuff up soon), in a form which (I hope) should be useful to anybody wanting to reimplement things themselves.

I've also set up a maillist/group as esp8266-re@googlegroups.com which I hope to be useful for people to get together and coordinate this sort of work, so we can all know what everybody else is doing etc.  Anyone interested is welcome and encouraged to join in!

projectgus, 2015-07-01:
Hi @foogod,

This is super useful, thank you for publishing it! I was somewhat dreading having to do this RE myself (and I don't have an IDA license either - although I have ScratchABit woking with xtensa now which is very helpful).

Although I was a little dismayed to see yet another esp8266 wiki, you've packed it with good info (and it's open access so thank you heaps for doing that.)

I've subscribed to the esp8266-re mailing list. I will definitely be referring to the docs you've published when it comes time to write new startup code for esp-open-rtos, and I'll make sure to contribute edits for any subtle differences I'll work out (and link back to the wiki where appropriate).

Thanks again!

pfalcon, 2015-07-01:
@foogod: Thanks for coming forward with this! My first comment would be that I'd personally prefer that was done on a popular hosting site (like, well, github), to make sure it can be easily and seamlessly continued even if original creator lost interest/resources to keep it up.

foogod, 2015-07-01:
Well, I had originally debated whether to set up a new wiki for this, but I really think there are some advantages to having a site specifically dedicated to RE (and all the mess that often entails) separate from the more general-purpose resources out there.  The idea is that the ESP8266-RE wiki can be used for all the gory details of ripping things apart (that only those involved in the work really care about), and then over time the insights which are useful to "normal" folks (those who just want to write apps for the ESP8266, etc) can be cleaned up and posted back to the more mainstream sites in forms that are actually useful to people who aren't reverse-engineers.

As for using GitHub for this, to be honest, I really just don't think GitHub is a good resource for this sort of general community wiki, and there are real advantages to using a much more full featured and extensible wiki system for this sort of thing..  This is something we can certainly discuss more on the esp8266-re maillist if folks want to, though (I don't want to hijack this issue thread too much with those sorts of side-discussions, and talking about the wiki is exactly the sort of thing that the list was created for)..  Regarding the maintainership question, I do definitely understand the concerns, but rest assured that the wiki isn't likely to be going anywhere any time soon, at least (the server it's on is already paid up for the next 10 months or so, MediaWikis are not hard to move from one place/person to another if necessary, and I've got no problems giving additional folks wiki admin access (and access to the offsite backups, etc) to help keep things running if I'm suddenly not around for some reason)

FWIW, most of the RE work on the SDK I've been doing thus far hasn't been with any kind of IDA, but just using xtobjdis (https://bitbucket.org/foogod/xtobjdis), a (non-interactive but full-featured) Xtensa disassembler I put together a couple of months back (I actually wrote it before I knew about ScratchABit, but I tend to prefer working with more unixy non-interactive tools anyway)..  I've also put together a few miscellaneous scripts/tools which I've been using to make working on the ESP8266 SDK easier as well, and am hoping to post some more info on those in the next day or two in case anyone's interested..

igrr, 2015-07-25:
>So, let's just pronounce what the fuzz is all about - it's about getting rid of proprietary callback-style espconn API, and being able to use standard BSD socket API, and standard sequential-style programs. To achieve that, one would need to deal with ets_task and ets_post (and more).

> So, need to apply "inversion of inversion of control" pattern ;-).

Just wanted to drop a note regarding this "inversion of control" — it is quite achievable by calling a custom version of `ets_run` from a function which needs to block (e.g. `read`). I used this approach early on during arduino core development, but ultimately discarded it because this caused stack to grow considerably.

Another note regarding "being able to write standard sequential-style programs" in context of IoT SDK. Since we have blocking network APIs in arduino, I know all too well that this causes issues when you have more than one network service running on ESP. While one service (e.g. HTTP) is waiting for more data to arrive, another (like mDNS) gets flooded with data, overflowing all the scarce buffers you have. Without a generous amount of RAM and multithreading, blocking network APIs just don't work well.

projectgus, 2015-07-26:
Hi @igrr,

Thanks for letting us know about your experiences. I'm guessing it's something of a twofold problem in the Arduino case because of the single threaded default programming style, so there's no obvious way to change the API and add either threads or lots of callbacks without making the API style a lot more complex than the typical Arduino code. Interested to see what you end up doing.

Very impressed by your esp8266 Arduino project in general btw, seems to be coming along in leaps and bounds! :)

Angus

foogod, 2015-08-19:
Just wanted to add a note here that I now have a reimplementation of almost all of libmain/app_main.o (including all the startup code) in C source form.  There are a couple of bits I didn't want to finish until we'd sorta sorted out how we were going to be doing the register headers, so I've been sidetracked by issue #8 instead for the past couple of days, but once I've finished up my work on that (hopefully in the next day or two), next on my list is working on integrating and testing the new startup code and submitting a PR here..

projectgus, 2015-08-19:
Fantastic news! Let me know if I can help with either task.

foogod, 2015-08-22:
So I have one question:

For my replacement app_main.c (which is currently a drop-in replacement for libmain/app_main.o), should I keep all the functions/etc prefixed with "sdk_" or not?

Is the "sdk_" prefix intended to convey "these are binary-only routines from the Espressif libs which we don't have source for" (in which case I should be removing the prefix when I create source versions) or does it mean "these are APIs invented by Espressif, not by us" (in which case they should presumably keep the prefix regardless of source/binary form)?

projectgus, 2015-08-23:
Hmm, yeah. I originally meant is as the first one ("this is a binary routine"), and also just to get around the fact that there were a few namespace collisions otherwise (Espressif had renamed things in lwip/axtls to avoid collisions, and I didn't want to go around renaming parts of upstream code).

However I think if something is a 1:1 recreation of the SDK code, you can make a good case for keeping the prefix. If that feels right to you then do it.

Long-term, I'd like to look at evolving the startup code a bit so it probably won't resemble the SDK startup code as closely. I'm sure you have ideas about this as well though, and just recreating the binary SDK functionality is a sensible first step before we talk about them. :)

(Long-long-term, once the first milesteones goals are all completed I'm thinking the logical second milestone is "remove SDK libmain entirely")

foogod, 2015-08-23:
Ok, yeah, the more I think about it the more I think keeping the prefix is probably a good idea..  For one thing, that way if people write apps that use the SDK routines, they don't have to keep changing their code every time we reimplement another piece of things.

It also means that there's a clear distinction between "old API" and "new API", and over time we can pick and choose which routines from the SDK we also want to become "official" parts of our esp-open-rtos API (and give them non-sdk names), and then eventually deprecate the others, etc..

It'll probably also be a bit easier to work with if we don't have to keep tweaking the rename-exclude lists whenever we write new replacements for SDK routines..

foogod, 2015-08-24:
Ok, getting closer, but I'm running into another issue..

Because I'm now adding app_main.o to the list of object files to remove, there are some symbols defined in app_main which now aren't getting auto-renamed, because the renaming stuff doesn't consider them part of the SDK libs anymore (but they do need to be renamed in the other libs in order for them to call my versions properly).

I can see a few different ways to resolve this:

1. Add a way to specify a (manual) list of "extra" symbols which should be renamed even though they don't show up automatically as part of the SDK-defined list.  The problem with this is it would become more and more of a pain to maintain by hand as we do more of this sort of thing.
2. Split the "remove" step into two:  One round runs before symbol substitution (like now) and another one that runs after (so all the symbols in those objects still go through the renaming phase, but then the objects are ultimately removed after so we can cleanly replace them)
3. Don't try to dynamically figure everything out each time, but just generate (once) a static list of "SDK symbols" to be renamed, and use that list going forward.  This has the benefit that it would also simplify the makefile considerably (and speed up the build slightly), and we wouldn't have anything change out from under us without realizing it.  The only drawback is it wouldn't be updated automatically if we changed the version of the SDK libs we're using (we'd need to regenerate it), but (given the current licensing situation) it seems very unlikely we'd end up doing that anyway.

I'm personally leaning toward option 3, but wanted to get opinions from other folks..

projectgus, 2015-08-24:
Oh snap, I should have realised that yesterday when I replied. Sorry.

There's an option (4), which is to drop the sdk_ prefixes from those functions after all. Unless you really feel strongly about it, that's probably the easiest option for now. Especially if you agree that keeping a 1:1 app_main.c is probably an interim measure anyhow, and it's going to evolve pretty quickly from there.

Angus

foogod, 2015-08-25:
I dunno.. as I said earlier, the more I think about it the more I think it'd be good to keep the 'sdk_' prefixes for this stuff.. if nothing else it means that the API won't be changing out from everybody as we go along, which I think is kinda important..

The other thing is that some of these exported symbols that Espressif has come up with are named things like "info", which (on top of being wonderfully non-descriptive as a structure name) is just begging for naming conflicts with user code, so I was actually kinda happy to see it turned into "sdk_info" in esp-open-rtos..

What are your feelings on the other options?  I have no problem doing the work to implement one of them, if folks are OK with the approach being taken..

projectgus, 2015-08-25:
Hmm, solid point re:"info" and similar. And as we dig out more parts of the binary libraries, there will only be more of those names cropping up.

Let's do the static rename list option 3, then. Less makefile complexity is a big plus, and as you say the libraries are unlikely to change in the future. (The more binary functionality gets replaced, the bigger a job that would potentially become anyhow, above and beyond any symbol name changes.)

foogod, 2015-08-29:
[image]

Good news, everyone!

I finally have a fully working implementation of app_main.c!  It boots and runs and everything!  I'm not going to submit a PR just yet, as I want to do a bit more testing (to be honest, thus far I've only actually confirmed that the 'blink' example works)..  However, it is currently up in my fork in the 'open-startup' branch (https://github.com/foogod/esp-open-rtos/tree/open-startup) if anybody else wants to take a look at it / test it out / etc.  (I'd welcome any help in testing it, as it's kinda a fundamental part of things I'd like to make sure it's really right before potentially merging it into the main line.)

projectgus, 2015-08-30:
Wow, really impressive work! Just reading the code is quite educational about the ESP startup process.

I can run the 'simple' and 'http_get' examples, and 'http_get' connects to my wifi and pulls down a page so that's very encouraging. :D 

I got some very odd behaviour loading the ota_basic example with rboot. Lots of very early boot loops (possibly rboot crashing?), I got it to run only once and I'm not sure how (I think it was a combination of chip erase and power cycling to clear RTC memory), then the next time I pressed  reset it landed in the boot loop. The board I have here (nodemcu v1.0 knockoff) doesn't support 74800bps (nor or 74906bps) so I'll try again tomorrow with a logic analyzer when I'm back in my workshop, see what it's actually outputting before it crashes.

projectgus, 2015-08-30:
Hmm, actually I get this even without the OTA stuff. If I upload http_get then it works immediately when the bootloader jumps out, but then if I press 'Reset' I get a boot loop. Will investigate further.

foogod, 2015-08-30:
Hmm, ok, I see part of the problem, I think.. When it reboots, it's reporting a system param error (which basically means the checksum check on the g_ic saved data it read from flash failed).  This might just be a flaw in the checksum validation, or it might somehow be reading the wrong bit of the flash or something.  Unfortunately, the real problem is that I apparently left out a bug fix I had made to the config-sector-dump routines, so when it reports the system param error, it then calls those routines, which go into an endless loop (oops)..

I've pushed up the fix for the config dump stuff, so at least it shouldn't go into an endless loop when it calls that (sorry about that).  Now I just need to figure out why the checksum validation is failing..

foogod, 2015-08-30:
Ok, I think I found it.. I had an off-by-one error when calculating the flash sector to read the g_ic data from..  Should be fixed in my branch now.

projectgus, 2015-08-31:
Yeah, that was it. Working great now!

closed 2015-09-23 by projectgus

slaff, 2016-07-27:
> Not wanting to rewrite axTLS for lwIP raw sockets (I started down this path back in April but it's quite a substantial set of changes).

@projectgus Just as an information: Some time ago I wrote a "glue" for LWIP raw mode and axTLS and it works quite nicely. If you are interested take a look at [Lwirax](https://github.com/attachix/lwirax)

PR #49 (2015-09-23): Open startup
I'd like to look at merging this, it seems pretty stable in the testing I've done. Closes #2.

Some discussion about next stages here:
https://groups.google.com/forum/#!topic/esp-open-rtos/D7SPgMrqc9E