SuperHouse/esp-open-rtos #3: Security bug: No entropy source

projectgus, 2015-06-02:
Similar to the official SDKs pre-v1.1.0, esp-open-rtos uses no entropy source so the random values used in TLS, WPA, etc. are wholly predictable.

It should be possible to use internal timers and/or ADC values to inject entropy. Possibly implement our own versions of `rand()` and `srand()` that combine the entropy source with PRNG?

podema, 2015-08-26:
Since we have our own libc is it not possible to port the rand function from the gnu libc or any open source implementation?

projectgus, 2015-08-26:
The libc already has rand()/srand(), but this isn't really the problem. If you don't have a high entropy source to seed srand() with, your values aren't really random - and that's the bug.

I'm sure there is a high entropy source we can use, I think probably reading the MAC timers from the CPU timer tick interrupt may work, or vice versa - they just have to be out of sync so there's some amount of jitter in one value relative to the other.

Once open source startup code (#2) is in place, that's probably the sensible place to first seed the random number generator as well. Not 100% sure where the best place to get that data after reset is, maybe the ADC or something in the RF internals.

podema, 2015-08-28:
I'm new in embedded development, just wondering why was not possible

foogod, 2015-08-29:
I've actually been thinking about this a bit recently.. I think we should do this right by starting by implementing a proven, cryptographically-strong PRNG.  I'm thinking maybe something like Fortuna, which should be reasonably easy as presumably we could use the existing block cipher capabilities in axTLS as a base (though I haven't yet evaluated memory utilization requirements, etc, which might be an issue)..

Once we have that, then comes the issue of seeding / feeding.  A few thoughts on that:
* It may be possible to get a fair bit of entropy by repeatedly sampling the lowest bit of the ADC input (especially if the TOUT pin is left disconnected, but this may be less useful if it actually needs to be connected to something for the user's application).
* If we can get enough insight into the PHY layer, we might be able to pull some entropy from raw radio sampling / RSSI levels / etc.  (The one drawback of this is that it could potentially be influenced delibrerately by an external attacker, so might give us less entropy than expected under attack situations.)
* The situation can be improved by additionally mixing in secret information not known outside of the ESP module itself.  For example, the Flash chip on every module actually has a unique ID code which could be used for this sort of purpose.  That way even if an external attacker can completely control all the entropy sources, they still won't necessarily know the seed.
* I can't help thinking that with everything else this chip has, it might actually have a hardware entropy source as well, but unfortunately we're kinda reliant on Espressif for any information about that.

Ideally, it'd also be good to save the seed info to flash periodically so it can be read back on startup, avoiding the need for re-seeding every time the system resets.

projectgus, 2015-08-30:
All good suggestions.

When I originally reported this bug to Espressif I suggested the ADC, but their solution was to call system_get_time() and feed that into the axTLS entropy pool, which seems pretty non-ideal (that function apparently returns a non-persistent system uptime in microseconds, 'wDevTimOffset + *0x3ff20a00;).

Other ideas that have crossed my mind are:

* The RTC has an internal RC clock source (confirmed by heating/cooling the ESP and watching the frequency drift up and down relative to the crystal), so it jitters quite a lot of compared to the stable crystal clock source. Mixing this in periodically, say on every RTOS tick, should provide a decent bit of entropy. Less useful at first startup though.

* Saving seed info to flash and/or to RTC memory seem like good ideas. RTC memory may be particularly good as it seems to be pretty garbled on cold start, so you get a reasonably random (to be confirmed) value back regardless. Maybe there are some unused fields in the reserved 'SYSTEM' space?

* The NMI "MAC" counters do seem to be out of step with the RTOS ticks somehow, so it can probably be mixed in as well. Needs a bit more verification of how this comes to be though (they're both probably sourced from the same master clock, so I'm not sure exactly what the phenomenon is that puts them out of step and how predictable it is... Possibly just that sometimes the RTOS tick happens while the NMI is running, which may be super predictable)

* The hardware ADC unit seems to be connectable to other things internally, for example the system_get_vdd33() function has at least a couple of bits of noise in it. It'd be ideal if there's a way to disconnect the internal mux somehow and just read pure noise... The Arduino folks have some interesting info on this, I'm not sure where it originally comes from: https://github.com/igrr/arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_phy.c#L199

* The ADC also seems to be very closely related to the WiFi hardware, so need to make sure any method that uses it regularly/transparent doesn't impact WiFi performance. It's probably an excellent one to use at startup, though.

[/braindump]

foogod, 2015-08-30:
Good thought about the RC / xtal jitter.. We should measure how predictable that is under controlled temperature conditions, though.

The advantage of saving things to flash instead of RTC mem is that you only ever have to fully seed the thing once.  Using the contents of the RTC mem on cold start for an entropy source might be useful, but I'm a little wary as to exactly how random it is (or if, for example, it comes up as basically the same value every time).

Anyway, lots of things to test! :)

FWIW, I'm pretty sure at this point that the reason the ADC is tied up with the WiFi stuff is mainly that the same ADC hardware is used internally by the WiFi transmitter to measure VDD so it can maintain a consistent output power regardless of the system voltage (this is how they can meet the FCC power regulation requirements over such a wide potential supply voltage range, and probably the main reason that the ADC hardware exists in the chip in the first place).  What this means, though, is that the WiFi transmitter has to be switched off any time the CPU wants to use the ADC hardware, because the ADC needs to be switched over to talking to the main CPU instead of talking directly to the PHY module.  All indications I've seen, though, suggest that the ADC _input_ can only be switched between TOUT or VDD, though.  One of the open questions I have a "TODO" on the ESP8266-RE Wiki about is whether you can measure VDD while TOUT is physically connected to something else or not (or if, for example, the act of measuring VDD actually connects TOUT to VDD internally, and thus puts out a voltage on the TOUT pin, etc).  If you can do that safely, then measuring VDD would probably be a fairly portable way to get some entropy, at least on startup, but if you can't, then trying to do that could potentially do nasty things depending on how the developer has hooked up TOUT..

The info you link to from the Arduino site about `vdd33_const` appears to be straight out of the SDK docs, actually, though it is useful that they appear to have confirmed that the phy_info structure is actually the same one that's sort of obliquely referred to in the Espressif docs as "esp_init_data_default.bin" (I had a suspicion, but hadn't been able to actually confirm that), which is potentially very useful for other things..

foogod, 2015-09-09:
FYI, I've been working a bit on developing a modified Fortuna implementation for the ESP8266.. It's a bit slow-going because there are a lot of rather subtle and interdependent architectural and security considerations, but it's coming along.

One problem with any kind of good CSPRNG is going to be how to handle entropy accumulation.  To be robust against attack, an algorithm like Fortuna needs to accumulate entropy in multiple "pools" over a significant amount of time before mixing them in.  The problem is that any entropy pools being kept in RAM will disappear as soon as the ESP8266 goes into deep-sleep, which a lot of applications will want to do.

My current design is basically the stock Fortuna (https://www.schneier.com/fortuna.pdf) with a randomized pool scheduler similar to that suggested by Dodis et al (https://eprint.iacr.org/2014/167.pdf), with the following other modifications:

* The core generator uses AES-128 in counter mode.  Running entropy pools are automatically compressed down to 128 bits each using a hashing function to save space while preserving the entropy. (This is basically the same as the normal Fortuna, with the exception that it uses AES-256 and hashes the pools down to 256 bits each (using SHA-256).)
* This implementation of necessity will not have the full 32 entropy pools of the classic Fortuna algorithm.  I'm currently thinking 6 or 7..  The first couple of pools (the ones that are mixed in most frequently) will be kept in RAM.  The intermediate pools (128 bits each) would need to be saved to RTC memory on sleep, along with the current RNG key (128 bits).  The last (long term) pool or two would be saved in a couple of Flash sectors (I have some techniques to minimize the need for erase cycles while adding entropy).
* (I've also got a couple of (I think) clever techniques to mix the RAM pool entropy into the NVRAM pools when sleeping, so the entropy won't actually get lost, and applications which sleep really frequently will still get entropy updates on a reasonable time schedule..)
* The RNG key will be saved periodically to another flash sector (again, with some techniques to reduce the need for erase cycles) so that we can recover it and continue on after a power loss, etc.

The biggest thing here is the need for a decent chunk of RTC memory to make the whole thing work well.. If we persist the RNG state and three pools, we're looking at needing about 68-72 bytes of RTC memory dedicated to just the CSPRNG.  Personally, however, I think this might still be worth it, particularly since I have a suspicion (I still need to do more double-checking) that there's some significant portions of the "system region" of the RTC memory that the SDK stuff doesn't actually use, which could potentially be used for some of this..

Of course I also haven't even gotten around to the question of entropy sources for the whole thing...

Thoughts?

projectgus, 2015-09-10:
This all sounds really good to me. Carving off some "system" RTC memory sounds viable, and like a good solution.

I have some related things that I was putting off discussing, but in this context they seem good to talk about. Have posted to mailing list:
https://groups.google.com/forum/#!topic/esp-open-rtos/btUfyPyWOuU

foogod, 2015-09-11:
Eureka!  (and fooey)

So I had a thought about finding entropy sources on the ESP8266..  I scanned through all of the known hardware-IO memory ranges, dumping each range twice and comparing the results to see if any addresses had values which changed with repeated reads on an otherwise idle system.  I thought I might be able to find one or two things which had a few bits of entropy naturally (like a RSSI indicator, etc).  I found several things that were obviously just various versions of incrementing counters, but then I found:

**0x3ff20e44**

I don't know what this is connected to, exactly (it's in the "wdev control registers" memory range, but in the middle of a patch of registers that don't appear to be ever touched by the SDK), but every single time I read it, it gives me a value that's completely different.

More importantly, I've pulled substantial amounts of data (32KB at a time) from this port as fast as I could manage (tight loop into a buffer, then send the buffer) and as best I can tell, it's not just widely varying, but actually *completely random*.  It's completely incompressible, visually uniform, unbiased, and passes the TestU01 Alphabit, Rabbit, and NIST (FIPS-140-2) statistical test suites for randomness, all with flying colors.  I've sampled the same data repeatedly at startup from a cold-start and can find no statistical correlations between multiple runs.  This thing is as close to a perfect hardware random number generator as I've ever heard of.

Of course, that might be a problem..  This thing is almost too good to be believed.  It appears to be a completely unbiased hardware random number generator that can generate 100% entropy continuously at around 300 million bits per second.  Either that, or somebody has implemented a cryptographic-quality PRNG in hardware with a sufficiently random seed to make it indistinguishable from one.

Either way, though, it may well be more than good enough for what we need.  Even if we don't trust it completely, it could potentially provide enough external entropy for a cipher-based CSPRNG to completely remove the need for persistent entropy pools, etc..   It looks like all my work on Fortuna may have been a waste of time (sigh)..

projectgus, 2015-09-11:
Wow! That's amazing work! And, well, weird...

Does it keep doing it if you turn off the wifi? Maybe it's some property of the radio baseband that's sufficiently noisy, although that's _very_ noisy if all your stats are to be believed...

This might be something to actually ask Espressif about on their forum. Pretty funny if their hardware people designed in a hardware RNG, and their software people were using nothing...

projectgus, 2015-09-11:
Other crazy thought is that if you're associated to a wifi access point while doing this, is it possible that's a random value coming over from their end?

foogod, 2015-09-11:
No, the wifi wasn't even configured to do anything at the point I was dumping this (I did it inside user_init, so the base hardware was initialized, but it was before it even configures it for AP or station mode, etc).  Even if that were the case, the 300-million-bits-per-second would mean that we'd have to be receiving pure random data from a remote source at or above the theoretical maximum transmission speed of 802.11n (which I think would noticeably impact other things trying to use the spectrum)..  I've also tested it from within a upload-to-RAM image via the bootloader (so no wifi hardware initialization at all) and get pretty similar results..

One of the things that gets me is the purity of the output.  Every bit from 0 to 31 is completely uniform and unpredictable.  I'm pretty much of the opinion that this must have been deliberately engineered to be a random-data source of some kind.  The only real question is how much of it is actual hardware randomness and how much may be pseudorandom scrambling..

(well, and the other good question which you noted, which is that if that's the case, why the heck are Espressif's SDK authors using a crappy PRNG algorithm seeded with a predictable counter?  Though to be honest, this isn't the first thing that I've seen that's made me think that the Espressif hardware guys are brilliant and the software guys are kinda just stumbling through..)

One thing I do want to test is whether the behavior changes any when I am actually associated and using the wifi for communication..  (some other thoughts I've had include sticking it in a metal box, sticking it in the fridge, and performing dumps from multiple modules simultaneously, to see if there are any detectable environmental factors.. I also want to do some tests for longer-term patterns..)

projectgus, 2015-09-11:
> pure random data from a remote source at or above the theoretical maximum transmission speed of 802.11n (which I think would noticeably impact other things trying to use the spectrum)

Hmm yeah good point. :)

Good luck with your continued tests...

projectgus, 2015-09-28:
I'm closing this as our entropy support is at least as good as the Espressif SDK now, almost certainly better. If @foogod finds any problems during further investigation of the Hardware RNG then I'm sure it'll get fixed in esp-open-rtos as well. :)

Other relevant reference (linked in code comments):
http://esp8266-re.foogod.com/wiki/Random_Number_Generator

closed 2015-09-28 by projectgus