SuperHouse/esp-open-rtos #5: Move interrupt vectors to open source

projectgus, 2015-06-02:
Relates to #2 and other libmain replacement functions.

Interrupt vectors are currently in libmain, however some of the user interrupt handling code can be found in the bottom of FreeRTOS/Source/portable/esp8266/port.c

These should probably be combined into a single source file that handles the Xtensa interrupt vectors and the user interrupts.

projectgus, 2015-06-08:
Have started a file `include/xtensa_interrupts.h` which has inline versions replacing some simple interrupt management functions that were defined in the SDK.

As work on libmain replacement continues then this can probably be fleshed out and made totally independent of libmain.

projectgus, 2015-08-05:
This is almost entirely done, it just needs some clean up and consolidation as the exception & interrupt handling is currently split between a few different files.

foogod, 2015-09-23:
Is this done now?  the exception_vectors.S file has been in master for a while now and appears to work fine..

projectgus, 2015-09-23:
There's still some consolidation for the interrupt side of things. There's a bunch of generic interrupt (rather than exception) management stuff in FreeRTOS/Source/portable/esp8266/port.c and scattered around a couple of other places, it'd be good to have just in a single file in 'core' as they're not really FreeRTOS specific.

projectgus, 2015-09-24:
I made a bit of progress on this one this morning, will push something for review soon.

projectgus, 2015-10-06:
Closed as part of #50

closed 2015-10-06 by projectgus