SuperHouse/esp-open-rtos #8: Register access header

projectgus, 2015-06-02:
Flesh out https://github.com/SuperHouse/esp-open-rtos/blob/master/include/esp8266.h to provide register masks, values for known ESP8266 registers - GPIO, UART, etc.

Try to avoid clunky WRITE_PERI_REG-style macros, it should be possible to write clean-looking code that accesses registers directly.

pfalcon, 2015-06-07:
Please consider going data-driven way of providing a description of hardware registers, from which various "sources" and "headers" can be generated. https://github.com/pfalcon/libperipha is dedicated to such approach, a description example: https://github.com/pfalcon/libperipha/blob/master/arm/cortex-m/nrf51/gpio.yaml

projectgus, 2015-06-08:
Hi Paul,

Thanks for the suggestion and the link. That's a really nice way of generating headers in different styles.

The only sticking point for me is that it adds another dependency, Python. Not a big deal for Linux users like us, but a minor hassle for Windows users and possibly OS X users depending on their version.

I'm hoping that esp-open-rtos will only require GNU Make and a few base utils, besides the cross-toolchain. This makes it easiest for the most number of people to get up and running and contribute.

Once the known hardware register set becomes reasonably stable then maybe this is something that could be revisited and adopted, with the idea being that you'd only need Python if you wanted to change the register headers.

Cheers,

Angus

pfalcon, 2015-06-08:
> The only sticking point for me is that it adds another dependency, Python.

It adds a dependency on a Python in the same way as gcc adds a dependency on dejagnu. This is 100% true, right? Dejagnu is required to run testsuite for gcc, so you cannot imagine gcc without dejagnu. But most people don't even know about that - they're given gcc binaries and they run them. Just the same here, layman people don't even need to know that headers are generated from a description, they just compile using them.

 > I'm hoping that esp-open-rtos will only require GNU Make and a few base utils, besides the cross-toolchain. 

Cool, just like I write above.

> Once the known hardware register set becomes reasonably stable then maybe this is something that could be revisited and adopted, with the idea being that you'd only need Python if you wanted to change the register headers.

Sorry, it doesn't work like that. When something will be reasonably stable (which is usually never for amatuer reverse engineering project - hardware gets obsolete first), then nobody will spend any time on it. wm8505 (which is first time we talked, to remind) is "stable", and nobody does anything on it.

The whole idea is to break vicious cycle and do it right from the beginning. Doing right is stopping to pretend that there's anything new or current project is anyhow special. It's the same old time-wasteful thing as gazillion projects before it and gazillion after it. But there's something to change - to make sure that all these gazillion projects are part of big picture. That all old projects are not lost, but available in a central place, and that any new project is easy to start and lead based on accumulated experience of all the older projects.

> with the idea being that you'd only need Python if you wanted to change the register headers.

Sorry, to change registers, one needs to do do a reverse engineering or at least a lot of data mining. Being able to run Python is minuscule nuisance comparing to that, and it's fair to expect that anyone who possesses the above magic skills will be able to figure how to install and run Python (damn it's easy).

Obviously, feel free to consider the above a rhetoric figure, the point is that until open-source reverse-engineering community understands that there should be integration and sustainability of effort, any work done now will be just the same waste as all (well, most) work done before.

foogod, 2015-08-17:
Personally, I don't think that Python is a particularly onerous dependency for anyone getting into hacking on an OS..  Especially since it's already installed on pretty much any modern posix-like system anyway, and would be a development-only dependency (not required if you just want to download and build the OS/SDK, only required if you're doing your own hacking on certain parts of the internals), similar to Automake, etc.  That having been said, I also don't think it would be particularly difficult to switch from one approach to another down the line, so I don't see that we necessarily have to hold off working on this until we've figured out the absolute best possible way to do things.  (Unfortunately, I can't currently take a look at libperipha because that GitHub repository is apparently down, so I'll have to take a look at that in more detail later.)

In terms of the actual C header files/macro format, I would like to suggest a "volatile struct" setup, along the lines of https://gist.github.com/foogod/5a5cef60208b9de2f7bd

This provides for a reasonably intuitive/idiomatic syntax of &lt;device&gt;(&lt;number&gt;).&lt;register&gt;, with no need for access-wrapper macros, for example:

    SPI(0).CTRL0 |= SPI_CTRL0_QIO_MODE;

This syntax also allows the device-type and device-number to be kept together (i.e. "SPI(0)" instead of having to put the "(0)" at the end after the register name), which I think makes more sense, while still allowing the device number to be a dynamic variable if desired (but allowing it to optimize-down efficiently if it's a constant).  FWIW, the above code results in the following object code (disassembly) output:

    l32r         a2, .Llit001 /* 0x60000200 */         
    l32r         a3, .Llit002 /* 0x01000000 */         
    memw                                               
    l32i.n       a4, a2, 8 /* [0x60000208] */          
    or           a3, a4, a3                            
    memw                                               
    s32i.n       a3, a2, 8 /* a3 -> [0x60000208] */    

A few other notes/suggestions:
* All register bitfield definitions are prefixed by the device type ("SPI"), the full register name ("CTRL0"), then the bitfield description/etc ("QIO_MODE").  This makes it a lot easier to see at a glance whether you're using the right bitfield constant for the register you're accessing (for example, if you see something like "SPI(0).CLOCK |= SPI_CTRL0_QIO_MODE" in the code somewhere, you can be sure that's almost certainly wrong)
* Avoid unnecessary abbreviations, especially uncommon ones.  Yes, it will make the constant/macro names longer, but the compiler won't care, and it can make it a hell of a lot easier for other people to know at a glance what's actually going on if they aren't intimately familiar with every single bit of every register of every device type off the top of their heads (as a personal example, it took me a good week or so of looking at stuff before I realized that the "BK" suffix I was seeing on several of the PIN_MUX constants in the Espressif headers was supposed to mean "blink".  Really?  Just to save three characters?)  Of course, having said that, there are a lot of abbreviations that are so common or well known that they're not a big deal (such as CTRL for "control"), so it's not exactly a hard-and-fast rule.
* In my example SPI header, I kept the Espressif convention for multi-bit fields of defining an (unshifted) mask and a shift value (mainly because that usage pattern does actually map pretty well to the Xtensa opcodes, and thus is one of the most efficient ways to do things), but I did add a "_M" suffix to the names of anything that's actually a mask value (to match the "_S" for shift values).  It's possible there's a better way to do this stuff, but I do think however we do it, having explicit identifiers to indicate the type/usage of a constant (at least for anything which isn't just a pure bit value) is preferable to implicit.

(Actually, now that I write them up I realize that pretty much all of the above are examples of two of my favorite "rules of thumb" from the Python world:  "Explicit is better than implicit" and "Code is more often read than written", both of which I think are good principles for any software project.)

On the subject of mask/shift access, we might also want to add some constructs for the most common operations to common_macros.h or something:

    #define VAL2FIELD(fieldname, value) (((value) & fieldname##_M) << fieldname##_S)
    #define FIELD2VAL(fieldname, regbits) (((regbits) >> fieldname##_S) & fieldname##_M)
    #define SETFIELD(regbits, fieldname, value) (((regbits) & ~(fieldname##_M << fieldname##_S)) | VAL2FIELD(fieldname, value))

(names are just off the top of my head and might be improvable)

pfalcon, 2015-08-17:
> Personally, I don't think that Python is a particularly onerous dependency for anyone getting into hacking on an OS.

There doesn't even need to be (user-side) dependency on Python, just the same as there's no need to have autoconf/automake installed to use GNU configure based software - headers would be just pregenerated, ready for anyone to pick up.

 > (Unfortunately, I can't currently take a look at libperipha because that GitHub repository is apparently down, so I'll have to take a look at that in more detail later.)

Strange, it's accessible for me just as any other repo, and here's example of the above: https://github.com/pfalcon/libperipha/tree/master/arm/cortex-m/nrf51

> (names are just off the top of my head and might be improvable)

Compare with an uncommitted change I have:

~~~
#define FIELD(val, fldname) (((val) & fldname##_mask) >> fldname##_shift)
~~~

(The change to add auto reg dumper for https://github.com/xobs/fernly . That went backlogged as I figure I'd spread too thin hacking on different hardwares which will become irrelevant in a year. But the point is that - once you have declarative description of registers, you can generate C header or Python ctype files for it, register dumpers in human-readable form, or whatever.)

projectgus, 2015-08-17:
Hi @foogod,

This looks like a good approach to me. I think the existing https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/esp/registers.h could be brought over to this without too much refactoring. Any chance you have time to make the changes and send a PR?

Aiming for explicitness & readability seems very sound. Even though lines with register access are going to be quite long, they're going to be quite readable. Also I'm pretty keen to wrap any commonly used register access into higher level access functions anyhow, like what's in https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/esp/gpio.h & https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/esp/timer.h - so most code doing "common" register things can stay pretty succinct but still readable (and, even better, the implementation of the wrappers stays readable).[*]

We might be about to push the limits of a single all-peripherals registers.h though. It might be time to split that out into multiple register headers. What do you think?

Angus

[*] I got a bit clever writing timers-private.h/timers.c with what really may be premature optimisation, so some of that cleverness I should probably roll back in pursuit of readability.

foogod, 2015-08-18:
@projectgus:

Yeah, I'll look into refactoring registers.h and submit a PR if you think that's a good way to go..

I do think that sooner or later we will want separate headers for different groups of registers anyway.  Doing this also helps keep things more modular and explicit (and it's always posssible to create a registers.h that just includes all the other individual files, if somebody really wants one, but it's not possible to go the other direction)..

I think we probably also will need to make some explicit choices about how we handle GPIO/IOMUX/pin labelling with regard to constant/register names, etc, too.  My personal preference (contrary to what Espressif does) is to name all pins and IOMUX registers with primary names of GPIO&lt;num&gt; (and if necessary, have aliases for their other names, but even that I think may be unnecessary).  This looks to be similar to what you were doing with iomux.h already, I think?

The more I think about it, I'm also thinking that things like SPI's W0-W15 registers would be better as an array (i.e. "SPI(0).W[0]" .. "SPI(0).W[15]")..

@pfalcon:

Looks like GitHub has fixed their issue, it's not telling me the repository is down anymore..

First thoughts on looking at libperipha (very briefly):
* It would be nice if it had a bit more documentation :) (Not a deal-breaker, though)
* I do like the idea of defining these things in a language-agnostic form (such as yaml) and generating C headers (or Python modules or etc) from there.
* Unfortunately, the use of YAML does make for another dependency (which is a bit more annoying than just being dependent on Python)
* It looks like it's only really designed to generate headers one particular way, and it looks like it would need some hacking to do some things like the "SPI(i)" macro in my proposed header format.  I think I would also prefer a more template-based approach to make it more generally applicable to various projects' preferences/coding-guidelines/etc..
* Would prefer it if the C source it generated had proper formatting/indentation (minor nitpick, but still)..
* Not sure I'm a big fan of having to include libperipha_types.h in all the headers just for a couple of defines that aren't really necessary or particularly useful anyway. (smells just a bit like tool-branding?)
* The Javadoc-style comments are a nice touch, but it would be nice to be able to turn them off or use alternate formatting (such as other Doxygen-compatible styles) instead.

Still, it might be an interesting starting point for something..

pfalcon, 2015-08-18:
> First thoughts on looking at libperipha (very briefly):

As a usual intro, when working on open-source projects, I always do prior art search and strongly prefer to base on somebody else's code rather than starting from scratch. libperipha implementation is based on code contributed by some to libopencm3 (and later removed, and salvaged by me to a separate project).

> It would be nice if it had a bit more documentation :) (Not a deal-breaker, though)

Yup, if only somebody contributed it ;-). I'd however prefer if people contributed actual descriptions, while keeping them clean, so they served as by-example documentation ;-).

> Unfortunately, the use of YAML does make for another dependency (which is a bit more annoying than just being dependent on Python)

Well, that was already answered - it's not that important what dependencies are to *generate* headers, as long as most people can use already generated headers. As for YAML specifically, when I surveyed the landscape, there were 2 choices to base from: chrysn's YAML descriptions for libopencm3 and vendor CMSIS SVD descriptions. SVD is XML, and I hope you do agree that it's much less human readable/writable.

> It looks like it's only really designed to generate headers one particular way, and it looks like it would need some hacking to do some things like the "SPI(i)" macro in my proposed header format.

Well, of course it can, depending on your attachment to particular characters. But before going forward, it makes sense to consider that they may be more or less established formats for doing that in the industry. Indeed, there's, ARM CMSIS provides conventions for accessing register across different hardware implementation (not for 1 or 2 particulars). It uses following convention:

~~~~
BLOCK->REGISTER
~~~

Here's how several similar blocks are defined :
https://github.com/pfalcon/libperipha/blob/master/arm/cortex-m/lm4/gpio.h#L94 , here's how described in libperipha: https://github.com/pfalcon/libperipha/blob/master/arm/cortex-m/lm4/gpio.yaml#L21 . Can you use "." instead of "->"? Yes, you can, just define `#define GPIOA (*(GPIO_Type*)GPIOA_BASE)`.
Can you use GPIO(n) instead of GPIOA/GPIOB/GPIO1/GPIO2? Yes, you can (with minor modifications to libperipha-tool indeed), e.g. on YAML level it would be defined as:

~~~~
    - name: GPIO(n)
      base: 0x40004000
      size: 0x1000
~~~~

On header level as:

~~~~
#define GPIO(n) (*(GPIO_Type*)(GPIO_BASE + n * GPIO_SIZE))
~~~~

So yes, you can, but why? Just to be different? Espressif does it very well with naming GPIO signals something completely different.

Note that libperipha's aim is NOT implement ARM CMSIS format. it's aim is to allow to freely use software (including existing software) without dependence on vendor headers. And ARM CMSIS is just one of formats it supports to that end. (This disclaimer is for a case that ARM goes mad and tries to copyright/patent uncopyrightable/unpatentable things, like "Novelty way to use C '->' operator to access hardware registers".)

pfalcon, 2015-08-18:
> I think I would also prefer a more template-based approach to make it more generally applicable to various projects' preferences/coding-guidelines/etc..

I smell contradiction here. You complain about YAML dependency, but let me translate what you say above: "I would prefer to add another external dependency on a templating engine. I would also prefer to spend my time on fighting that engine's peculiarities (as there're, with any such engine)." Both chrysn and me worked off following requirements: 1) have declarative, human readable register descriptions; 2) get from that to actual C headers ASAP. So, chrysn whipped a script which does templating in Python, and I picked it up for usage and evolutionary changes. If someone will join project with another requirement of "going from 1 to 2 should be using a template engine", let be it, but I'd rather recommend focusing on other aspects.

> Would prefer it if the C source it generated had proper formatting/indentation (minor nitpick, but still)..

I at first didn't understand what you mean. Do you mean `__rw` annotations and cases where they're missing? Minor indeed, patches welcome (but then there're >1 way to do it).

> Not sure I'm a big fan of having to include libperipha_types.h in all the headers just for a couple of defines that aren't really necessary or particularly useful anyway. (smells just a bit like tool-branding?)

Tool-branding? :-D Nope, just obvious application of DRY and avoid-naming-conflicts-with-namespacing principles. But tool-branding? Well, not bad idea, given that only way such project can make an impact is by many people be aware of it and contribute to it. But if there's idea to squash contents of that header into generated ones (via templating of course), well, sounds good too.

> The Javadoc-style comments are a nice touch, but it would be nice to be able to turn them off or use alternate formatting (such as other Doxygen-compatible styles) instead.

Those are exactly Doxygen comments, as implemented by chrysn. Again, I smell a contradiction - you started with asking for documentation, and now ask for a way to get rid of it ;-). Can be done. Usual question - why? There can be added options for every minor variation in output format, but the only gross effect is to support 40-50 output formats instead of 4-5 (most people will still use 4-5).

foogod, 2015-08-18:
> Well, of course it can, depending on your attachment to particular characters. But before going forward, it makes sense to consider that they may be more or less established formats for doing that in the industry. 

Well, for starters, some of the differences we're talking about cannot just be condescendingly dismissed as "attachment to particular characters" or "just to be different".  There are significant advantages to the parameterized-macro approach that your CMSIS-like model simply doesn't support.

To be honest, I care much less about adhering to some arbitrary convention from some other platform (which somebody else may or may not have put much thought into in the first place) than trying to have an access methodology which is intuitive, easy to work with, and reasonably efficient in a variety of contexts on the hardware actually being used.  It does make sense to look at what other people have done before and evaluate that based on its own merits, but mindlessly propagating other people's conventions just because "they did it that way before" is really the coding equivalent of "style over substance" in my opinion.

> I smell contradiction here. You complain about YAML dependency, but let me translate what you say above: "I would prefer to add another external dependency on a templating engine. [...]

Well, first of all, you don't have to use a third-party templating engine to add some basic templating capability to a piece of software, and even if you do that doesn't necessarily need to be an external dependency.  However, I never said that dependencies were forbidden either.  The ultimate question is always: is the effort of installing the tool made up for by the benefit the tool provides?  In this context, templating capabilities might (or might not) be worth the negatives of an extra dependency (particularly since the bigger hurdle is always the first dependency anyway (which in this case is YAML), the difference between 0 and 1 is always much larger than the difference between 1 and 2 in practice).

>> Would prefer it if the C source it generated had proper formatting/indentation (minor nitpick, but still)..

> I at first didn't understand what you mean. Do you mean __rw annotations and cases where they're missing? Minor indeed, patches welcome (but then there're >1 way to do it).

Actually, I was mainly commenting on the fact that the output struct definitions have no indentation at all..  Though since you mention it, I don't really see the point to custom opaque "__rw" and "__ro" types (which you have to go look through a separate header file to understand what they actually translate to in terms of C semantics) over simply using "volatile" and "volatile const" in the declarations (which any C programmer knows what they mean).

>> The Javadoc-style comments are a nice touch, but it would be nice to be able to turn them off or use alternate formatting (such as other Doxygen-compatible styles) instead.

> Those are exactly Doxygen comments, as implemented by chrysn. Again, I smell a contradiction - you started with asking for documentation, and now ask for a way to get rid of it ;-).

No contradiction.  First, I asked for more documentation _on how to use the tool_, not inserted in the produced output, which are in no way the same thing.  Second, I'm not asking to get rid of it, I'm suggesting an _option_ to configure it and/or change the format of the output.  Also, my other point that you apparently completely missed was that Javadoc comments are not the only way to do Doxygen (and Doxygen is not even the only way to do documentation-oriented-comments, for that matter).

Anyway, I'm not saying you have to fix all of my complaints (and I'm not the arbiter for whether to use any particular tool in this project anyway) these were just some thoughts I had when first looking over libperipha, that's all. Folks here can take them or leave them as they see fit..

pfalcon, 2015-08-18:
@foogod: I appreciated your feedback on libperipha, and made sure to answer your questions and describe why it is made the way it is so far. As many folks who work on open-source projects in free time, the choice I usually have is not "answer in way1" or "answer in way2", but "answer" vs "not answer". So, I can just hope that the answer helped you to see ideas behind libperipha better and decide whether it would work you. If it does, your contributions would be much welcome.

> Folks here can take them or leave them as they see fit..

Projects like libperipha are in rather subtle position. First of all, folks who work with only one piece of hardware per unit of time don't see need for it at all - they just deal with whatever they got on their hands. Then, folks who need to get things done per specific plan, also don't have time to fly in meta-levels - it's always easier to hack something up directly. Then, folks who take it as a hobby, can do only limited work on it. For example, I put up libperipha after I figured I waste more time adapting inconsistent vendor crap for use in https://github.com/pfalcon/PeripheralTemplateLibrary . But I transcribed just GPIO/UART registers for handful of boards I had around. Finally, companies are not interested in such projects, because they make their product consistent and reusable with the other products, and their aim is to "differentiate". So, it's pretty hard to find "folks" who can "take" it. For me, I saw a problem, and wanted to make sure I have a ready tool next time I face such problem again. So, @foogod, if you like the idea, and see a need to move it forward, please join in.

pfalcon, 2015-08-19:
Btw https://github.com/pfalcon/libperipha/commit/2e861242fe3f3786dd5974169275d3f512c23770 shows, why having "->" to access hw blocks is better than ".": because you can pass pointer to a block around as needed. (No indentation again!!11 - always leaving low-hanging fruit for people to jump into contributing ;-) ).

foogod, 2015-08-24:
Feel free to work on the rest of them.. I had mainly done the ones which were needed for the startup code I'm working on, which I've done, so I didn't have any immediate need/plans to do the others..

foogod, 2015-08-24:
Note that you may want to revisit some of your helper functions for some of this stuff.. Under the new header style, a bunch of them reduced down to operations which are as simple (or simpler) than the helper function call (for example, timer_get_count(frc) is now just the same as "TIMER(frc).COUNT"), so we may want to just eliminate or change some of them.. I figured I'd let you make the call on that, though..

projectgus, 2015-08-24:
Hmm, yeah. I think for as long as there are some wrappers, it's worth having a full set of wrapper functions for the common functionality - just so that code written stays in a consistent style right through.

It is definitely the case that if the registers are this readable people might prefer to just use them directly though, and that's fine too. Will see how we go I guess.

projectgus, 2015-08-24:
Hmm I didn't explain that well, still pre-coffee here in Australia. What I mean by "full set" is, for example, if we have some timer wrapper functions for common timer functionality then it should be possible to write a program that does generic timer things using just the wrapper layer - so that your program reads consistently.

At least that's the way I'm leaning now, maybe it'll turn out to be different in practice.

foogod, 2015-08-25:
So FYI, I just happened to notice that apparently the `__attribute__ (( packed ))` on the register structures causes GCC to access the members with multiple 8-bit accesses instead of single 32-bit (i.e. it uses unaligned-fixup-load/store code even for aligned volatile 32-bit addresses) when compiled with `-O0` (but only under `-O0` (that is, debug builds)).

If nobody else gets to it first, I'll submit a PR this evening to remove all of the `packed` attributes on these structs.. We shouldn't actually need them (they just seemed like a good idea at the time), and if the compiler does add in some padding somewhere where it shouldn't, it should get caught by the `_Static_assert`s anyway..

projectgus, 2015-10-12:
Remaining registers to add definitions for:

Working from http://esp8266-re.foogod.com/wiki/Memory_Map#mmio

* RTC
* I2S
* SLC (ie DMA)
* PHY
* SAR

Everything here but RTC may be mysterious enough that they should be split out to future work, though.

foogod, 2015-10-12:
FYI, I put together an esp/rtc_regs.h header as part of commit 1da996e (in the open-libmain branch).. it could be pulled out separately and merged in if people want it sooner, though..

projectgus, 2015-10-13:
Awesome! Sorry I hadn't looked at that branch yet. I have one in a private branch somewhere as well with some comments documenting what I figured out about behaviour. I'll combine them into a separate commit and then probably close this Issue, break out the gnarlier headers to their own Issues.