SuperHouse/esp-open-rtos #11: Allow storing string literals in irom (flash) instead of dram

projectgus, 2015-06-10:
(This is spinning off a discussion from #1.)

A large part of DRAM usage is currently .rodata, and a substantial part of that is string literals. It'd be nice to store string literals in SPI flash (irom), but irom requires 4-byte aligned word reads..

These are the options that I know of, with a summary from my point of view:

## Option A

Suggested by @pfalcon, add a compiler frontend attribute that forces all pointer access to go via 32-bit aligned reads.

Advantages of being generic. Disadvantage of requiring separate functions to propagate the attribute, due to C's type system, and requiring gcc changes to front-end and back-end. Tricky to apply to existing binary library code.

## Option B

This is a suggestion based on Option A, which is that it would be possible to implement Option A just in terms of existing gcc primitives. ie base it around a macro that dereferences `char *` via a 32-bit read. I think this would end up looking similar to avrlibc's `avr/pgmspace.h` functions. 

Advantages/disadvantages similar to Option A, although the separate functions will be more complex to write than for Option A (but no compiler changes).

## Option C

This is @jcmvbkbc's suggestion, to implement an exception handler that catches the faulting read and completes it.

Advantages of requiring basically no additional code changes, could potentially work with any/all of .rodata automatically, and is potentially applicable to binary libraries by moving .rodata symbols to irom (symbols that are accessed while SPI flash is unmapped notwithstanding). Disadvantage of being slow, @jcmvbkbc thinks 2 orders of magnitude slower.

## Option D

This was my original plan, which is to wrap printf(), write() and any other likely functions at compile time. Where the argument is a const at compile time, store it statically in IROM then load it into stack or heap dram at runtime and pass it to the inner wrapped function.

Advantages of being pretty easy to implement (I think), and relatively fast. Disadvantages of only applying to certain functions, although I think printf will cover a large number of use cases. Can't be applied to binary library code.

## ***

My gut feeling at the moment is I'd like to implement (D) soon, and maybe experiment with (C) when I have some more time as it sounds interesting. I'd be open to patches that try any of these approaches, though!

projectgus, 2015-06-10:
Just to make sure I understand how Option (C) would work, do I have it right that the exception handler would need to be something like this:

* Save register state
* Check faulting instruction opcode was an l8 (maybe also l16 non-aligned l32)
* Determine source address and destination register from opcode.
* "`source &= ~3`"
* 32-bit read from aligned source address.
* Shift right to match the original source address & read width.
* Overwrite saved destination register with correct result.
* Restore register state, update PC/exception state as if the read succeeded.

... otherwise, continue with normal exception handler - register dump and halt.

jcmvbkbc, 2015-06-10:
> Determine source address and destination register from opcode.

Source address will be recorded in the EXCVADDR SR.

> Shift right to match the original source address & read width.

There's special SAR load instruction for that case that produces correct shift amount from the address: ssa8l.

dpgeorge, 2015-06-10:
How about a simplified version of option A: go conservative and just force all non-32bit reads to go via a l32?  This requires changes only to the backend emitter and would be very localised.  Just replace all emissions of l8/l16 with l32+sar.  No need to worry about writes because they are always to dram.  Just need a compiler option (--mforce-l32) and no changes to the source code whatsoever.

I checked the disassembly of the ESP8266 port of MicroPython to get some stats on number of loads.  The text size is 311800.  I did a simple wc -l on the output of objdump -xd.  I get:
- total lines: 79162 (includes some meta data, not just assembly code)
- total l32: 10554
- total l16: 141
- total l8: 485

So around 5% of all loads are l8 or l16 which is pretty low.  The code size increase by converting these to l32+sar would be minimal, and drop in efficiency would not be too bad.  Very much worth the reclaimed RAM.

jcmvbkbc, 2015-06-10:
> Just replace all emissions of l8/l16 with l32+sar

l16 looks a bit trickier than l8, because if we generate code that assumes aligned data it won't get exception if unaligned pointer is passed to it and will silently return wrong result. And handling unaligned l16 in the backend doesn't seem feasible as gcc data movement patterns are not allowed to use temporary registers.

dpgeorge, 2015-06-10:
> l16 looks a bit trickier than l8,

l8 on its own would be a good start, as proof of concept.  It might also be enough since I guess that l16 is not used for string processing (except maybe in optimised strcpy/memcpy etc functions which can be de-optimised by hand).

> gcc data movement patterns are not allowed to use temporary registers.

Can you globally reserve one of the registers?

jcmvbkbc, 2015-06-10:
> Can you globally reserve one of the registers?

Yes. I thought that may impact generated code size, but a quick test with -ffixed-<register name> showed no difference on http server project.

pfalcon, 2015-06-10:
> l16 looks a bit trickier than l8, because if we generate code that assumes aligned data it won't get exception if unaligned pointer is passed to it and will silently return wrong result.

@jcmvbkbc: I guess that's reasonable assumption and compromise, in a sense that other RISC architectures may require natural alignment for accessing sub-word units, and have varying behavior if that doesn't hold (behavior ranging from silently returning undefined data to a fatal fault). ARM, MIPS all work this way. And that's pretty different from requiring *all* memory accesses to be on word alignment. I'm not CPU architectures scholar, and I know only about one more or less popular CPU to have similar restriction - custom XAP core as (was) used in CSR BlueCore Bluetooth SoCs (e.g. https://github.com/pfalcon/blutunode). That was strictly 16-bit CPU, and not only had no 8-bit operations (like zero/sign extension), but even memory laod/stores were only 16-bit. So it's char was 16-bit, which is of course not memory efficient. So, that's the issue we seek to resolve, not an arbitrary case. Thanks for looking into this, let me know if you'd like to have proper gcc-xtensa ticket backpointing to these discussions.

pfalcon, 2015-06-10:
@projectgus:

> Where the argument is a const at compile time, store it statically in IROM then load it into stack or heap dram at runtime and pass it to the inner wrapped function.

That means limiting maximum size of a string/array, or risk stack/heap overflow.

projectgus, 2015-06-11:
I've added an "Option E" to the top of the issue to account for the compiler option suggestion, just to keep track of everything. Sorry I missed it when I created the issue in the first place.

Re: Option D and loading into dram at runtime, memory usage is definitely of concern. The benefit is that the max memory footprint caused by string literals would go down from the sum size of all string literals, to the size of the largest individual string literal.

I really like the sound of this compiler option ("Option E") though, if that turns out to be possible.

jcmvbkbc, 2015-06-11:
Ok, here's my lame attempt to add -mforce-l32 (both for bytes and shorts) to current gcc: 
https://github.com/jcmvbkbc/gcc-xtensa/commit/6b0c9f92fb8e11c6be098febb4f502f6af37cd35
Passes basic sanity tests.
Please let me know what you think.

light, 2015-06-11:
May I suggest to adopt an option that preserves choice for the end programmer ? The trade off between speed and memory usage is not always in the same direction, not even inside one program. For example if I have a configuration page I don't care if it is a bit slow but there is probably a lot of help text so I don't want to use too much memory for it. On the other hand if I need to compare command strings in my API I want it to be as fast as possible for better response time and to save power.

jcmvbkbc, 2015-06-12:
> an option that preserves choice for the end programmer

Sorry, I'm not sure I understand what choice.
With the current implementation it's possible to pass -mforce-l32 to the compiler only for those source files that need it. And it's just for a PoC to see if it's  useful at all.

pfalcon, 2015-06-12:
Ok, preliminary results using -mforce-l32 on https://github.com/micropython/micropython ("smoke test", just building existing code w/ and w/o option):

~~~~
- 310642    1396   54176  366214   59686 build/firmware.elf
+ 320094    1396   54176  375666   5bb72 build/firmware.elf
~~~~

So, the code growth is noticeable, but not too bad (~3%).

~~~~
-.irom0.text     0x0000000040210000    0x3fda0
-.text           0x0000000040100000     0x72f2
+.irom0.text     0x0000000040210000    0x42074
+.text           0x0000000040100000     0x750a
 .data           0x000000003ffe8000      0x574
 .rodata         0x000000003ffe8580     0x4ce0
 .bss            0x000000003ffed260     0xd3a0
~~~~

-mforce-l32-compiled binary boots and passes smoke test ;-).

pfalcon, 2015-06-12:
Ok, using a quick hack to put .rodata to .irom0.text - ... doesn't boot. It of course may be something with linker script (or esp8266 bootrom handling).

jcmvbkbc, 2015-06-12:
> put .rodata to .irom0.text - ... doesn't boot. It of course may be something with linker script

Blobs and interrupt handlers also may not be ready for that.

pfalcon, 2015-06-12:
Well, right. Doing it in less aggressive manner boots:

~~~~
.irom0.text     0x0000000040210000    0x45be9
.text           0x0000000040100000     0x750a
.data           0x000000003ffe8000      0x574
.rodata         0x000000003ffe8580     0x1180
.bss            0x000000003ffe9700     0xd3a0
~~~~

pfalcon, 2015-06-12:
Ok, final picture, confirmed via map file that all upy objects are in .irom0.text, and .rodata consists only of SDK/libc stuff.

~~~~
.irom0.text     0x0000000040210000    0x4693e
.text           0x0000000040100000     0x750a
.data           0x000000003ffe8000      0x574
.rodata         0x000000003ffe8580      0x41c
.bss            0x000000003ffe89a0     0xd3a0
~~~~

Great work, @jcmvbkbc ! (But that's only smoke test, yet need to run testsuite.)

light, 2015-06-12:
@jcmvbkbc my comment was general and not about your solution particularly, since all the options proposed in the first comment seemed still open. If your implementation can be switched per file, then it's perfect for me.

projectgus, 2015-06-14:
Thanks for this @jcmvbkbc, very interesting development.

I applied the patch against gcc 4.8.2 instead of 5.1.0, using the lx106-g++ branch of your crosstool-NG fork and this seemed to also apply & build cleanly.

Adding -mforce-l32 to TARGET_CFLAGS in crosstool produced a libc with about 15% larger .text (255282 ->  292967 bytes), so more than what @pfalcon saw with micropython - probably not surprising for an unlinked library that's not representative of any actual code usage patterns.

Compiling the libc made me realise that there are complications similar to @light's concerns, though. If libc is built with -mforce-l32 then the performance hit applies to all use of libc functions. If it's built without -mforce-l32, any libc function that is passed a .rodata argument from irom will fault. I can't see a way around that apart from linking two parallel libcs!!

***

There's one other thought I had. I know this is asking a lot, but I was wondering if this optimisation could be applied to an intermediate stage rather than the code generation stage.

Easiest to explain with an example. This code:

```c
const uint32_t *something = (uint32_t *)some_extern;
uint8_t a = (*something >> 24) & 0xFF;
uint8_t b = (*something >> 16) & 0xFF;
uint8_t c = (*something >> 8) & 0xFF;
uint8_t d = *something & 0xFF;
printf("%d %d %d %d\r\n", a, b, c, d);
```

Compiled with -O2 produces:

```asm
 b:   0368            l32i.n  a6, a3, 0
 f:   753860          extui   a3, a6, 24, 8
12:   754060          extui   a4, a6, 16, 8
15:   745860          extui   a5, a6, 8, 8
18:   746060          extui   a6, a6, 0, 8
```
(Non-relevant compiler interleaved instructions removed.)

OTOH, this code:
```c
const uint8_t *something = (uint8_t *)some_extern;
printf("%d %d %d %d\r\n", something[0], something[1], something[2], something[3]);
```

Compiles with -mforce-l32 to:
```asm
2e:   0338            l32i.n  a3, a3, 0
32:   431b            addi.n  a4, a3, 1
34:   532b            addi.n  a5, a3, 2
36:   633b            addi.n  a6, a3, 3
3b:   402300          ssa8l   a3
3e:   413230          srli    a3, a3, 2
41:   1133e0          slli    a3, a3, 2
44:   0338            l32i.n  a3, a3, 0
46:   913030          srl     a3, a3
49:   743030          extui   a3, a3, 0, 8
4c:   402400          ssa8l   a4
4f:   414240          srli    a4, a4, 2
52:   1144e0          slli    a4, a4, 2
55:   0448            l32i.n  a4, a4, 0
57:   914040          srl     a4, a4
5a:   744040          extui   a4, a4, 0, 8
5d:   402500          ssa8l   a5
60:   415250          srli    a5, a5, 2
63:   1155e0          slli    a5, a5, 2
66:   0558            l32i.n  a5, a5, 0
68:   915050          srl     a5, a5
6b:   745050          extui   a5, a5, 0, 8
6e:   402600          ssa8l   a6
71:   416260          srli    a6, a6, 2
74:   1166e0          slli    a6, a6, 2
77:   0668            l32i.n  a6, a6, 0
79:   916060          srl     a6, a6
7c:   746060          extui   a6, a6, 0, 8
```
(Again, non-relevant interleaved instructions removed.)

My hope is that if the 32 bit reads were forced at a stage before optimisations occur, the code generated could look very similar to the first example - which would greatly reduce the performance hit where there is a lot of localised access to adjacent 8-bit values (ie string functions in particular).

I don't know anything about gcc internals though, so I have no idea how complex (or impossible) such a change would be - I'm guessing complex! I just thought I'd flag it while we're discussing the various options. When I have a moment I'll dig around and see what I can figure out myself, but I'm probably not up to implementing it.

I still want to play with the xtensa mforce-l32 option some more, as well. I haven't had a chance to run any code on actual hardware yet, I'll hopefully get a chance to try that tomorrow. Keen to hear what @pfalcon finds with micropython as well. Thanks again to jcmvbkbc for enabling all of this experimentation!

projectgus, 2015-06-14:
Derp, I don't know why I treated everything big-endian in that first example. Please ignore that snafu!

projectgus, 2015-06-14:
Hmm I should have thought about this more before I posted, sorry. Without knowledge about the alignment of a given `char *`, or the structure of a loop, the code can't possibly look as neat as the example I posted. It's more like this:

```
#define DR32(P) ( *((uint32_t *)(((intptr_t)(P))&~3)) >> (8 * (((intptr_t)(P))&3)) & 0xFF )

void using_uint32(void)
{
    const uint8_t *something = (uint8_t *)some_extern;
    printf("%d %d %d %d\r\n", DR32(&something[0]), DR32(&something[1]),
           DR32(&something[2]), DR32(&something[3]));
}
```

And -O2 doesn't optimise this any better than the -mforce-l32 method, the -mforce-l32 method actually produces less instructions.

I should know better than to have so much faith in compiler optimisations!

OTOH, if you know that the symbol is 4-byte aligned then you can probably hand-write functions that work on 4-byte chunks, and those will be quite efficient. Though hand-writing special cases is roughly what this discussion is about avoiding, I guess.

pfalcon, 2015-06-14:
>  If libc is built with -mforce-l32 then the performance hit applies to all use of libc functions.

Well, sure, looking the MicroPython's map file I easily spotted first victims - memcpy, strlen, etc, (uPy has own libc subset, coded of course mostly in naive way). And the only way around that is of course to have them be special version, working on word-sized data.

pfalcon, 2015-06-14:
> Compiling the libc made me realise that there are complications similar to @light's concerns, though.

@light's concerns (as far as I can read them) are my concerns with which I started, and why I proposed to have attribute, which can be applied to individual variable declarations. Note that I don't think it's worth implementing any longer, seeing how easy -mforce-l32 works. Well, to elaborate, of course I think it would be nice to implement it, but who would do it? I wouldn't ask @jcmvbkbc about it again, my assessment as it's not worth his precious time. -mforce-l32 is similar thing to -mlongcalls - it's useful to both vendors and end users, it's platform enablement features, allow to port existing software easily and maintain it for the platform without headache. Granted, it's not efficient, but -mlongcalls is not efficient either, but show me someone who doesn't use it and instead structures his link units to avoid them as much as possible and mark remaining functions manually? Only someone who does would have moral right to ask @jcmvbkbc for more fine-grained way to apply -mforce-l32, otherwise it's easy to predict that nobody will really bother to go thru all his variable and patch their declarations.

dpgeorge, 2015-06-14:
Personally I'm completely happy with just having -mforce-l32.  An increase of 3% in code size, and a bit of a performance hit, is well worth the approx 16k RAM that is reclaimed.

BTW, if you want more fine grained control you can separate out certain functions (eg strcpy_dram) into a file and compile that without -mforce-l32.

jcmvbkbc, 2015-06-15:
> -O2 doesn't optimise this any better than the -mforce-l32 

Adding peephole optimization here aimed at loads from consecutive addresses may help, as that's what's expected to happen often. I'll look at it.

> OTOH, if you know that the symbol is 4-byte aligned then you can...

It appears that the backend can find out the alignment of memory it's asked to load from, so there's a chance to generate shorter load sequences in these cases. I'll look at it as well.

> -mforce-l32 is similar thing to -mlongcalls

longcalls that appear to be reachable by mere call are folded by linker back to the short form, thus not making every call pay that price. OTOH force-l32 affects the whole translation unit and the compiler is in the best position to decide what code to use.
So now that you guys have at least one way to deal with that issue I'll try to improve it. Probably not that fast though, as I've never dealt with these areas before.

> Thanks again

You're welcome. It's a pleasure for me, actually, so thank you.

projectgus, 2015-08-10:
Experimented a bit with an exception handler to "fix" unaligned reads. Still at proof of concept stage. Branch includes an "experiment" program with some microbenchmarks:

```
Testing DRAM (0x3ffe9fb8) 'O hai there! %d %d %d'
Formats as: 'O hai there! 1 2 3'
 .. against                        null op:    14 instructions
 .. against           memcpy - aligned len:    52 instructions
 .. against         memcpy - unaligned len:    61 instructions
 .. against   memcpy - unaligned start&len:    74 instructions
 .. against                         strcpy:   103 instructions
 .. against                   naive strcpy:   179 instructions
 .. against                        sprintf:  2614 instructions
 .. against             sprintf format arg:   103 instructions
Testing IRAM (0x40100d30) 'O hai there! %d %d %d'
Formats as: 'O hai there! 1 2 3'
 .. against                        null op:    14 instructions
 .. against           memcpy - aligned len:    72 instructions
 .. against         memcpy - unaligned len:   340 instructions
 .. against   memcpy - unaligned start&len:   466 instructions
 .. against                         strcpy:   139 instructions
 .. against                   naive strcpy:  2885 instructions
 .. against                        sprintf:  9505 instructions
 .. against             sprintf format arg:   139 instructions
Testing Cached flash (0x4024e998) 'O hai there! %d %d %d'
Formats as: 'O hai there! 1 2 3'
 .. against                        null op:    14 instructions
 .. against           memcpy - aligned len:    72 instructions
 .. against         memcpy - unaligned len:   340 instructions
 .. against   memcpy - unaligned start&len:   466 instructions
 .. against                         strcpy:   133 instructions
 .. against                   naive strcpy:  2885 instructions
 .. against                        sprintf:  9505 instructions
 .. against             sprintf format arg:   133 instructions
Testing 'Uncached' flash (0x4024e998) 'O hai there! %d %d %d'
Formats as: 'O hai there! 1 2 3'
 .. against                        null op:  1512 instructions
 .. against           memcpy - aligned len:   969 instructions
 .. against         memcpy - unaligned len:  1058 instructions
 .. against   memcpy - unaligned start&len:  1370 instructions
 .. against                         strcpy:  1772 instructions
 .. against                   naive strcpy:  3421 instructions
 .. against                        sprintf: 15444 instructions
 .. against             sprintf format arg:  1772 instructions
```

My observations:
* Newlib uses aligned word operations whenever it can, so with aligned addresses & lengths IRAM performance is almost as good as DRAM (interesting it's not the same, but this might just be an artifact of NMI "jitter" in the benchmark).
* As soon as the length is unaligned then the performance blows out by a factor of ~4x, probably around 250 instructions.
* In the worst case, "naive" byte-to-byte operations are about 16x slower reading from IRAM.
* sprintf is about 3.5x slower if the format string is in IRAM (obviously the actual slowdown will depend on the format string itself).
* If the SPI flash cache is hit, performance is identical between SPI flash and IRAM (as expected).
* If the SPI flash cache is forced to miss, there looks like an overhead of about 600 instructions to "page in" to the cache.

There's definitely room to optimise the loader routine some more, but I think this looks like a good approach to keep following - at least for esp-open-rtos.

I haven't compared performance with -mforce-l32 yet. I also don't know if the "libc" routines in the Espressif SDK do the same kinds of optimisations as newlib.

Comments/patches/performance tips all welcome. :)

projectgus, 2015-08-10:
Probably the main optimisation for the loader routine is not to save all 16 general registers every time, refactor to use a minimal set of working registers and then use a jump table (?) to complete the load if the target register isn't one of the working registers.

projectgus, 2015-08-11:
Pushed a few more commits to https://github.com/SuperHouse/esp-open-rtos/tree/experiments/unaligned_load - now supports l16si, shaved some instructions - "naive strcpy" worst case is now ~13.2x slower from IRAM than DRAM.

Still need to make this work with DoubleException before trying to move all of .rodata into instruction memory.

projectgus, 2015-09-10:
Continuing discussion of rodata-in-IROM on the mailing list:
https://groups.google.com/forum/#!topic/esp-open-rtos/btUfyPyWOuU

projectgus, 2016-03-09:
Current status of this issue:

esp-open-rtos transparently loads bytewise data from flash using the unaligned load interrupt handler. There's some performance impact but otherwise it's fully transparent to the calling code.

The .rodata section is still loaded to RAM by the linker script, because there are some symbols in there that are accessed either before flash is mapped or from the NMI. There is a dirty hack in the linker script so the .rodata of certain files (like the cert.c file in the http_get_mbedtls example) are loaded to flash.

The remaining task before we can close this is to find the symbols (or object files) that are accessed in non-flash-mapped contexts and explicitly load those into a RAM .rodata section, so that the default .rodata can go to flash.

Whatever mechanism we use for loading .rodata into RAM should also be documented so people with constant data that they don't want stored in flash (for performance reasons) can mark those easily.

commit d72aedf7 (2016-05-28): Store .rodata in flash by default

Closes #11
closed 2016-05-28 by projectgus