SuperHouse/esp-open-rtos #15: About local.h

kanflo, 2015-07-23:
I proposed the following change to esp_iot_rtos_sdk wrt wifi SSID and password management. I find it rather attractive as there will be a file containing dummy SSID/pw following a git clone and changes to that file will be ignored by git. I'd be happy to incorporate that change here if more people than me likes the idea :)

https://github.com/kanflo/esp_iot_rtos_sdk/commit/3d0f7be119388ebdc3517a534b3adb2f895e59b0

projectgus, 2015-07-26:
Hi,

Sorry for the slow reply on this, I wanted to think it over before I replied. It's an interesting approach, I never thought about keeping the file in git like this and just telling the local client to ignore it.

I am a bit uncomfortable with the way local.h is implemented at the moment - you may have seen already, but the makefile checks if it's present and then adds it to the preprocessor flags:
https://github.com/SuperHouse/esp-open-rtos/blob/master/common.mk#L246

... which is kind of icky "magical behaviour", which I don't like. And the makefile is already plenty complex and doesn't really need more weird one-off sections! This approach seems more explicit, which I like.

The other nice thing about it is that "PROGRAM_DIR/include" is placed ahead of root/include/ in the include flags, so if you want per-program ssid_config.h then you can just create one and put it there, and the search path will find it before the global one.

So, yes - I'd happily accept a PR that removed local.h and used this instead. Thanks!

Angus

kanflo, 2015-07-28:
Great, what do you think of this? https://github.com/SuperHouse/esp-open-rtos/pull/16 
I like that you can have separate credentials for  different projects.

Johan

projectgus, 2015-08-05:
(Closing this issue as it's covered by #16)

closed 2015-08-05 by projectgus