SuperHouse/esp-open-rtos #18: sdk_uart_rx_one_char crashes

kanflo, 2015-08-04:
Calling 
```
int ch = sdk_uart_rx_one_char();
```
leads to a crash inside sdk_uart_rx_one_char (if I understand the reboot log correctly):
```
s8i     a6, a2, 0
```
Since this is an SDK provided function I guess something has not been set up correctly.

projectgus, 2015-08-04:
Hi Johan,

Thanks for reporting this. It seems likely that's the case. Will take a look, hopefully later this week.

Angus

kanflo, 2015-08-05:
Your'e welcome as always :) I am thinking we could ditch ```sdk_uart_rx_one_char``` altogether in favor of IRQ driven RX. I have completed support for it but would like to check with you where to put it.

I use a circular buffer for RX and a counting semaphore making the caller block when there is no data available. Counting semaphores require ```configUSE_COUNTING_SEMAPHORES``` which currently is zero. It also requires an init function for setting up the circular buffer and UART RX IRQ. I could place it in ```newlib_syscalls.c```and check if it has been init'ed in every call to ```_read_r``` but rather not. Do you have any thoughts on this?

foogod, 2015-08-05:
I just took a quick look at the disassembly for uart_rx_one_char from the SDK, and it looks like you may be invoking it wrong..

It appears its prototype should actually be:

int sdk_uart_rx_one_char(char *buf);

(that is, it takes as its first argument a pointer to a char address to store the result in, and then always returns 0)

I'm not really sure why they implemented it that way, as admittedly it does seem counterintuitive..

foogod, 2015-08-05:
Oh, my mistake, I see what they're doing now..

uart_rx_one_char actually returns 0 on success, and 1 if there was nothing in the FIFO to retrieve.

projectgus, 2015-08-06:
>It appears its prototype should actually be:
>
> int sdk_uart_rx_one_char(char *buf);

Sorry, that signature was added by me not Espressif. I'm not sure where I got it from but I obviously never tested it!

If someone can confirm this then I'll update the header file in esp-open-rtos as well.

>  I am thinking we could ditch sdk_uart_rx_one_char altogether in favor of IRQ driven RX. I have completed support for it but would like to check with you where to put it.

Awesome, I agree this is a great long term solution.

It'd be nice to eventually remove all the SDK UART functions, and just keep in stubs for any SDK functions that use the UART themselves (this is already 99% of the way there already as printf/ets_printf from the SDK calls directly into newlib).

(That's a longer term goal that might depend on #2, though, as the [SDK startup code calls os_install_putc1](http://esp8266-re.foogod.com/wiki/User_start_%28IoT_RTOS_SDK_0.9.9%29)).

> Counting semaphores require configUSE_COUNTING_SEMAPHORES which currently is zero

This is fine by me. It'd be nice to have a non-blocking function as well (similar to the current uart_rx_one_char) that just "peeks" the UART buffer, but I'm guessing you've thought of that too.

>  It also requires an init function for setting up the circular buffer and UART RX IRQ. I could place it in newlib_syscalls.cand check if it has been init'ed in every call to _read_r but rather not.

The list of things that need to be explicitly initialised is growing! I think my preference is that you do it this way for now, and when #2 is done (I hope soon) we can move it into the startup code along with a bunch of other things that need to be initialised pre-`main()`.

The other interesting thing that I'd like to support at some point in the future is uart1. Just throwing that out there though, I don't expect you to add it!

foogod, 2015-08-06:
The IRQ-driven UART stuff sounds useful for some applications, but also sounds like unnecessary bloat for others.  I'd very much prefer if this sort of thing was included but optional, instead of always initialized on startup for everything..

(This is actually one of my pet peeves with the current SDK:  The startup code automatically pulls in routines from 31 different files in 6 different libraries before even invoking any user app code..  as a result, the simplest "hello world" program (which doesn't even use any net functions) takes up a minimum of 200KB of program code.)

kanflo, 2015-08-06:
> The IRQ-driven UART stuff sounds useful for some applications, but also sounds like unnecessary bloat for others.

You have a point there. The IRQ-driven UART should be considered a 'driver' just as there might be a driver for SPI TFT displays and we would not want to include that in every build. 

> The list of things that need to be explicitly initialised is growing! I think my preference is that you do it this way for now.

As there is no driver concept in e-o-r I will look at adding weak references in ```_read_r``` and initialize the driver if it was added to the project. Does that sound ok @projectgus , @foogod?

> It'd be nice to have a non-blocking function as well (similar to the current uart_rx_one_char) that just "peeks" the UART buffer, but I'm guessing you've thought of that too.

Yup!

kanflo, 2015-08-06:
> uart_rx_one_char actually returns 0 on success, and 1 if there was nothing in the FIFO to retrieve.

Thanks for the clarification @foogod! I am however getting very strange behavior with this code:
```
    while(1) {
        char ch;
        while (sdk_uart_rx_one_char(&ch)) ;
        printf("'%c'\n", ch);
    }
```
The MIPS burned in the while loop apart, there seems to be some kind of buffering going on as I eventually receive the characters I type, although the function doesn't return 1 for every character typed.

I will yield on this one and lay my time on the IRQ driven UART driver :)

kanflo, 2015-08-06:
The new prototype is ready to be pulled. Even though sdk_uart_rx_one_char stil does not work as expected it does not crash. I will close this issue and open a new one for discussion about how to include the IRQ driver UART driver.

closed 2015-08-06 by kanflo

foogod, 2015-08-06:
Hmm, that's odd..  the uart_rx_one_char code from the SDK is actually very simple, and just does (literally) this:

```
int uart_rx_one_char(char *buf) {
    int a3, a4;

    a3 = READ_PERI_REG(UART_STATUS(0)) & 0xff;
    a4 = 1;
    if (a3 != 0) {
        a4 = 0;
        *buf = READ_PERI_REG(UART_FIFO(0));
    }
    return a4;
}
```

I agree, though, it's probably not worth pursuing too much further at this point..  if somebody actually wants to use this particular routine instead of the IRQ-based stuff for some reason, and then still can't get it to work, we can always revisit..

projectgus, 2015-08-07:
>As there is no driver concept in e-o-r I will look at adding weak references in _read_r and initialize the driver if it was added to the project. Does that sound ok @projectgus , @foogod?

Something like this sounds good to me.

Once we have "proper" startup code, this should actually be easy to have linked automatically. First the _read_r syscall needs to go in its own object file, which contains the _rear_r function and a _read_r_uart_init method marked with __attribute__((constructor)).

If you link something which calls read() -> the linker will find the _read_r() syscall, loads its the object file, and adds the constructor function to the list of functions to call during startup.

If you don't link something which calls read() -> the linker never pulls in the syscall object file, so nothing related to the UART IRQ driver gets linked into the binary at all.

In the end it might turn out to be more useful having explicit initialisation rather than "magic" constructors, but I think this would also work well. _After_ we have open source startup code, that is!