SuperHouse/esp-open-rtos #19: Problem compiling examples

jpbarba, 2015-08-05:
I get this error when compiling examples. I have download the source code and compiler is in PATH

../../include/espressif/esp_misc.h:9:26: fatal error: lwip/ip_addr.h: No such file or directory
 #include "lwip/ip_addr.h"

projectgus, 2015-08-05:
Hi jpbarba,

esp-open-rtos uses "git submodules" to manage LWIP and some of the other third party components. This means you need to do a `git clone --recursive` when first getting it, you can't click the "download" button (you won't get all the components.) The README on the main page has the steps to follow.,

If you did a git clone already but didn't add "--recursive", you can run `git submodule init` then `git submodule update`.

Let me know if you're sure you have the submodules already (for example, the directory `-root-/lwip/lwip` shouldn't be empty.

Angus

jpbarba, 2015-08-05:
Hi Angus
You are right, but now I get this error. Its like a include problem. I have clone with recursive option and I have xtensa in PATH too

CC /home/esp8266/esp8266/esp-open-sdk/esp-open-rtos/examples/simple/simple.c
In file included from ../../include/xtensa_interrupts.h:14:0,
                 from ../../core/include/esp/gpio.h:15,
                 from ../../core/include/esp8266.h:18,
                 from ../../FreeRTOS/Source//portable/esp8266/portmacro.h:73,
                 from ../../FreeRTOS/Source//include/portable.h:318,
                 from ../../FreeRTOS/Source//include/FreeRTOS.h:87,
                 from /home/esp8266/esp8266/esp-open-sdk/esp-open-rtos/examples/simple/simple.c:5:
../../FreeRTOS/Source//portable/esp8266/xtruntime.h:29:32: fatal error: xtensa/config/core.h: No such file or directory
 #include <xtensa/config/core.h>
                                ^
compilation terminated.
make: *** [build/program//simple.o] Error 1

projectgus, 2015-08-05:
Thanks for reporting this. What toolchain are you using, esp-open-sdk or something else?

On my system xtensa/config/core.h seems to be supplied by the crosstool-NG toolchain built by esp-open-sdk. However maybe the paths aren't quite set up right in esp-open-rtos.

kanflo, 2015-08-05:
@jpbarba, you might have tried this already but building the toolchain with ```make STANDALONE=n``` is vital. My original toolchain was built the other way around resulting in a lot of compiler errors (cannot remember exactly what though).

jpbarba, 2015-08-05:
Im gonna try. It'd be good to have a virtual machine with this stuff.

Thanks

projectgus, 2015-08-21:
I ran into this exact error (`xtensa/config/core.h`) earlier myself setting up Travis, and it's caused by the lx106 libhal headers not being installed as part of the toolchain.

esp-open-sdk should build libhal and install the headers. If you're using a toolchain from somewhere else then it might not have them, and this will be the problem.

If the esp-open-sdk build process errors out part way, you might end up with a situation where you have a compiler but not libhal - which will give this error as well.

foogod, 2015-08-21:
Might be good to start a "troubleshooting" page on the project wiki or something for "if you get this error, check this thing" kind of notes..

projectgus, 2015-08-24:
Good suggestion, done:

https://github.com/SuperHouse/esp-open-rtos/wiki/Troubleshooting

I'm going to close this issue for now, @jpbarba, but please reopen if you think that wasn't the problem you were experiencing.

closed 2015-08-24 by projectgus