SuperHouse/esp-open-rtos #25: I2C Communication

mikejac, 2015-08-20:
Hi,

AFAIK, the ESP8266 does not have hardware support for the I2C protocol. Thus it must be implemented in code ('bit-banging').

If I'm correct, any thought on how to implement that? Timing critical, and don't really wan't to disable interrupts while doing I2C...

/michael

PS. Is there someplace else where we can sort of discuss instead of creating an Issue? Google Groups, Google+, other...

kanflo, 2015-08-20:
Luckily i2c isn't timing critical and can easily be implemented using bit banging with IRQs on.
Even more luckily, there is already an i2c driver for this project: https://github.com/kanflo/esp-open-rtos-driver-i2c and on my todo list is to add it to ```esp-open-rtos/extras```.

As for where to discuss these matters, I would guess Stack Overflow when it comes to (possible ESP8266 related) i2c details.

mikejac, 2015-08-20:
Great! :-)

The SCL and SDA pins can be (almost) any pins one want's to use? I.e. not fixed to 0 and 2.

The "discuss" part wasn't meant to be specific for I2C - I was thinking about all things related to esp-open-rtos ...

kanflo, 2015-08-20:
That's correct, you can select any pins you like.

Oh, my missunderstanding. I think Github issues are perhaps not the best place for general discussion, a Google Group would probably be better. Any ideas @projectgus?

foogod, 2015-08-20:
Actually, I believe the initial assumption is incorrect.  There are indications that the ESP8266 does indeed support hardware I2C (No bit-banging is required).  There are several routines in the ROM which appear to be related to this (rom_i2c_readReg / rom_i2c_writeReg / etc).  Newer versions of the SDK appear to actually have a whole set of "i2c master APIs" listed in the SDK programming guide, but it looks like unfortunately the 0.9.9 version esp-open-rtos is based off of does not include those routines (so we'll probably need to reverse-engineer them sooner or later).

foogod, 2015-08-20:
Actually, I just realized that the "i2c master" routines from Espressif aren't built into the SDK libraries, but are just supplied in source-code form under the examples/driver_lib/driver/i2c_master.c directory, so if you don't mind the license-tainting, it might not be too hard to copy those out of a newer version of the SDK and fix them up to work with esp-open-rtos..

(Unfortunately, we can't include them directly in esp-open-rtos due to licensing issues..  We'll probably still want to eventually come up with some of our own..)

mikejac, 2015-08-20:
Good enough for me. Just ordered some MCP23016 I/O expanders with I2C communication. Think I'll try @kanflo's library when they get here.

pfalcon, 2015-08-20:
According to @jcmvbkbc, bootstrap sequence of esp8266 involves programming PLLs over I2C, so there's definitely hardware I2C, and maybe even few blocks (buses) if hw I2C is multiplexed over GPIO.

As for bit-banging, one boring things about it is that it needs to be reimplemented for each new MCU. Trying to abstract it with means C offers (function calls) immediately leads to performance hit. Maybe acceptable for low-speed I2C, but getting noticeable for SPI, and you want be able to bit-bang USB with such approach at all. Can be cured with metaprogramming techniques, with the most available language being C++. That's the idea behind PeripheralTemplateLibrary I mentioned recently. Here's I2C bit-banging protocol implementation for example: https://github.com/pfalcon/PeripheralTemplateLibrary/blob/master/include/i2c.hpp . Instantiated with suitable GPIO template class for a particular MCU, it allows to achieve assembly-level efficiency (while staying completely platform-independent).

projectgus, 2015-08-21:
> I think Github issues are perhaps not the best place for general discussion, a Google Group would probably be better. Any ideas @projectgus?

Yes, I think you're probably right - we're hitting limits of the Issues system for discussions. I'm currently tossing up between setting up Discourse, registering a Google Group, or maybe just firing up a http://gitter.im chatroom (the esp8266 Arduino project has one and it's quite a good place, you can connect via IRC if you don't like web clients). If you all have any preferences for any of these (or some fourth option), let me know.

foogod, 2015-08-21:
I do also agree that Github wasn't meant to be a discussion forum, despite the fact that some people try to use it that way.

A chatroom might be handy, but I don't think it's a substitute for an actual maillist/forum of some kind..  Particularly with this kind of subject there are often longer-running or more in-depth discussions that don't really suit a chatroom environment (and, for example, due to the nature of my work I'm not always able to be hanging around paying attention to an online chatroom all day, but I can periodically check a forum or email)..

foogod, 2015-08-21:
FYI, back on the original subject of this issue:  I realized when I looked at the Espressif I2C driver in more detail that it's apparently just doing software bit-banged I2C too, so maybe there isn't any hardware I2C support built into the chip after all.  (Does make one wonder what those ROM routines are doing, though.. I may have to investigate that soon)

kanflo, 2015-08-21:
I agree with @foogod, a chat room probably isn't well suited. The signal to noise ratio seems poor if you want to follow a single subject :)

mikejac, 2015-08-22:
Discourse (discourse.org) looks nice. Never used it though...

foogod, 2015-09-02:
@mikejac is the original issue of this ticket taken care of to your satisfaction?  If so, I'm thinking we might as well close this issue..

mikejac, 2015-09-02:
@foogod yes it is, go ahead and close this issue.

If I create a group on discourse.org will you visit? :-)

mikejac, 2015-09-02:
Forget the part about discourse.org - it's not what I thought it would be...

foogod, 2015-09-02:
Cool, I'll close this then.. if we want to discuss forum options further, I've opened a separate ticket to track getting something set up for that..

closed 2015-09-02 by foogod