SuperHouse/esp-open-rtos #26: Create test case app(s)

foogod, 2015-08-20:
The more we start hacking on the internals of the existing SDK, the more useful I think it would be to have one or more standard test case apps we can use to try to make sure we don't break existing behavior.  Ideally, I'm thinking a single, reasonably complex application which attempts to exercise most of the types of functionality people currently use which could be compiled and run, and then a testing procedure designed to make sure it does actually function as intended.  That way anybody who wants to has a well-defined procedure to test that at least basic functionality still works as expected on any given esp-open-rtos build.

Unfortunately, all of the current examples are fairly limited in scope for this purpose..  It might be good to start by finding an existing app which exercises a reasonable range of behavior, and then work from there.

projectgus, 2015-08-21:
Agreed.

I'm taking some baby steps in this direction at the moment setting up Travis, which will at least confirm that all the examples & experiments build on each commit.

That's obviously a much weaker condition to verify than "basic functionality still works", though. It'd be better to verify that!

Having something where anyone can run some kind of automated test script might be interesting for doing actual CI, I can leave an ESP8266 board and a little router (Raspberry Pi maybe?) in a corner somewhere and have it go through some motions on each commit, check for certain strings in the serial output. If the hardware setup is reasonably well defined then other people can make their own local test setups as well...

A manual test case program is a good first step in this direction, anyhow. As we start adding more third party libraries it should become easier to put something reasonably complex together without a lot of program code (for instance: fire up MQTT, and a web server, and some kind of i2c peripheral all concurrently and check for defined behaviour).

jcmvbkbc, 2015-08-21:
> single, reasonably complex application which attempts to exercise most of the types of functionality people currently use which could be compiled and run, and then a testing procedure designed to make sure it does actually function as intended.

Seems to me that a number of small independent simple tests would be easier to maintain and debug, than one big and complex.

foogod, 2015-08-21:
> Seems to me that a number of small independent simple tests would be easier to maintain and debug, than one big and complex.

In most cases, I would agree, but there were two reasons I didn't suggest that approach here (at least for the initial way of doing things):

1. Somebody would have to write all the individual test cases
2. Without some sort of automated testing setup, I think having to build, flash, and manually test a large series of individual ESP8266 firmwares every time we wanted to check we hadn't broken anything would get extremely tedious. (And because of the nature of the architecture, getting an automated testing framework (especially one that can test a lot of the things we actually need to test) isn't just going to be a matter of spinning up some build-server software somewhere..)

A hybrid approach might be doable:  Create a bunch of independent test case code samples which then get compiled into a single ESP8266 app which runs through them in a semi-automated fashion..  Of course that still requires writing and verifying test cases for everything we want to test..

projectgus, 2016-02-22:
There are the makings of an automated test infrastructure in this branch:
https://github.com/SuperHouse/esp-open-rtos/tree/feature/tests

Some discussion here:
https://groups.google.com/forum/#!topic/esp-open-rtos/Kgzbpw07jDM

Once I have a Jenkins server up (and some documentation) I'll merge to master.