SuperHouse/esp-open-rtos #28: Separate library build from app build

foogod, 2015-08-21:
The makefiles are currently set up to build all of the libraries as part of building each app (using the app's own build directory for the results, etc).  This means that, for example, if you build all of the examples, you end up recompiling the same libraries 9 times.

The makefiles should be set up so that the libraries are compiled into their own destination directory, which only needs to be built once, and then is just linked in by any application build which needs it. 

(Eventually, we should probably also add a top-level makefile with build/install targets for those people who just want to download and build esp-open-rtos and then just reference the prebuilt libs as a normal build dependency from their own (external) project directories/makefiles.)

kanflo, 2015-08-21:
That is a good point. One issue to be solved is how to treat different configurations of the libraries as done in https://github.com/kanflo/esp-open-rtos/blob/master/examples/terminal/FreeRTOSConfig.h
Here I needed a FreeRTOS feature not normally used and thanks to the e-o-r make system it was easy to reconfigure the FreeRTOS build for this app.

foogod, 2015-08-21:
Also a good point..  Maybe we should keep both ways of doing it and let the user switch between them with a config variable in the app makefile..

It seems to me we might want even more granularity, though..  For example, just because you're building the FreeRTOS core with an extra feature doesn't necessarily mean you should have to recompile lwIP, axTLS, etc..

projectgus, 2015-08-22:
The "everything builds together" behaviour was originally a design decision based on my personal preferences when doing microcontroller stuff.

You make some really good points though, especially about integrating with third party build systems. I don't usually use IDEs but I'm guessing that integrating esp-open-rtos into Eclipse, for example, would be a total pain.

FWIW it is quite possible to have third party programs that build totally out of tree (we have private SuperHouse firmwares that are implemented like this). Limited to sticking with the esp-open-rtos Makefile system, of course. I should add a wiki page explaining how to do this (it's really just changing the path where you include common.mk from).

Another minor note is that setting "CROSS=ccache xtensa-lx106-elf-" will dramatically speed up building all examples at once, because most of them have the same library-visible headers. Although this is totally a crutch.

Here's what I think the pros/cons of the currrent approach are:

Pros
* Build examples or new programs after a single git clone.
* Easily make changes to library-level component configuration or compiler settings and have them propagate to all effected components.
* Fairly easy to override settings without needing to dig through the Makefile to do it (may need more documentation to make this clear to people other than me).

Cons
* Locked in to one fairly obscure and non-standard build system, with lots of non-recursive make gibberish in it.
* Unnecessary recompiling of files when top-level config headers may or may not have significantly changed.
* Most config changes at the "library" level will probably fail at link time (ie missing functions) even if they didn't cause the client code to recompile.
* Changing the Makefile causes everything to rebuild, which is a pain if you're not actually changing the compiler command line.

... I think there definitely is a better approach that keeps most of the Pros while losing a lot of the Cons. Very open to suggestions/patches!

At first thought, one of the significant changes for "split libraries" is probably a big reorganisation of the directory layout, for a single "include" directory similar to how Espressif (and most C-based projects) do it. Which is perfectly fine, it just means some headers which are currently only stored in git submodules will need to be copied into the tree.

foogod, 2015-08-22:
I don't think there's any reason we can't keep all those pros while making the build design a bit more modular..

I think it's important to recognize that this is going to have to be an evolutionary process.  As a first step, what I'd suggest is just changing the makefiles to have a different "build dir for application code" (relative to app dir) vs "build dir for SDK/OS libs" (default relative to esp-open-rtos root, and thus shared by different apps).  That way the lib build process will still automatically be pulled into the application build if necessary, but once they're built once, successive app builds, even of different applications, don't need to rebuild them (unless the app makefile overrides the lib build dir location (for the custom-config-of-FreeRTOS situation, etc)).

I had been thinking about suggesting a reorganization of the whole project directory tree anyway (it would make a lot more sense to me to move most of the Espressif SDK stuff into its own subdirectory, for example, and maybe move core/include up to be the top-level /include, etc), but haven't had a chance yet to put together a real proposal for that..

As far as the header locations, etc, what we probably ultimately want is a "build the OS" target, which builds all the libs and then puts them into "libs/" and also copies any needed public headers out of their individual module locations in to a common "include/" dir.  It would then become trivially easy to also make a 'build install' target, that just copies the contents of libs/ and include/ to an appropriate system location (i.e. '/usr/local/esp-open-rtos' or some such)

projectgus, 2015-08-23:
This all sounds good to me. "Building" the header files to an include directory is an easy solution.

If you want to leave this with me then I'm happy to have a go at it, put something up for review?

foogod, 2015-08-23:
Sounds good to me (you're more familiar with the makefiles anyway) :)

projectgus, 2015-10-13:
Quoting @foogod from [this mailing list thread](https://groups.google.com/forum/#!topic/esp-open-rtos/MC6zJyIwZjM):
> I do like the idea of having /include eventually become "the one true                                                 
> include directory for app builds", which would mean eventually it would
> also contain a bunch of FreeRTOS headers, and relevant stuff from all the
> other modules (presumably copied there during the library build)..  This
> would require changes to the build process which I think are currently a
> bit beyond me, though..  (Likewise, I think it'd be good to eventually move
> /lib somewhere else and turn /lib into a "library build output directory"
> for compiled libraries, so that if one wanted to build an app from
> "outside" our build environment, all they need to do is... *snip*

+1 for this.

I haven't forgotten about this issue, but there are blocker issues I want to close first so that the Makefile's overall complexity can go down rather than up (it's already too complex):

* #58 Assemble binary SDK source files in place
* #38 Linker script needs simplification

Once these are done it should be pretty easy to have a dedicated library build as the default (with the option of per-compile library rebuilding if you want to tweak settings for flexibility, resources, size, whatever.)

DagAgren, 2015-11-12:
As an entirely unsolicited opinion, I also prefer the approach of just building everything in one go and in one place. It keep everything conceptually simple and easier to modify and reuse, especially in cases where you want to make big changes.

DagAgren, 2015-11-12:
Also, I'd like to be able to use the SDK by just putting it all in a subdirectory of my project, and adding a list of its .c files to my own Makefile, and include paths to my CFLAGS, and be done with it. This seems somewhat easier if the SDK is already doing pretty much the same.

projectgus, 2015-11-12:
> As an entirely unsolicited opinion, I also prefer the approach of just building everything in one go and in one place. It keep everything conceptually simple and easier to modify and reuse, especially in cases where you want to make big changes.

Unsolicited opinions are fine if you're interested in using esp-open-rtos. :)

> Also, I'd like to be able to use the SDK by just putting it all in a subdirectory of my project, and adding a list of its .c files to my own Makefile, and include paths to my CFLAGS, and be done with it. This seems somewhat easier if the SDK is already doing pretty much the same.

This might get unwieldy, I think. There are a lot of source files, including C and assembler source, and the exact list changes fairly regularly. Some internal "components" are built with private header search paths and macro definitions that are required to build them but not for any code linking to them. At best compiling everything with the same CFLAGS would be a very long list of CFLAGS, including a lot of cruft. At worse there will be namespace conflicts and it won't compile at all.

How would it suit to be able to call out from your makefile to a target inside esp-open-rtos to "update libraries if needed"? And for your own code the makefile just links to the built esp-open-rtos libraries and has the correct include path for the header files?

(In case it's not clear, you can currently use the existing makefile system to build-out-of-tree or to place the esp-open-rtos directory as a subdirectory of your program. I have projects that use it like that. And you can also override CFLAGS in your Makefile for the "library" components, before you include common.mk.

You do have to follow the conventions that the other esp-open-rtos Makefiles use though - for how you list source files for the build, etc. Which is a part that we want to fix down the track.)

DagAgren, 2015-11-13:
> Unsolicited opinions are fine if you're interested in using esp-open-rtos. :)

Seems I am. Usually I try to stay as close to the source as possible for these things and avoid third-party systems, but I've now tried working with both SDKs from Espressif, and they are just too terrible. The whole time I just wanted to rip everything out and rewrite everything.

I'm going to be trying to use I2S next, so I might contribute register headers for that and SLC, unless you already have those nearly done.

> This might get unwieldy, I think. There are a lot of source files, including C and assembler source, and the exact list changes fairly regularly. Some internal "components" are built with private [...]

It's OK, I won't be upgrading constantly. I already got this set up now, actually, and I didn't seem to need any macros at all, but I'm not using any of the extras yet. The list of files isn't too unwieldy (except I did cheat with $(wildcard) in a few places).

The biggest pain is the gigantic list of include paths, and preprocessing the linker script and SDK libraries. I see you've fixed the linker script, which will make things much smoother, and I ended up just doing the libraries in a separate script, and checking the results into my repo.

> How would it suit to be able to call out from your makefile to a target inside esp-open-rtos to "update libraries if needed"? And for your own code the makefile just links to the built esp-open-rtos libraries and has the correct include path for the header files?

That could be a compromise if things get out of hand. Especially if it put all libraries and headers in one place.

> (In case it's not clear, you can currently use the existing makefile system to build-out-of-tree or to place the esp-open-rtos directory as a subdirectory of your program. I have projects that use [...]

> You do have to follow the conventions that the other esp-open-rtos Makefiles use though - for how you list source files for the build, etc. Which is a part that we want to fix down the track.)

It's a bit too magical. I try to avoid magic in my build system at all costs, I want it simple and easy to follow along when something breaks or needs changing, so I prefer my own. Makes it easier for others to replace the build system too if they want.

(My philosophy is that if you allow too much cleverness in the build system, you start relying on it, and understandability of your system suffers, which leads to poorer code.)

foogod, 2015-11-13:
> I'm going to be trying to use I2S next, so I might contribute register headers for that and SLC, unless you already have those nearly done.

I'm pretty sure nobody else has gotten around to tackling those yet, so please do feel free! :)

> The biggest pain is the gigantic list of include paths, and preprocessing the linker script and SDK libraries

We're actually working on fixing all of those issues already.. The SDK lib preprocessing should go away once we switch over to rebuilding them all from C / assembler source, which is currently a work in progress..  Also, once we get the reverse-engineered version of libmain into the mainline source, we should be able to remove a bunch more of the complexity/preprocessing in the linker scripts, as we'll have a lot more control over what's actually inside the libraries (most of the current complexity is just to workaround limitations in the binary-only Espressif libs).  I had already proposed a start on a scheme to work toward consolidating and better organizing the include directories (https://groups.google.com/d/msg/esp-open-rtos/MC6zJyIwZjM/NmVsrkUOBgAJ) but just haven't had time to really do any of the work yet..

However, fixing the include directory situation is actually part of the reason I want to try to separate the esp-open-rtos build from the application build process, because currently the headers you potentially need to include from an app are strewn across a bunch of different submodules (many of which are not ours).  My current plan is for part of the esp-open-rtos build to copy all the include files that are needed by apps into one common include directory as part of the "build" of the OS libs, so that apps don't have to include tons of different dirs when they compile their own stuff, but in order to do this the esp-open-rtos build process needs to happen (at least somewhat) separately from the app build process.

>> How would it suit to be able to call out from your makefile to a target inside esp-open-rtos to "update libraries if needed"? And for your own code the makefile just links to the built esp-open-rtos libraries and has the correct include path for the header files?

> That could be a compromise if things get out of hand. Especially if it put all libraries and headers in one place.

That is the plan, really.  What I would like to aim for is:
* You can stick the esp-open-rtos sources anywhere relative to your app build directory and it doesn't care
* You don't need to use esp-open-rtos's Makefiles/etc to build your app at all (but you can if you want to, at least for simple things, so it's easy for folks to get up and running quickly)
* You can either manually build esp-open-rtos separately, and then just reference its libs/includes from your project, or you can have you Makefiles (or whatever) call esp-open-rtos' to automatically build/update the OS libs as part of your build.
* esp-open-rtos' Makefiles will build everything into one 'lib' dir and one 'include' dir (you can either use its default or tell it where to put them).  Then all you need to do when building apps is add those dirs to the include path and libpath (and link in the libs your app uses, of course).

Regarding "magical" build scripts:  We all agree that the current ones are a bit too magic, and have plans to try to clean some of that up (but it's a question of priorities and manpower, really).  On the other hand, it is possible to go too far the other direction.  Making things too simple can result in a project that's impossible to maintain as it grows and doesn't allow the flexibility to support multiple users' use-cases.  It's important to find a balance.

This is another reason, however, why I believe it's good to compartmentalize and separate the build processes for different parts of things, so that you don't have to overcomplicate your app build scripts with all of the stuff necessary to build esp-open-rtos correctly, and esp-open-rtos doesn't need to overcomplicate its system to support building all kinds of different apps in addition to the core OS libs.  Each piece should pick one thing, do it right, and do it cleanly instead of trying to have a one-solution-for-everything monster build system that nobody really understands.

projectgus, 2015-11-13:
FWIW I agree with everything in @foogod's post. :)

DagAgren, 2015-11-13:
That all sounds quite reasonable. You do seem to have your priorities quite straight with this project, which makes it a lot more appealing to use.

resetnow, 2016-08-07:
I have tried to use esp-open-rtos with Eclipse CDT and as good as it is (I did the same for Espressif RTOS SDK and it was a nightmare to get it working right) one of the things that could be improved is resolving circular dependencies between the libraries. For some reason Eclipse does not provide a way to link a group of libraries (--start-group / --end-group), which leads to some nasty hacks like putting list of the libraries into a compiler command template. Maybe combining some of binaries into one (<a href="http://stackoverflow.com/a/3821949">using ar</a>, for example), would help.

Another thing I have noticed is that <a href="esp-open-sdk">https://github.com/pfalcon/esp-open-sdk</a> already ships the lwip library which sometimes leads to conflicting includes. I was under the impression that many people who use esp-open-rtos also use esp-open-sdk; what would be the best way to avoid lwip conflicts?

mhanuel26, 2016-09-13:
Hello resetnow,

Would you mind to share how you integrate esp-open-rtos under eclipse?
I have done basic integration of Espressif NONOS SDK, but have read from you and others here that is a nightmare.

I really will appreciate your response,

resetnow, 2016-09-13:
@mhanuel26 you can either add your esp-open-rtos project as a “Makefile project with existing code” but if you want to use CDT builder (which eliminates the need to edit makefiles manually) I have created a gist <a href="https://gist.github.com/resetnow/5d2eaea0588ffdbd150616acda1eb51c">here</a>. It would still be a pain to set up, you could edit .project file by hand though and then reuse it.

Sorry for the comment spam.