SuperHouse/esp-open-rtos #37: PWM output

blipi, 2015-08-27:
First of all, I do not know if this is the correct place to post this (is there a forum, mailing list, w/e?). Sorry in case it is not.

I've been trying to understand how the GPIO works here, and so far it is quite intuitive. But when it comes to PWM, I have no clue on how to enable it.
I successfully did so on the bare Espressif SDK, but had no luck so far with this RTOS.

Any help would be really appreciated.

Thank you!

projectgus, 2015-08-28:
It's the correct place for now. There is no forum or mailing list yet, but I'm working on it...

PWM is not something that the ESP hardware does automatically, it's implemented in the Espressif SDK using a timer interrupt and some software.

esp-open-rtos doesn't have this yet. :( Someone needs to write a PWM implementation that uses one of the timer interrupts, and contribute it under `extras/pwm` or similar.

Sorry! I'm going to keep this issue open as a feature request though, because it's something we definitely should have but don't.

blipi, 2015-08-28:
Oh I see, if that's the case let me take a look at Espressif's implementation and see if I can get something to work.

I can't really test now, as I don't have the ESP with me, but I'll keep you updated in case I do any progress.

Don't worry, it's nice to have such support and effort put in this project ;) I'll try to contribute too.

projectgus, 2015-09-18:
I'm going to close this issue and move discussion to the PR in #40

closed 2015-09-18 by projectgus