SuperHouse/esp-open-rtos #39: TLS handshake failed

hawkhsieh, 2015-08-28:
I have tried the http_get_ssl example, and the result is below:
```
ip:10.0.0.11,mask:255.255.255.0,gw:10.0.0.1
DNS lookup failed err=202 res=0
top of loop, free heap = 29932
Running DNS lookup for www.google.com...
DNS lookup succeeded. IP=74.125.23.147
... allocated socket
... connected. starting TLS session...
initial status 0x3fff9610 -261
SSL handshake failed. :( -261
```
Please give me a hand. Thank you.

foogod, 2015-09-02:
Hmm.. according to axtls/ssl/ssl.h, error code -261 is `SSL_ERROR_INVALID_PROT_MSG`, which suggests that it got a SSL message packet it didn't like for some reason.  Unfortunately, it looks like this error can be triggered in several different cases (including just getting a packet type it doesn't recognize)..

I haven't personally ever gotten around to trying the http_get_ssl example, but when I have a chance I'll see if I can reproduce your issue and hopefully get a bit more insight as to what might be going on..

projectgus, 2015-09-28:
Hi @hawkhsieh,

axTLS had some problems with modern cipher suites, which I think included this problem as google.com were pretty strict about not accepting old/outdated suites.

I've just merged support for the newer/better mbedTLS library. If you try out the http_get_mbedtls example then you should get better results, hopefully. If it doesn't work with mbedTLS then please open a new ticket.

Angus

closed 2015-09-28 by projectgus