SuperHouse/esp-open-rtos #41: Usage of strftime(..) leads to link error ...

Angus71, 2015-08-30:
Hi,

While converting time into a user understandable form the following link error occurs. I also attached a minimal c file, which in my case reproduces the problem. (Checked with latest master branch SuperHouse@2fe1b7a00d8d1b33c61cee2437622fc70ba479a2)

```
../../libc/xtensa-lx106-elf/lib/libc.a(lib_a-sysopen.o):(.literal+0x0): undefined reference to `_open_r'
../../libc/xtensa-lx106-elf/lib/libc.a(lib_a-sysopen.o): In function `open':
/home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/syscalls/../../../../../newlib/libc/syscalls/sysopen.c:21: undefined reference to `_open_r'
collect2: error: ld returned 1 exit status
```
Is there something I've missed in the Makefile?

Makefile
```
PROGRAM=simple
include ../../common.mk
```
Sample Code
```
/* Does not link ...
 */
#include "stdio.h"
#include "time.h"

void user_init(void)
{
   time_t rawtime;
   struct tm *info;
   char buffer[80];

   time( &rawtime );

   info = localtime( &rawtime );

   strftime(buffer,80,"%x - %I:%M%p", info);
   printf("Formatted date & time : |%s|\n", buffer );

   while(1) {};
}
```

foogod, 2015-08-31:
This is a bit odd.. The problem appears to be that the use of `strftime` pulls in something which ultimately tries to `open` a file on the filesystem, and since we have no filesystem, we have no underlying implementation for that function..  But I don't know off hand what `strftime` would be calling that would have that sort of dependency..

Note that since the ESP8266 does not actually have a real-time clock, `time` et al are very unlikely to provide valid wall-clock time information anyway (since they're reset to 0 every time it boots).  To get around this you'd probably need to use NTP or something to synchronize the clock over the network before doing this sort of stuff.  (AFAIK, the libc routines also currently have no support for custom time zones either, so `localtime` will (I believe) always return GMT..)

We should still try to figure out why `strftime` thinks it needs `open`, though, and fix that if possible..

projectgus, 2015-09-01:
I think what's happening is strftime calls `__get_current_time_locale()` to get the locale for formatting date/time fields as strings ("January", etc.), which then calls through to `__part_load_locale()` in `libc/locale/ldpart.c` which then either returns the locale from the cache, or tries to load it from the filesystem if it's not cached.

There doesn't seem to be a way to compile newlib with hardcoded or disabled locale support, which seems a bit odd to me. A quick search didn't show up anyone discussing it as a problem, either. Maybe people on microcontrollers just usually avoid strftime?

It could be something you could bring up on the newlib list, I'm guessing they might be amenable to accepting a patch that disabled loadable locale support...?

foogod, 2015-09-01:
It occurs to me that we could mitigate some of this sort of thing by just implementing sub file IO routines which always return error responses (for example, ENOENT for read and EROFS for write)..

projectgus, 2015-09-01:
Yeah, that might be the best option. I mean, it would be better not to pull in those routines at all and unnecessarily bloat the firmware with code that can never work. But as a stop-gap it's pretty good, and routines like strftime() are always going to come with a fairly big code size overhead.

projectgus, 2015-09-01:
(@Angus71, the place to implement those any stub syscall _open_r/etc syscall routines will be in `core/newlib_syscalls.c`)

projectgus, 2015-09-01:
In fact now I look there are already fstat() and close() stubs in there, just no stub for open(). The _read_r() and _write_r() syscalls currently assume you mean stdout, which is probably something to fix just in case someone doesn't check the result for open().

projectgus, 2015-09-03:
I think that should be fixed now. I wrote a quick test program and was able to parse a date/time with `strptime()` then print it with `strftime()`, and everything worked even simple C locale features like month abbreviation.

(It did bloat the code size out by 8kB, but if you need that functionality then you need it, I guess!)

commit 920ee8bf (2015-09-03): libc syscalls: ENOSYS implementations for open,fstat,close,lseek

Closes #41
closed 2015-09-03 by projectgus