SuperHouse/esp-open-rtos #42: Make int32_t/uint32_t non-long

foogod, 2015-09-02:
This is something that arguably isn't major, but keeps bugging me, and I'm sure is going to bug a bunch of people out there who might try to use esp-open-rtos, so it'd be good to fix..

stdint.h currently typedefs `int32_t` as `long int` and `uint32_t` as `unsigned long int`, despite the fact that int/unsigned on this architecture are 32-bits, so these could/should just be `int` and `unsigned` instead.  From an ABI standpoint, there's no difference, but a few things do care (most notably, the printf-format checking that GCC does (`-Wformat`) complains bitterly and stops the build if you try to print an `int32_t` value using a `"%d"`, for example, which is just dumb).

The esp-open-sdk stuff does not have this problem by default (things built with esp-open-sdk outside of esp-open-rtos define `int32_t` as `int`), so it's an issue specific to our particular configuration, not a general GCC thing.  From what I can tell so far, the problem seems to be that we're using the `stdint.h` from newlib, whereas esp-open-sdk normally uses the one from gcc.

I'm not sure the best way to fix this.  The most straightforward would probably be to include our own copy of `stdint.h` somewhere in the search path before the newlib stuff (which it occurs to me might be easier to do cleanly if we had a standard "include" directory for everything, as suggested in issue #28, but that's not necessarily a prerequisite)..

kanflo, 2015-09-02:
Not having looked too deep into the new lib sources it surprises me this is not configurable. Printfs with %d has been bothering me too when porting code to esp-open-rtos so I second it being fixed. Depending on include order has a "hackish" kind of feel to it, as you probably feel to. Unless this can be fixed upstream I guess we just have to deal with it in whatever way possible. And we should.

jcmvbkbc, 2015-09-02:
> From what I can tell so far, the problem seems to be that we're using the stdint.h from newlib, whereas esp-open-sdk normally uses the one from gcc.

AFAIK there's no stdint.h with type definitions in the gcc, only in the libc. And newlib-2.0.0 port for xtensa has this particular fix in it, here: https://github.com/jcmvbkbc/crosstool-NG/blob/feb1fb829d5cb4f8739e488cbd2f0b72304705bb/local-patches/newlib/2.0.0/0001-xtensa-add-port.patch#L841

foogod, 2015-09-02:
Ah, ok, I guess instead of "gcc" I should have said "the one from crosstool-NG"..  in that case it looks like it's because esp-open-rtos is using a different source for newlib than the version in crosstool-NG (and thus doesn't have that patch)..

@projectgus I believe the version of newlib used by esp-open-rtos is something you put together..  Any ideas on the right way to fix this?

projectgus, 2015-09-02:
Short answer: Yeah, this is my fault - sorry. We should probably fix it.

Long answer: Newlib 2.2.0 changed the way stdint.h is laid out and that part of the newlib-2.0 patch didn't apply against 2.2.0 (rest is [here](https://github.com/projectgus/newlib-xtensa/commit/54eb5243a28d2949807c75534c4603c8f9e72cb5). I dropped that part, then obviously didn't think twice while adding 'l' to lots of printfs...

Newlib 2.2.0 is currently getting the definition of int32_t/uint32_t [from gcc in _default_types.h here](https://github.com/SuperHouse/esp-open-rtos/blob/master/libc/xtensa-lx106-elf/include/machine/_default_types.h#L63), via the built-in `__INT32_TYPE__` macro:

According to this [10 year old patch](https://sourceware.org/ml/newlib/2005/msg00824.html), the reason newlib went this way (and it seems gcc does this) is due to "subtle issues in the PRI*32 macros" - also in newlib. This is probably no longer the case though? Maybe it's just to encourage people to use PRI*32 even when on 32-bit?

I think the "correct C" answer is to use the PRI*32 macros in any case, but given we're not writing portable C here that feels like total overkill (whereas using stdint sizes feels right to me.)

... anyway, I think the solution is just to patch projectgus/newlib-xtensa appropriately. I think what this new newlib structure is encouraging us to do is create a machine/xtensa/_default_types.h, although the only way to make this change without copying all of _default_types.h looks like it will be to redefine or undefine `__INT32_TYPE__`, which seems dangerous.

Anyone have thoughts/suggestions?

(A patch to gcc to change `__INT32_TYPE__` would also fix this particular problem, but I feel that's a bad idea. Plus it would mean everyone using esp-open-rtos would need a new toolchain, lest they drown in an avalanche of printf formatting errors!)

projectgus, 2015-09-02:
(Comment updated to fix markdown formatting problems.)

projectgus, 2015-09-02:
On my system arm-none-eabi-gcc defines `__INT32_TYPE__` as 'long int' as well, although x86 is 'int' for -m32 and -m64. Is the 'long int' thing a side effect of bootstrapping gcc with newlib?

foogod, 2015-09-03:
Interestingly, it looks like since that patch, newlib has changed to prefer whatever gcc thinks should be done (using `__INT32_TYPE__`, etc), but crosslib-NG apparently is setting `__INT32_TYPE__` to long specifically because it thinks newlib wants it that way (from gcc/config/newlib-stdint.h, which I assume is included when `CT_LIBC_newlib=y`):

```
/* newlib uses 32-bit long in certain cases for all non-SPU
   targets.  */
#ifndef STDINT_LONG32
#define STDINT_LONG32 (LONG_TYPE_SIZE == 32)
#endif
```

This smells a bit like it may actually be a case of chicken-and-egg for a problem which doesn't exist anymore, in which case the "correct" fix would arguably be to change newlib-stdint.h so that crosstool-NG produces a GCC which defines `__INT32_TYPE__` "normally" even when being built with `CT_LIBC_newlib=y`

But it's not entirely clear whether that actually is the case, and as mentioned, that still wouldn't really fix the immediate problem for lots of people, so we probably also want to change the stdint.h define that gets included with esp-open-rtos somehow anyway.  I'm not familiar with how newlib handles platform-specific defines, but I think I'd probably favor creating a `machine/xtensa/_default_types.h`, if it's actually that easy (I don't see a big problem with just copying the whole file and making our changes.. I doubt that it's likely to change that frequently)

projectgus, 2015-09-05:
How does that look? (Second commit 7ed064a not the first one, first one was a mistake that github hasn't cleaned up yet...)

projectgus, 2015-09-07:
If noone has any objections I'll merge this tomorrow.

projectgus, 2015-09-09:
Merged after a couple more minor fixups to fix least_int32_t and also PRI_32. Thanks @Frans-Willem for the least_int32_t fix.

commit 7ed064a7 (2015-09-05): libc: Redefine int32_t/uint32_t as 'int' not 'long'

libc built from projectgus/newlib-xtensa@007777696321407

Closes #42
commit 1d7bfeec (2015-09-09): Merge branch 'fix/int32_t'

Closes #42
closed 2015-09-09 by projectgus