SuperHouse/esp-open-rtos #43: Need a discussion forum for esp-open-rtos

foogod, 2015-09-02:
As discussed a bit on issue #25, GitHub issues are not necessarily the best way to discuss some issues, and a more general-purpose maillist / forum / etc would be very useful.

I believe @projectgus was looking into some options, but am not sure the status on that.  This ticket is just to track the progress on getting something set up (and will hopefully be closed soon :) )

projectgus, 2015-09-07:
Decided to bite the bullet and just go with a mailing list:
https://groups.google.com/forum/#!forum/esp-open-rtos

Links are added on the main README as well.

projectgus, 2015-09-07:
(Ping @mikejac & @kanflo who were involvede in the earlier discussion, in case they don't see this otherwise.)

closed 2015-09-07 by projectgus

seemasinghh, 2019-06-14:
I don't have an idea esp-open-RTOS whereas I think to get the training of [RTOS Training in Noida](

https://www.cetpainfotech.com/technology/Rtos-training) so I want to know about RTOS.