SuperHouse/esp-open-rtos #44: Errors in the build of all examples

CHERTS, 2015-09-03:
I'm using Ubuntu 14.04.3 LTS
When build any example there is the same error log assembly cite below.
Examples of esp-open-sdk going without problems.

sleuthhound@esp-server:~/esp8288/esp-open-rtos$ make flash -j1 -C examples/blink ESPPORT=/dev/ttyUSB0
make: Вход в каталог `/home/neo/esp8288/esp-open-rtos/examples/blink'
CC /home/neo/esp8288/esp-open-rtos/examples/blink/blink.c
AR build/program.a
CC /home/neo/esp8288/esp-open-rtos/FreeRTOS/Source/croutine.c
CC /home/neo/esp8288/esp-open-rtos/FreeRTOS/Source/list.c
...
CC /home/neo/esp8288/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c
AR build/freertos.a
CC /home/neo/esp8288/esp-open-rtos/lwip/esp_interface.c
CC /home/neo/esp8288/esp-open-rtos/lwip/sys_arch.c
CC /home/neo/esp8288/esp-open-rtos/lwip/lwip/src/api/api_lib.c
....
CC /home/neo/esp8288/esp-open-rtos/lwip/lwip/src/netif/ethernetif.c
CC /home/neo/esp8288/esp-open-rtos/lwip/lwip/src/netif/slipif.c
AR build/lwip.a
CC /home/neo/esp8288/esp-open-rtos/axtls/axtls/crypto/aes.c
...
CC /home/neo/esp8288/esp-open-rtos/axtls/axtls_esp_stubs.c
AR build/axtls.a
CC /home/neo/esp8288/esp-open-rtos/core/esp_gpio_interrupts.c
CC /home/neo/esp8288/esp-open-rtos/core/esp_iomux.c
CC /home/neo/esp8288/esp-open-rtos/core/esp_timer.c
CC /home/neo/esp8288/esp-open-rtos/core/newlib_syscalls.c
CC /home/neo/esp8288/esp-open-rtos/core/sdk_compat.c
AS /home/neo/esp8288/esp-open-rtos/core/exception_vectors.S
AR build/core.a
SDK processing stage 1: Removing unwanted objects from ../../lib/libmain.a
cat ../../lib/symbols_norename.txt | grep -v "^#" | sed ':begin;N;s/\n/\\|/;tbegin' > build/sdklib/norename.match
SDK processing stage 2: Building symbol list for build/sdklib/libmain_stage1.a -> build/sdklib/libmain.rename
SDK processing stage 1: Removing unwanted objects from ../../lib/libnet80211.a
SDK processing stage 2: Building symbol list for build/sdklib/libnet80211_stage1.a -> build/sdklib/libnet80211.rename
SDK processing stage 1: Removing unwanted objects from ../../lib/libphy.a
SDK processing stage 2: Building symbol list for build/sdklib/libphy_stage1.a -> build/sdklib/libphy.rename
SDK processing stage 1: Removing unwanted objects from ../../lib/libpp.a
SDK processing stage 2: Building symbol list for build/sdklib/libpp_stage1.a -> build/sdklib/libpp.rename
SDK processing stage 1: Removing unwanted objects from ../../lib/libwpa.a
SDK processing stage 2: Building symbol list for build/sdklib/libwpa_stage1.a -> build/sdklib/libwpa.rename
cat build/sdklib/libmain.rename build/sdklib/libnet80211.rename build/sdklib/libphy.rename build/sdklib/libpp.rename build/sdklib/libwpa.rename > build/sdklib/allsymbols.rename
SDK processing stage 3: Renaming symbols in SDK library build/sdklib/libmain_stage1.a -> build/sdklib/libmain.a
SDK processing stage 3: Renaming symbols in SDK library build/sdklib/libnet80211_stage1.a -> build/sdklib/libnet80211.a
SDK processing stage 3: Renaming symbols in SDK library build/sdklib/libphy_stage1.a -> build/sdklib/libphy.a
SDK processing stage 3: Renaming symbols in SDK library build/sdklib/libpp_stage1.a -> build/sdklib/libpp.a
SDK processing stage 3: Renaming symbols in SDK library build/sdklib/libwpa_stage1.a -> build/sdklib/libwpa.a
LD build/blink.out
/home/neo/esp8288/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(mem_manager.o): In function `xPortGetFreeHeapSize':
(.irom0.text+0x4): multiple definition of `xPortGetFreeHeapSize'
./build/freertos.a(port.o):/home/neo/esp8288/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:215: first defined here
./build/ld-16/eagle.app.v6.ld:10: undefined symbol `malloc' referenced in expression
collect2: error: ld returned 1 exit status
make: *** [build/blink.out] Ошибка 1
make: Выход из каталога `/home/neo/esp8288/esp-open-rtos/examples/blink'

projectgus, 2015-09-03:
Hi CHERTS,

Did you build esp-open-sdk without adding STANDALONE=n? In the default esp-open-sdk mode it links the SDK libraries permanently into the toolchain, so you can't use it with esp-open-rtos.

If you run `make STANDALONE=n` in the esp-open-sdk directory then it should fix the problem.

If that isn't the problem, can you pastebin the 'build/blink.map' file (or any other example .map file) please?

Angus

CHERTS, 2015-09-04:
I used esp-open-sdk build make STANDALONE=y
Today, I try to make STANDALONE=n
Thank you

projectgus, 2015-09-18:
I'm going to close this issue as I'm fairly sure it was caused by STANDALONE=y.

Please reopen if you don't think that was the problem after all.

closed 2015-09-18 by projectgus