SuperHouse/esp-open-rtos #45: blink example does not work

atommann, 2015-09-09:
Hi,

In the blink example.

void blinkenTask(void *pvParameters) does not work.
void blinkenRegisterTask(void *pvParameters) works with GPIO4 an GPIO5 as parameters.

I checked eagle_soc.h found that some GPIOs are controlled in other way.
PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDI_U, FUNC_GPIO12); // Set GPIO12 function
PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTCK_U, FUNC_GPIO13); // Set GPIO13 function
PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTMS_U, FUNC_GPIO14); // set GPIO14 function

Thanks.

foogod, 2015-09-09:
That's odd.. I just tested the blink example and both methods seem to work fine for me using both GPIO 4 and 5.

When you say "doesn't work", you mean it compiles and flashes normally, but you just don't see any change on the GPIO pin when it runs?  Does it print anything to the UART?

(Are you sure you're looking at the right pins?  Note that on some ESP-12 modules (I'm not sure about other ESP modules) the GPIO4 and GPIO5 pins actually have their labels swapped (so the pin for GPIO4 says it's "GPIO5" and vice-versa)..)

Those macros in eagle_soc.h are not used in esp-open-rtos.  We're actually aiming to have a more intuitive and easy-to-use way of doing a lot of this stuff, so we're designing a somewhat different interface (as demonstrated in the blink example, etc)..  The way that blink.c is written should work correctly for any GPIO 0-15..

atommann, 2015-09-10:
Yes. It's odd. esp-open-rtos/examples/simple works fine.

It compiles and flashes normally. Nothing on the GPIOs. Then I add printf("hello world!\n"); in blinkenTask(), computer can receive these strings.

Yes. I am using ESP-12E module. And I checked both GPIO4 and GPIO5 with a multimeter, no voltage changing.

Then I tried the blinky example from repo https://github.com/esp8266/source-code-examples, I walked thru every GPIOs, the blinky code can blink every GPIOs. This means the hardware itself is still alive.

atommann, 2015-09-10:
I attached the code below.

GPIO14 doesn't work.
GPIO4 and GPIO5 works.

void blinkenRegisterTask(void *pvParameters)
{
    GPIO.ENABLE_OUT_SET = BIT(4);
    GPIO.ENABLE_OUT_SET = BIT(5);
    //IOMUX_GPIO14 = IOMUX_GPIO14_FUNC_GPIO | IOMUX_PIN_OUTPUT_ENABLE; /* change this line if you change 'gpio' */
    IOMUX_GPIO4 = IOMUX_GPIO4_FUNC_GPIO | IOMUX_PIN_OUTPUT_ENABLE; /* change this line if you change 'gpio' */
    IOMUX_GPIO5 = IOMUX_GPIO5_FUNC_GPIO | IOMUX_PIN_OUTPUT_ENABLE; /* change this line if you change 'gpio' */
    while(1) {
        GPIO.OUT_SET = BIT(4);
        GPIO.OUT_SET = BIT(5);
        vTaskDelay(1000 / portTICK_RATE_MS);
        GPIO.OUT_CLEAR = BIT(4);
        GPIO.OUT_CLEAR = BIT(5);
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
}

void user_init(void)
{
    sdk_uart_div_modify(0, UART_CLK_FREQ / 115200);
    //xTaskCreate(blinkenTask, (signed char *)"blinkenTask", 256, NULL, 2, NULL);
    xTaskCreate(blinkenRegisterTask, (signed char *)"blinkenRegisterTask", 256, NULL, 2, NULL);
}

projectgus, 2015-09-10:
Thanks for posting the code. I took the liberty of editing your post to add Markdown 'code' tags around the code.

Are you also changing the BIT(4)s to BIT(14) in the ENABLE_OUT_SET, OUT_SET, OUT_CLEAR sections?

Can you post the exact code which doesn't work when you try to use GPIO 14? Does that mean GPIO4 and GPIO5 work now as well?

Sorry for the confusion, just trying to understand which combinations are working and which aren't.

atommann, 2015-09-10:
Thanks for adding the markdown tags.

I just downloaded the blink.c from github and swapped the tasks in user_init(void) and there's no output on GPIO14.

```
/* The classic "blink" example
 *
 * This sample code is in the public domain.
 */
#include "espressif/esp_common.h"
#include "espressif/sdk_private.h"
#include "FreeRTOS.h"
#include "task.h"
#include "esp8266.h"

const int gpio = 14;

/* This task uses the high level GPIO API (esp_gpio.h) to blink an LED.
 *
 * Even though it reads better than the register-level version in blinkenRegisterTask,
 * they compile to the exact same instructions (except gpio_enable also set the output type in
 * the GPIO control register).
 */
void blinkenTask(void *pvParameters)
{
    gpio_enable(gpio, GPIO_OUTPUT);
    while(1) {
        gpio_write(gpio, 1);
        vTaskDelay(1000 / portTICK_RATE_MS);
        gpio_write(gpio, 0);
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
}

/* This task uses all raw register operations to set the pins.

   It's not fully parameterised, as the IOMUX_GPIO# macros involve a non-linear
   mapping from GPIO to IOMUX ports.

   There is no significant performance benefit to this way over the
   blinkenTask version, so it's probably better to use the blinkenTask
   version.

   NOTE: This task isn't enabled by default, see the commented out line in user_init.
*/
void blinkenRegisterTask(void *pvParameters)
{
    GPIO.ENABLE_OUT_SET = BIT(gpio);
    IOMUX_GPIO14 = IOMUX_GPIO14_FUNC_GPIO | IOMUX_PIN_OUTPUT_ENABLE; /* change this line if you change 'gpio' */
    while(1) {
        GPIO.OUT_SET = BIT(gpio);
        vTaskDelay(1000 / portTICK_RATE_MS);
        GPIO.OUT_CLEAR = BIT(gpio);
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
}

void user_init(void)
{
    sdk_uart_div_modify(0, UART_CLK_FREQ / 115200);
    //xTaskCreate(blinkenTask, (signed char *)"blinkenTask", 256, NULL, 2, NULL);
    xTaskCreate(blinkenRegisterTask, (signed char *)"blinkenRegisterTask", 256, NULL, 2, NULL);
}
```

atommann, 2015-09-10:
And now I am trying the bmp180_i2c sample and the below code give me:

SDK version : 0.9.9
GIT version : 421f8dd

Is this normally? (I mean the SDK verson)

```
    // Just some infomations
    printf("\n");
    printf("SDK version : %s\n", sdk_system_get_sdk_version());
    printf("GIT version : %s\n", GITSHORTREV);
```

projectgus, 2015-09-12:
Hi @atommann,

Thanks for reporting these problems. I did some tests and was able to reproduce the issue, in fact I found two related bugs (commits above).

Please let us know if the GPIOs work as expected now.

Seeing these bugs makes me very keen to find a way to do some automated testing on every commit!

>SDK version : 0.9.9
> GIT version : 421f8dd
> Is this normal? (I mean the SDK verson)

Yes, esp-open-rtos is based on SDK 0.9.9 which was the last "MIT licensed" IoT RTOS SDK from Espressif, before they swtiched licenses.

Angus

atommann, 2015-09-13:
Hi,

Nice to see the commits.

I tested the latest repo with ESP-12-Q module from AI-Thinker, and here are the results:

Method of void blinkenTask(void *pvParameters)
GPIO5 works
GPIO4 works
GPIO0 works
GPIO2 works
GPIO15 works

GPIO16 (not tested)
GPIO14 works
GPIO12 works
GPIO13 works

method of void blinkenRegisterTask(void *pvParameters)
GPIO5 works
GPIO4 works
GPIO0 works
GPIO2 works
GPIO15 works

GPIO16 (not tested)
GPIO14 works
GPIO12 works
GPIO13 works

I found that GPIO2 is special. When the code blink GPIO2, the onboard blue LED also blinks, it means the blue LED is connected with GPIO0. But when I excute "make flash", this LED also flashes, this means this LED is also connected with a UART. I want to know the exact connection inside.

foogod, 2015-09-13:
I believe the bit about GPIO2 is because the Boot ROM configures that pin to be IOMUX_GPIO2_FUNC_UART0_TXD_BLINK by default, not a GPIO, so its default behavior is to blink when there's TX activity on UART0.  When you run the blink.c with GPIO2, you're switching it over to be an independent GPIO (IOMUX_GPIO2_FUNC_GPIO) instead (and thus from that point on, it presumably won't blink with UART activity anymore).

projectgus, 2015-09-18:
Thanks for going through all of the GPIOs. I'm hoping we'll have some kind of automated test for this stuff (and lots of other stuff) in the coming months.

I see the same behaviour that @foogod describes when testing that pin, also. Which is quite neat, I'd assumed the onboard LED was just connected directly to the TX pin so it's good to know it can be repurposed without losing UART functionality.

closed 2015-09-18 by projectgus