SuperHouse/esp-open-rtos #48: strtok_r errors

podema, 2015-09-18:
The chip freezes when strtok_r is in use. It doesn't even print previous statements.
if strtok_r line is commented then the program runs just fine.

Does someone has used succesfully strtok on the esp?
Any suggestions?
```c
void worker(void * pvParameters){
  char cR[BUF_SIZE];
  char *cMethod;
  //char *cURI;
  char *rest;
  int iBites;
  int sockClient;
  xQueueHandle xWorkersQueue=*(xQueueHandle *)pvParameters;
  while(1){
    //waiting for clients
    while( xQueueReceive(xWorkersQueue , (void *)&sockClient , portMAX_DELAY ) != pdTRUE );
    //get client request
    iBites=lwip_recv(sockClient, cR, sizeof(cR), 0);
    printf("-----REQUEST-------\n");
    printf("%.*s\n",iBites,cR);
    //parsing request
    //strcpy(cMethod,strtok_r(cR," ",&rest));
    //strcpy(cURI,strtok_r(NULL," ", &rest));
    cMethod=strtok_r(cR," ",&rest);
    printf("----------DEBUG----------\n%s\n%s\n",cMethod,"");
   
    
    strcpy(cR, "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\nprova");
    printf("-----RESPONSE-------\n");
    printf("%s\n",cR);
    lwip_send(sockClient, cR, strlen(cR), 0);
    lwip_close(sockClient);
```

foogod, 2015-09-18:
One question when looking over your code:  Are you sure that `cR` is zero-terminated?  It looks like you're just doing a raw recv and then passing that to `strtok`, but `strtok` expects a zero-terminated string (and if it doesn't find a zero, will just keep reading off the end into who knows what memory regions (and then write zeros into them))..

(Note that `lwip_recv` is also not guaranteed to return all of the sent data in one go, and you may have to call it multiple times to get the full text. (so, for example, it's quite possible that instead of returning "GET / HTTP/1.0", when you call it the first time it'll return "GE" and the second time it'll return "T / HTTP/1.0".  This may not have anything to do with your problem here, but it's something your code should be prepared for..)

podema, 2015-09-19:
So if i strncpy or write at the end a \0 before passing it to strtok it
should work?
Thanks for the tip, never took that into account, it was driving me crazy.
I will try.
How is it however possible that even previous statements didnt work. I mean
if a overwrited a sensible region of memory with strtok it makes sense that
i doesnt work. But first it have to arrive to strtok statement. I will like
to know.
Another time thanks for your help

foogod, 2015-09-20:
Note that when you send output over the UART using `printf` et al, what it actually does is stick the data into a hardware FIFO buffer on the UART, and the UART chip then works on actually sending that out over the wire while the CPU continues on with other things.

Even at 115200 bps, the rate the UART can actually send data out is glacially slow compared to how fast the CPU keeps on executing instructions, so it's quite possible that the previous statements did actually execute, but you never saw their output because it just never had time to make it out of the UART FIFO buffer before `strok`'s rogue memory accesses locked up the whole system (including the UART).

If you want to be absolutely sure that your output has left via the UART before doing the next step, you can put something like the following in your code to wait until the FIFO is actually empty before continuing (which should make sure you see all messages up to that point, even if something later goes completely bonkers):

    while (FIELD2VAL(UART_STATUS_TXFIFO_COUNT, UART(0).STATUS) != 0) {}

podema, 2015-09-21:
after modifying the program, manually zero terminating the string has not solved the problem:
```
  while(1){
    //waiting for clients
    while( xQueueReceive(xWorkersQueue , (void *)&sockClient , portMAX_DELAY ) != pdTRUE );
    //get client request
    /*
    counter=cR;
    acc=0;
    do{
      if(acc <= sizeof(cR)-sizeof(char) ){
        iBites=lwip_recv(sockClient, counter, sizeof(cR)-sizeof(char)-acc, 0);
        acc=acc+iBites;
        counter=&cR[acc/sizeof(char)];
      }
      else{
        iBites=lwip_recv(sockClient, &acc, sizeof(acc),0);
      }
    }
    while(iBites>0);
    */
    iBites=lwip_recv(sockClient, cR, sizeof(cR)-sizeof(char), 0);
    cR[BUF_SIZE]='\0';
    printf("-----REQUEST-------\n");
    printf("%.*s\n",iBites,cR);
    //parsing request
    cMethod=strtok_r(cR," ",&rest);
    cURI=strtok_r(NULL," ",&rest);
    printf("-----DETAILS-----\n");
    printf("Method: %s\nURI: %s",cMethod,cURI);
```

projectgus, 2015-09-22:
Hi podema,

A couple of thoughts:

On Mon, Sep 21, 2015 at 10:19:11AM -0700, podema wrote:
> ```
>     cR[BUF_SIZE]='\0';
> ```

you probably want to do `if(iBytes > 0) cR[iBytes] = 0;` here, because otherwise there could still be leftover junk in the buffer between after the data that you just read, but before the end.

```
acc <= sizeof(cR)-sizeof(char)
```

This code was commented so it's probably not the problem, but you should check sizeof(cR) isn't 4. Probably safest to use BUF_SIZE instead, if that's the size of the buffer pointed to by cR.

> ```
>     cMethod=strtok_r(cR," ",&rest);
>     cURI=strtok_r(NULL," ",&rest);
> ```

If you add a line below these two

```
printf("iBytes %d cR %p cMethod %p cURI %p\n", iBytes, cR, cMethod, cURI);
```

... then what does the full output from the code you pasted (including the other debug output) look like?

Angus

projectgus, 2015-10-06:
@podema, are you still having any problems with strok_r, or is it OK to close this issue?

podema, 2015-10-10:
Sorry for the delay, i had a busy week. 
I tried what you suggested, i tried even rewriting it from scratch, i'm still have the same error. However this easy program works correctly:
```c
void unused_function(){
  char foo[30];
  char *rest;
  while(1){
    strcpy(foo,"FOO BAR");
    printf("%s\n",strtok_r(foo," ",&rest));
    printf("%s\n",strtok_r(NULL, " ",&rest));
  }
}
void example(void *pvParameters){
  while(1){
    printf("Hello world!");
  }
}

void user_init(void)
{
  sdk_uart_div_modify(0, UART_CLK_FREQ / 115200);
  xTaskCreate(unused_function, (signed char *)"unused_function", 256, NULL, 1, NULL);
}
```

I suppose i'm violating some memory sector, and as you said corrupting the UART buffer. 
So it's okay to close the issue since it's not!
Thanks for your attention

projectgus, 2015-10-12:
OK, I'm going to close this for now.

If you're still stuck, you might find it useful to post your code (or a link to it) on the mailing list:
https://groups.google.com/forum/#!forum/esp-open-rtos

closed 2015-10-12 by projectgus