SuperHouse/esp-open-rtos #51: Configure as AP -wifi_softap_get_station_num

yogk, 2015-10-03:
I am trying to write a small application to configure esp8266 as SoftAP using esp-open-rtos in the example folder.

As below:
extern struct station_info* sdk_wifi_softap_get_station_info();
extern bool sdk_wifi_softap_free_station_info();
extern  uint8_t sdk_wifi_softap_get_station_num();
/**************************************************

task()
{              while(true)
          {
               uint8_t stationCount = sdk_wifi_softap_get_station_num();
        	if(stationCount)
        	{
        		printf("No of stations=%d\n",stationCount);
            	        sleep(1000);

            	       struct station_info *stationInfo = sdk_wifi_softap_get_station_info();
                	if (stationInfo != NULL)
            	   {
            		while (stationInfo != NULL)
            		{
            			printf("Station IP: %d.%d.%d.%d\n", IP2STR(&(stationInfo->ip)));
            			stationInfo = STAILQ_NEXT(stationInfo, next);
            		}
            		sdk_wifi_softap_free_station_info();
            	}
 } 
}
I defined station_info  structure in one of the .h files as seen in standard iot-sdk.

But i get linker and compiler error as "undefined reference" to these below functions:

extern struct station_info* sdk_wifi_softap_get_station_info();
extern bool sdk_wifi_softap_free_station_info();
extern  uint8_t sdk_wifi_softap_get_station_num();

Are these not part of libmain.a as are other wifi functions?

How can I configure esp8266 in AP mode and get the list of stations connected using esp-open-rtos?
Are there some other API's to configure as AP?
/**************************************************

foogod, 2015-10-03:
If you could provide a full program instead of a snippet it would make things easier (I wasn't able to easily figure out how to make your snippet compile, because it's obviously missing a fair bit of stuff, so I didn't actually test it myself)..

`sdk_wifi_softap_get_station_info` and `sdk_wifi_softap_free_station_info` are both present in the libraries and should not give you link errors (I just tried adding them to a sample program of my own and it compiled/linked without complaint).

However, `wifi_softap_get_station_num` is not present in the (0.9.9) Espressif RTOS libraries which esp-open-rtos is based on.  It may have been added after those were released, or it may have just been mysteriously left out by Espressif for no apparent reason (it's happened with other things), I'm not sure..

Unfortunately we haven't gotten to the point of pulling apart the wifi APIs enough to write any of our own..  From the description in the docs, it sounds like this function probably just walks through the linked list returned by `sdk_wifi_softap_get_station_info` and tallies up a count of entries, so it might not be too hard to just write your own version instead..  (if you want to write a `sdk_wifi_softap_get_station_num` function that matches Espressif's semantics, we'd be happy to incorporate it back into esp-open-rtos)

yogk, 2015-10-04:
sorry for the not so correct complete code given. below details:
created a folder called wifitest in examples with the file wifiap.cpp and a makefile (like other example with C++). I want it in C++ and not in C.
The complete code is as follows.

#include "task.hpp"
#include "queue.hpp"
#include "string.h"
#include "espressif/esp_common.h"
#include "sdk_internal.h"

extern struct station_info* sdk_wifi_softap_get_station_info();
extern bool sdk_wifi_softap_free_station_info();
extern  uint8_t sdk_wifi_softap_get_station_num();
/******************************************************************************************************************
 * task_1_t
 *
 */
class task_1_t: public esp_open_rtos::thread::task_t
{
public:
    esp_open_rtos::thread::queue_t<uint32_t> queue;
    
private:
    void task()
    {
        printf("Station Connected ::task(): start\n");
        while(true)
        {
        	uint8_t stationCount = sdk_wifi_softap_get_station_num();
        	if(stationCount)
        	{
       		      printf("No of stations=%d\n",stationCount);
            	      sleep(1000);
                      struct station_info *stationInfo = sdk_wifi_softap_get_station_info();
            	      if (stationInfo != NULL)
            	       {
            		   while (stationInfo != NULL)
            		   {
            			printf("Station IP: %d.%d.%d.%d\n", IP2STR(&(stationInfo->ip)));
            			stationInfo = STAILQ_NEXT(stationInfo, next);
            		   }
            		          sdk_wifi_softap_free_station_info();
            	         }
        	      }
               }
    }
}; //end of class

task_1_t task_1;
extern "C" void user_init(void)
{
    sdk_uart_div_modify(0, UART_CLK_FREQ / 115200);
    struct sdk_softap_config apconfig;

    memcpy(apconfig.ssid,"ESP8266YK",10);
    memcpy(apconfig.password,"123",3);
    apconfig.ssid_hidden=0;
    apconfig.max_connection=4;
    apconfig.authmode=AUTH_OPEN;
    sdk_wifi_softap_set_config(&apconfig);
   
    task_1.task_create("tsk1");
   
}
Added station_info struct in esp_softap.h 
struct station_info {
	STAILQ_ENTRY(station_info)	next;

	uint8_t bssid[6];
	struct ip_addr ip;
};

When i compiled and link I got undefined reference to sdk_wifi_softap_get_station_info and sdk_wifi_softap_free_station_info.

As "sdk_wifi_softap_get_station_num" is not in the 0.9.9 espressif sdk libraries itself,
it will take sometime for me to implement sdk_wifi_softap_get_station_num() and submit to esp-open-rtos as i am new to esp8266 and esp-open-rtos.

Thanks,
YK

yogk, 2015-10-04:
One further thing I investigated is that when I use a C file instead of C++ the function ,sdk_wifi_softap_free_station_info() and sdk_wifi_softap_get_station_info() got resolved and there is no linker error. I resolved it by adding extern "C" when i declared the prototypes.
extern "C" struct station_info* sdk_wifi_softap_get_station_info();
extern "C" bool sdk_wifi_softap_free_station_info();

-YK

projectgus, 2015-10-04:
Hi YK,

Does resolving the linker error fix the problem?

It looks like the functions and the structure are there, they just weren't in Espressif's header files. The data types look very simple so I think we can add both the structure definition and the functions to the SDK headers without any licensing issues.

I'm not sure if the ip_addr fields will automatically fill in, or if the DHCP server is expected to fill them in itself.

The original DHCP server in the 0.9.9 RTOS SDK had to be removed due to licensing issues. I have a DHCP server implementation in the branch [features/dhcp-server](https://github.com/SuperHouse/esp-open-rtos/tree/feature/dhcp-server) but it could use some more testing (hint hint). :) There's an 'access_point' example which is written in C, but shows an example of running as an AP with DHCP.

In the DHCP server branch the dhcp server keeps track of the connected clients internally in its own list, but I think we should actually probably eventually just reuse the station_info data structure from the SDK. The only problem is the lack of a lease expiry time in the station_info structure itself. After we have the ability to manipulate the SDK assembly code we can probably extend the structure size, though... (this paragraph is all just me thinking out loud hypothetically, not really important for now!)

In any case, please let us know if the AP code now works for you (either in master or in the dhcp-server branch.)

Angus

yogk, 2015-10-11:
Hello ProjectGus,
I tested the AP code first on the master branch without using sdk_wifi_softap_get_station_num(); as its not a part of sdk.  The code I used is the same as mentioned above.
Here I could see from my phone the AP name "ESP8266YK" when I scan. But was not able to connect and get an IP address for my phone. 

So my next step was to include the dhcp server in extras folder and give a try. Included dhcpserver from your dhcpserver branch and then added the below code.
The behaviour is the same. It does not connect and so no IP address.

class task_1_t: public esp_open_rtos::thread::task_t
{
    
 private:
   void task()
    {
        printf("Station Connected ::task(): start\n");

        while(true)
        {

        
            	struct station_info *stationInfo = sdk_wifi_softap_get_station_info();
            	if (stationInfo != NULL)
            	{
            		while (stationInfo != NULL)
            		{
            			printf("Station IP: %d.%d.%d.%d\n", IP2STR(&(stationInfo->ip)));
            			stationInfo = STAILQ_NEXT(stationInfo, next);
            		}
            		sdk_wifi_softap_free_station_info();
            	}
        	
        }
    }
};

extern "C" void user_init(void)
{
   //Other AP initialisation code here (same as above).

  /* DHCP server code below */
       struct ip_info ap_ip;
       IP4_ADDR(&ap_ip.ip, 172, 16, 0, 1);
       IP4_ADDR(&ap_ip.gw, 0, 0, 0, 0);
       IP4_ADDR(&ap_ip.netmask, 255, 255, 0, 0);
       sdk_wifi_set_ip_info(1, &ap_ip);

       ip_addr_t first_client_ip;
      IP4_ADDR(&first_client_ip, 172, 16, 0, 2);
      dhcpserver_start(&first_client_ip, 4);

}

Hope these steps are correct with dhcp server.?
OR do I have to follow the same steps as in the example/access_point code when dhcpserver is added?

--YK

projectgus, 2015-10-12:
This looks right at a glance. I have a few more questions to help pin this down:

* If you run the examples/access_point example, does that work?
* Can you paste the serial output  from the access_point example for comparison, please?
* What phone are you using? Can you try any other DHCP clients, like your computer for example?

Something to keep in mind is that most phones (iOS and Android, at least) will usually disconnect from the WiFi after a few seconds if they can't find a route to the internet. See this mailing list thread:
https://groups.google.com/forum/#!searchin/esp-open-rtos/dhcpserver/esp-open-rtos/4BlYYrYRfH8/NCcENdF2AQAJ

yogk, 2015-10-17:
HI ProjectGus,

Please find the answers for the questions
> If you run the examples/access_point example, does that work?
 Yes it works . Was able to connect using my mobile and laptop to esp8266 as AP. Sometimes it crashed when I tested it multiple times (i have also included the logs when crashed)

>Can you paste the serial output from the access_point example for comparison, please?
below is the serial output for Access point example

Case 1: When i connect my mobile (I used xiomi) below are the logs
add 1
aid 1
station: 14:f6:5a:ec:91:c4 join, AID = 1
State dump. Message type 3
lease slot 0 expiry 360000 hwaddr 14:f6:5a:ec:91:c4
lease slot 1 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
DHCP lease addr 172.16.0.2 assigned to MAC 14:f6:5a:ec:91:c4
max connection!
station: 14:f6:5a:ec:91:c4 join, AID = 1
State dump. Message type 3
lease slot 0 expiry 360000 hwaddr 14:f6:5a:ec:91:c4
lease slot 1 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
DHCP lease addr 172.16.0.2 assigned to MAC 14:f6:5a:ec:91:c4

========================================================================
case 2: When I connect both my mobile xiomi and laptop below are the logs
add 1
aid 1
station: 14:f6:5a:ec:91:c4 join, AID = 1
State dump. Message type 3
lease slot 0 expiry 360000 hwaddr 14:f6:5a:ec:91:c4
lease slot 1 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
DHCP lease addr 172.16.0.2 assigned to MAC 14:f6:5a:ec:91:c4
max connection!
station: 14:f6:5a:ec:91:c4 join, AID = 1
State dump. Message type 3
lease slot 0 expiry 360000 hwaddr 14:f6:5a:ec:91:c4
lease slot 1 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
DHCP lease addr 172.16.0.2 assigned to MAC 14:f6:5a:ec:91:c4
max connection!
max connection!
max connection!
add 2
aid 2
station: ac:b5:7d:0c:b9:d8 join, AID = 2
State dump. Message type 3
lease slot 0 expiry 360000 hwaddr 14:f6:5a:ec:91:c4
lease slot 1 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
DHCP Server Error: Lease for address already taken
State dump. Message type 1
lease slot 0 expiry 360000 hwaddr 14:f6:5a:ec:91:c4
lease slot 1 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
State dump. Message type 3
lease slot 0 expiry 360000 hwaddr 14:f6:5a:ec:91:c4
lease slot 1 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
DHCP lease addr 172.16.0.3 assigned to MAC ac:b5:7d:0c:b9:d8
[... 124 lines cut ...]
IP4_ADDR(&ap_ip.gw, 0, 0, 0, 0);
IP4_ADDR(&ap_ip.netmask, 255, 255, 0, 0);
sdk_wifi_set_ip_info(1, &ap_ip);
memcpy(apconfig.ssid,"ydkesp",strlen("ydkesp"));
memcpy(apconfig.password,"1234",strlen("1234"));
apconfig.ssid_len=strlen("1234");
apconfig.ssid_hidden=0;
apconfig.max_connection=3;
apconfig.authmode=AUTH_WPA_WPA2_PSK;//AUTH_WPA2;
sdk_wifi_softap_set_config(&apconfig);
ip_addr_t first_client_ip;
IP4_ADDR(&first_client_ip, 172, 16, 0, 2);
dhcpserver_start(&first_client_ip, 4);
task_1.task_create("tsk1");
}

I am facing this new problem.

1. After I program esp8266 with the access_point example followed by my program which has set AP name as "ESP8266YK", I am not able to see my access point name in my phone & laptop. I tried changing my AP name to some other name like "ydkesp" , but no success even though I use
sdk_wifi_softap_set_config(&apconfig);
However strangely my phone and laptop are still getting connected to "esp-open-rtos AP" which is the AP name used in access_point example even after I program esp8266 with new AP name

I tried attaching the code as pdf file, but it does not give me permission to attach document.

--YK

projectgus, 2015-10-21:
Hi YK,

Sorry for the slow reply, I've had a busy week. I've gone back and edited your comment to use "code format" <code>```</code> markdown style tags at the top of bottom of the logs & code, to make it more readable. If you're posting big chunks of code then you can also do it via http://gist.github.com and paste a link (easier than posting a PDF).

Regarding the crash, I tried and I wasn't able to reproduce by connecting the devices I have here. Can you please run this command against the access point binary file:

> xtensa-lx106-elf-addr2line -e build/access_point.out

... and then copy-paste the number from the epc1 line into the command:

> epc1=0x401064cc

So in this case paste/type in 0x401064cc and press enter.

... and addr2line will output the line number where it's crashing.

> tried changing my AP name to some other name like "ydkesp" , but no success even though I use sdk_wifi_softap_set_config(&apconfig); However strangely my phone and laptop are still getting connected [...]

I think this may be because the PSK "1234" is too short. WPA PSK minimum length is 8 characters, so the sdk_wifi_softap_set_config() call is probably failing and not making any changes. The device loads the last used wifi settings by default though (this is a behaviour of the Espressif SDK we haven't removed yet).

yogk, 2015-10-24:
Hi ProjectGus,

thanks for educating me on the "code format" ```as I was not aware of this in github.
Here is the crash and reboot that i could see again with my laptop and mobile connected
```
lease slot 0 expiry 360000 hwaddr ac:b5:7d:0c:b9:d8
lease slot 1 expiry 360000 hwaddr 04:18:0f:9f:42:51
lease slot 2 expiry 0 hwaddr 00:00:00:00:00:00
lease slot 3 expiry 0 hwaddr 00:00:00:00:00:00
DHCP lease addr 172.16.0.3 assigned to MAC 04:18:0f:9f:42:51
max connection!
Fatal exception (29): 
epc1=0x40101368
epc2=0x00000000
epc3=0x401001d6
excvaddr=0x000f8008
depc=0x00000000
excsave1=0x4024326f
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30428, room 16 
tail 12
chksum 0xc4
ho 0 tail 12 room 4
load 0x3ffe8000, len 2984, room 12 
tail 12
chksum 0x0d
ho 0 tail 12 room 4
load 0x3ffe8bb0, len 6136, room 12 
tail 12
```

Now running   (since epc1=0x40101368 during crash)
xtensa-lx106-elf-addr2line -p -f -e examples/access_point/build/access_point.out 0x40101368 
gave me below output
uxListRemove at /home/esp8266/esp-open-rtos/FreeRTOS/Source/list.c:184

=================================================================

Yes you are absolutely right. The password "1234" was too short and hence it was loading the last stored SSID name. After changing the password to > 8 characters I was able to get the new SSID "ydkesp".

But the task or thread is not connecting to any station with a call to 
 struct station_info *stationInfo = sdk_wifi_softap_get_station_info();
```
        while(true)
        {
                sleep(100);
                struct station_info *stationInfo = sdk_wifi_softap_get_station_info();
                    while (stationInfo != NULL)
                    {
                        printf("Station IP: %d.%d.%d.%d\n", IP2STR(&(stationInfo->ip)));
                        stationInfo = STAILQ_NEXT(stationInfo, next);
                    }
                    sdk_wifi_softap_free_station_info();
        }
```
Any idea why this does not connect to any stations  using the sdk api's? Do we always need to use netconn API's ?

-YK

yogk, 2015-10-24:
I think the sdk api  sdk_wifi_softap_get_station_info() will connect to the station only when we use the original dhcp server of sdk and not the one which you used for access_point example, but since we cannot use original dhcp server due to licensing issues, I think this sdk_wifi_softap_get_station_info api() will not work with the dhcp derver you integrated as part of(https://github.com/SuperHouse/esp-open-rtos/tree/feature/dhcp-server) branch. Hope i am right ?

projectgus, 2015-10-27:
@yogk, I did some experiments and I think you're correct that this API doesn't work in esp-open-rtos. I never get any output back from calling `sdk_wifi_softap_get_station_info()` in the SDK.

I've pushed some experimental code to a new branch:  
https://github.com/SuperHouse/esp-open-rtos/compare/feature/dhcp-server...experiments/get_station_info?expand=1

One odd thing, the linked list itself seems to be referenced via two pointers in the global SDK state structure - `sdk_g_ic.v.station_info_head` & `sdk_g_ic.v.station_info_tail`. After the first call to `sdk_wifi_softap_get_station_info()` (which returns NULL), the tail pointer updates itself to point to data RAM. However the contents of that RAM doesn't seem to be valid.

At the high level view, there are really two sets of data that should be made available via an API. They're similar but different:

* Enumerate all current DHCP leases (IP & BSSID address).
* Enumerate all connected stations (ie wifi clients) via BSSID.

The first we can do just from the simple DHCP server code. But there may be stations connected that haven't been assigned IPs via DHCP, or there may be DHCP leases where the station is not currently connected.

The second enumaration must be something the SDK libraries have internally already, we just have to work out how to access that information properly.

Once both these are available, the most useful possibility is probably a combination API that returns current DHCP leases and also a flag for whether each client is currently connected to the AP.

I think the next step will be to dig around in the GPLed udhcpd binaries that Espressif released with the rtos SDK, and work out how they were interfacing with the rest of SDK.

rameshKrSah, 2017-07-31:
I am also trying to find out the number of clients connected the ESP in softAP mode, but not being able to do so. While experimenting I found that the dhcp leases once assigned are not set to zero after the client disconnect from the ESP. Is this the normal behavior, say that to know which clients were connected the ESP later or is this a bug?

Also, no call to **handle_dhcp_release** function is made during the disconnect process?

I read this thread and am not able to grasp the idea of how to solve this issue. Have you guys worked out a solution or anything? If yes, can you explain it again?

Regards
Ramesh

ourairquality, 2017-07-31:
@rameshKrSah As noted `sdk_wifi_softap_get_station_info()` should return the list of clients connected, but it only includes those with an ip addr noted by calling `sdk_wifi_softap_set_station_info()`. The dhcp server in the ipv2 branch includes this and numerous other improvements. We might need to add a hook to inform the dhcp server when clients leave and can probably add that to cnx_node_leave if necessary. Keeping the old information around seems ok, just in case the client reconnects.

rameshKrSah, 2017-08-15:
@ourairquality Thank you for the response, but I am not able to find the **lwip v2 branch**, in the branch list for esp-open-rtos. Can you kindly, link that branch here.

Regards
Ramesh

ourairquality, 2017-08-15:
@rameshKrSah See https://github.com/SuperHouse/esp-open-rtos/pull/437 You can use the whole patch set, or cherry pick from that if you wish but there might be other dependent patches - the set gives the order at least.