SuperHouse/esp-open-rtos #52: Reading from UART is broken

projectgus, 2015-10-05:
As per reports on mailing list and in comments of #31.

The current UART reading code in master doesn't seem to work.

Fix will be to implement our own UART wrapper functions and test whatever initialisation stages are necessary to set the UART RX FIFO up correctly and read/buffer characters in hardware on input.

kanflo, 2015-10-05:
I have started work on using the HW FIFO in #31 and guess that can be reused here. Just in case we don't want to duplicate our efforts.

projectgus, 2015-10-05:
Hi kanflo,

That'd be great, thanks! Are there more changes you've got in progress right now?

Angus

foogod, 2015-10-06:
I haven't had a chance to look at any of the existing code, but I'm not sure what you mean about initialization of the UART FIFO.. there should be no initialization required except setting up the baud rate divisor and whether it uses RTS/CTS handshaking.  the UART automatically uses the hardware FIFO (and I don't think there's even a way to turn it off), so there should be no special configuration needed..

I've actually successfully used the UART with no special initialization even with bare code (no SDK, just booting straight from the ROM into my own entry point), so I know this to be true..

Are you sure you aren't overthinking this whole thing?  I suspect the problem may be much simpler (like just not reading the FIFO status right or something)

projectgus, 2015-10-06:
I could easily be overthinking it and doing something silly.

Just to be clear though, with your experience are you talking about UART RX or TX? I have no problems with UART TX (ie putc/printf), but working with UART RX seems trickier somehow...

foogod, 2015-10-06:
(non-interrupt-driven) RX and TX are almost exactly the same, really.. in the case of TX, you check the TX FIFO status to make sure it's not full (number returned is less than 127), block if necessary/appropriate until it's ok, then write the byte to the FIFO register.  RX, you check the RX FIFO status to make sure there's something in it (number returned is not 0), block if necessary/appropriate, then read a character from the FIFO register.

That's really all there should be to it..

I can take a look at the current state of things in more detail and try to figure out what's going on when I get home this evening, if you want..

projectgus, 2015-10-06:
This is also what I thought, but the status register wasn't playing ball for me when I tested it. I tried a few different CONF settings for different RX FIFO "full" sizes as well. I noticed some of the CONF fields (like data width & stop bits) were defaulting to wrong-looking values when read back, and the FRAME ERROR status bit was being set a lot. Which is the "initialisation" stuff I was obliquely referring to. I don't really understand why TX seems to work fine even despite those weird CONF values though.

If you have time to take a look that would be great. I'll push a work in progress of what I was doing to a branch, so you can probably spot what I was doing wrong. :)

Angus

yesco, 2015-10-06:
I'm happy with any cl that keeps this one working: a minimal emacs vt100
editor for esp

https://github.com/yesco/imacs

last days Hack while hanging out with my friend Mitch Altman in Hong Kong,
heading to china RSN.

foogod, 2015-10-06:
Ok, I think I see why getc/getchar aren't working right, and it looks like it has nothing to do with low-level UART stuff..  the implementation of _read_r isn't quite right.. it currently blocks until it can read `len` bytes of data, but it's supposed to block until it can read at least 1 (at *most* `len`) bytes and then return as much as it's got.

I actually realized I have a mostly-ready esp/uart.h with some basic wrapper routines sitting around I was meaning to contribute, so this seems like a good excuse to do that too, and then we won't need to rely on the ESDK sdk_uart_rx_one_char/etc routines for this stuff either..

I should be able to put up a fix in probably an hour or so..

projectgus, 2015-10-06:
That's not the problem I'm seeing, but it might be a problem all the same. My understanding was that _read_r should block, unless the fd has been set as non-blocking (assuming we want POSIX-like behaviour.)

Something I did realise which is probably bad is that with the very coarse locking mechanism I implemented for newlib, blocking in _read_r will probably happen with interrupts disabled because newlib locks the stdin file descriptor. Which is probably not at all what we want!

> I actually realized I have a mostly-ready esp/uart.h with some basic wrapper routines sitting around I was meaning to contribute, so this seems like a good excuse to do that too, and then we won't need to rely on the ESDK sdk_uart_rx_one_char/etc routines for this stuff either..

Cool! I also wrote an esp/uart.h implementation, but if you post yours first I'm happy to merge back later in the evening (especially if yours works for rx, as mine doesn't!)

projectgus, 2015-10-06:
@yesco - that is really cool!

foogod, 2015-10-06:
Posix behavior is that read() should block, until there is *any* data available.  It need not be enough data to fill the provided buffer (and in fact with stdio often isn't).

I got an inkling what might be going on when @yesco mentioned that `read(1)` worked correctly but `getc`/`getchar` blocked "forever"..   That made me suspect it was something in the stdio buffered-read part of things.  The problem is that the stdio routines call `_read_r` with a reasonably large buffer, and `_read_r` wasn't returning until it had completely filled the buffer (so in reality, I suspect if you pumped enough input into it over the serial port, eventually it would "wake up" and process it all in a huge batch)..

Anyway, I've just put up my modified version of things, which I've tested with a simple echo app and appears to read characters correctly via getchar()..

foogod, 2015-10-06:
You know, reading through the byzantine paths of some of this newlib stuff, it occurs to me it might be a good idea to point out somewhere in our documentation that while you *can* use the stdio stuff for input/output, if you're interested in performance it'd probably be better to use more low-level routines (like what's in esp/uart.h) instead..  There's really quite a bit of overhead with the stdio stuff, it looks like.

I'm also thinking we might want to look into whether there's an easy way to reduce the size of (or just eliminate) the buffer on stdin by default..  It looks like it's currently mallocing a 1KB buffer, which considering it will only ever be able to read 127 bytes at a time, seems like a complete waste of almost 900 bytes of memory..

projectgus, 2015-10-06:
Yeah, you're right about POSIX. My mistake of understanding, I thought serial ports were different somehow.

Good catch on the newlib buffer size as well. I just added a commit to PR #53 that reduces them to 128 bytes by default. You can still use setvbuf() to make them larger or smaller (and/or disable line buffering on stdout), but much better to save 1792 bytes of RAM by default.

I have a couple of other comments on the PR, but I'll make them there.

yesco, 2015-10-06:
I noticed I need to flush unless I'm printing newline. is there any good
reason at all to delay writing to serial port? I see only disadvantages.

on the other hand calling many putchar() is now quite fast. wouldn't like
that to change.

but I'd expect that any serial driver should initiate sending anything in
the buffer if hardware bigger is not full. just keep it streaming.

projectgus, 2015-10-06:
By default newlib line buffers its output. I think this might be useful in a multithreaded environment, because I think each thread/task gets its own buffer and therefore won't chop all over the other thread - although I haven't verified that. That's the only potential use I can think of for it, anyhow.

To disable line buffering, call `setvbuf(stdout, NULL)`. You'll also get the RAM which is used for buffering back by doing that.

Possibly we should call this as part of the startup code, to disable line buffering by default.

projectgus, 2015-10-06:
>I think this might be useful in a multithreaded environment, because I think each thread/task gets its own buffer and therefore won't mix its output with another thread - although I haven't verified that.

I did some basic experiments printing from competing threads and line buffering definitely helps a lot, although I think it would be reasonable to leave buffering off by default.

foogod, 2015-10-06:
I was internally debating whether stdout buffering is good by default or not..  One drawback is that if you do a `printf` and then do something like `uart_flush_txfifo` in an attempt to make sure everything's been sent before continuing on (i.e. into some potentially crashy code), you're not actually guaranteed everything is written unless you do a `fflush` in between, and a lot of people may miss that step..

kanflo, 2015-10-06:
@foogod, could `flush` make an implicit `uart_flush_txfifo`for stdout? I can see both pros and cons with buffering, but there must be an "intuitive" way of flushing the buffer.

kanflo, 2015-10-06:
> That'd be great, thanks! Are there more changes you've got in progress right now?

foogod raced past me :) I have a few things for `extras/`, HTTP file upload being one of them. Will make PRs for those once #31 is in.

foogod, 2015-10-07:
> @foogod, could flush make an implicit uart_flush_txfifo for stdout? 

Unfortunately, there's a couple of issues with that.. One is it'd require custom hacking inside newlib, which is a little undesirable, but more importantly (to my mind) is that (a) it makes assumptions about stdout that might not be true (at least once we add back in the ability to install a custom stdout handler), and (b) it actually breaks the normal semantics of `fflush` (which normally does not wait for output to be finished writing, it just sends it to the underlying subsystem to send it out "as soon as possible")..  Arguably, what you're describing would be more the semantics of `fsync`, but using `fsync` for a serial stdout is not a very common/intuitive idiom, methinks..

Maybe the best approach is to just define another helper function in uart.h (or somewhere) which does a combination of `fflush(stdout)` followed by `uart_flush_txfifo(0)` and just make a point of mentioning in the docs for `uart_flush_txfifo` that if you're using stdio routines with the UART you really should use that function instead of `uart_flush_txfifo(0)`..

Alternately, I think it could be argued that in an environment like this, using buffering on stdout should be an explicit choice of the app developer, and if they're actually doing something that would benefit from it, they should use `setvbuf` themselves to turn it on with an appropriately sized buffer for their application (and otherwise have it unbuffered by default).  This would save a bit of RAM in the large majority of cases where I suspect nobody will notice the difference, and makes it more likely the system will "do the right thing" even if somebody forgets a `fflush` somewhere, etc..

projectgus, 2015-10-07:
> Alternately, I think it could be argued that in an environment like this, using buffering on stdout should be an explicit choice of the app developer, and if they're actually doing something that [...]

I'm coming around to this conclusion as well. I think unbuffered is a sensible default, if newlib didn't buffer by default then I doubt we'd be having any of this conversation.

BTW, just out of curiosity, what are the circumstances where the esp8266 will software crash in a way that the TX FIFO doesn't empty? The most common crashes I see are unhandled exceptions and WDT timeouts, both of which I thought would empty the FIFO (provided newlib has flushed the stdout RAM buffer.)

kanflo, 2015-10-07:
> Alternately, I think it could be argued that in an environment like this, using buffering on stdout should be an explicit choice of the app developer

+1

> The most common crashes I see are unhandled exceptions and WDT timeouts, both of which I thought would empty the FIFO (provided newlib has flushed the stdout RAM buffer.)

That is a rather nice thought actually. I would expect that to not be the case however. If we are in control of the abort handler (haven't checked), that could/should be implemented. For WDT I assume the HW pulls the plug without any SW interference.

projectgus, 2015-10-07:
> +1

That's a unanimous vote for "disabled stdout buffering by default". Decided!

> If we are in control of the abort handler (haven't checked), that could/should be implemented. 

We don't have abort() yet, thanks for the reminder (added as #54). The fatal exception handler is under our control (currently in app_main.c) and it flushes the TX FIFO before resetting.

> For WDT I assume the HW pulls the plug without any SW interference.

Yeah, you're right. I was thinking on the case mentioned where you printf("about to do something risky\n"), and then do the potentially risky code, figuring if the code infinite looped then at least there would be enough time for the FIFO to flush before the watchdog kicked in. But, as you point out, this isn't guaranteed for all contexts where the watchdog might trip.

foogod, 2015-10-07:
> BTW, just out of curiosity, what are the circumstances where the esp8266 will software crash in a way that the TX FIFO doesn't empty? The most common crashes I see are unhandled exceptions and WDT [...]

As already pointed out, there are some conditions when the WDT could conceivably trip before things have been emptied (I'm actually not sure that the WDT itself would interfere with emptying the FIFO even if it did trip, but I'm pretty sure that one of the first things that the boot ROM does on any reset is to reinitialize the UART and reset the FIFOs, so anything that causes the chip to reset will probably lose anything in the FIFO at the time).

One point with the WDT, though:  Since the ESDK actually implements a dual-WDT (one that triggers in software first, then the actual hardware WDT later if that fails to work), we could conceivably add code to the software WDT handler to make sure the TXFIFO is flushed before rebooting..

The other situation is anything which causes a (even transient) power interruption to the chip, which is often easier to do than many people think (for example, accidentally driving an output pin high when it's connected to ground externally).  Also, there are certain memory regions which apparently cause the chip hardware to lock up hard if even read accesses are made to them (and I suspect that includes killing the UART)..

projectgus, 2015-10-12:
Closing as #53 fixed the underlying issue, and also turned off line buffering by default (which was the other major issue discussed here).

closed 2015-10-12 by projectgus