SuperHouse/esp-open-rtos #55: Add ability to install custom stdin/stdout handlers

foogod, 2015-10-09:
The Espressif SDK includes the option for an app to install a custom "putc" function if they want stdout to go somewhere other than UART0.  Currently with the newlib stuff, etc, esp-open-rtos does not actually support this, but it would be a good thing to bring back (and shouldn't be too hard).

(As a side-note, it might also be interesting to add support for registering a stderr handler as well)

projectgus, 2015-11-06:
I linked #61 to here as supporting POSIX read, write, close, fcntl transparently on both LWIP & UARTS (and someday probably filesystem operations) overlaps with this work as well.

Something that was added to mainline LWIP (much newer than our revision) this year is the ability to add an offset to the fds that LWIP allocates, which makes it pretty straightforward to identify whether a file descriptor belongs to newlib or LWIP:
http://git.savannah.gnu.org/cgit/lwip.git/commit/?id=8155b8cfb31e877c5c51cc539fa121347193f6ce
https://savannah.nongnu.org/patch/?7702

kanflo, 2016-03-14:
This is a feature I need and if no-one else has started looking into this, I would like to. Regarding the LWIP ```LWIP_SOCKET_OFFSET ```offset, could we apply that patch you liked to @projectgus to the existing EOR LWIP repository (for better or for worse) or is that too hawkish?

Regarding an optional replacement of ``putc``, it would be easy to install a custom handler as you mention @foogod. I think it would be useful to have the option of several (at least two) handlers for both local UART debug output as well as (say) UDP based output for remote debugging.

kanflo, 2016-03-14:
Also, when pulling the LWIP patch, we need to decide on ranges for the various file descriptors part from stdin/out/err: LWIP, SPIFFS, ...