SuperHouse/esp-open-rtos #56: flash file system

yesco, 2015-10-11:
I'm gonno create this a place holder, so I can track it.

If I'm correct, we currently don't have any flash file system under esp-open-rtos. 

nodemcu/lua uses the spiffs which is on https://github.com/pellepl/spiffs
However, their interface only allows one file open at a time. From viewing the
https://github.com/pellepl/spiffs/blob/master/src/spiffs.h it seems to be more 
capable.

Spiffs provides a flat file sytem with no directories, however, filenames aren't limited
and can contain "/" thus if one kept track of cwd/pwd one could easily make directly
listing functions that filtered correctly.

Any other ideas/preferences?

I created a simple test to see what would happen, currently we get linking errors:

      CC /home/knoppix/GIT/esp-lisp/lisp.c
      CC /home/knoppix/GIT/esp-lisp/tlisp.c
      AR build/program.a
      LD build/*.c.out
      ../esp-open-rtos/libc/xtensa-lx106-elf/lib/libc.a(lib_a-fopen.o):(.literal+0x10): undefined reference to `_open_r'
      ../esp-open-rtos/libc/xtensa-lx106-elf/lib/libc.a(lib_a-fopen.o): In function `_fopen_r':
      /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdio/../../../../../newlib/libc/stdio/fopen.c:141: undefined reference to `_open_r'

"/home/gus/" path is curious too ;-)

--- snipp ----

      void filetest() {
          FILE* f = fopen("afile", "r");
          if (!f) {
              f = fopen("afile", "w");
              fprintf(f, "Hello mother!\n");
          }
          int len = 1;
          while (len) {
              char buff[100] = {0};
              len = fread(&buff, 1, sizeof(buff), f);
              if (len > 0) printf(buff);
         }
          fclose(f);
      }

--- snopp ---

projectgus, 2015-10-12:
Good suggestion, and good idea to open an issue to keep track of the options for this.

Any filesystem implementation that uses open() will probably end up needing to dovetail with #55.

SaimenSays, 2016-07-10:
With a few simple adaptions you can also use [RTOS FAT SL](http://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_FAT_SL/FreeRTOS_Plus_FAT_SL.shtml)

kanflo, 2016-07-10:
Spiffs is great! Using FAT filesystems in embedded applications is not a very good idea as it is prone to corruption and has no wear leveling. The latter can be added but a poorly timed power loss may lead to the former.

sheinz, 2016-07-10:
I've integrated spiffs into esp-open-rtos.
Here's [branch](https://github.com/sheinz/esp-open-rtos/tree/feature/spiffs) with spiffs integrated. It includes spiffs image creation and flashing integrated into the build process.
Here's a separate [branch](https://github.com/sheinz/esp-open-rtos/tree/experiments/posix_fs) with spiffs integrated into newlib, so the common POSIX file operations are available.

I intend to merge it and create PR when it is ready.

Regarding number of open file descriptors. Spiffs is configured how many open file descriptors is available.
`size_t fdsBufSize = SPIFFS_buffer_bytes_for_filedescs(&fs, 5);`
In this example 5 files can be simultaneously opened.

projectgus, 2016-07-10:
Love your work @sheinz ! Particularly nice being able to integrate with the build process, that is excellent.

It would be nice to support multiple filesystem types, thanks @SaimenSays for pointing out that FAT implementation. Unfortunately there is a barrrier to using that particular implementation. The code is only available under the GPLv2 license (unless you purchase a commercial license). It's not possible to distribute esp-open-rtos applications under GPL, because we still include several binary libraries from Espressif - and we don't have GPL-licenseable source code for these libraries.

pellepl, 2016-07-13:
@kanflo you're so kind 😉
@sheinz looks great! I'd be happy to review the spiffs config before your branch is merged into eor. I think I  could cram some more bits out of it on the esp. Also, having the magic config would be nice.

sheinz, 2016-07-19:
There's a work in progress [PR](https://github.com/SuperHouse/esp-open-rtos/pull/163) for spiffs integration.
Anyone interested are welcome to have a look.

sheinz, 2016-11-25:
Currently esp-open-rtos supports spiffs for SPI flash and fatfs for SDIO.

closed 2016-11-25 by sheinz