SuperHouse/esp-open-rtos #60: Support for GPIO 16 (and beyond)

kanflo, 2015-10-28:
One thing we are currently lacking is support for handling GPIO > 15. Looking at the Arduino port, GPIO 16 is handled in other registers than GPIOs 0..15:

```
extern void __digitalWrite(uint8_t pin, uint8_t val) {
  if(pin < 16){
    if(val) GPOS = (1 << pin);
    else GPOC = (1 << pin);
  } else if(pin == 16){
    if(val) GP16O |= 1;
    else GP16O &= ~1;
  }
}
```

```GPIOS``` above corresponds to our ```GPIO.OUT_SET``` and ```GP16O``` resides at 0x60000768 where we have no mapping.

projectgus, 2015-10-28:
Yes, the standard GPIO interface in core/esp/gpio.h should support number 16.

GPIO16 is connected internally to the RTC hardware not the GPIO hardware, which is why it can be used for RTC wakeup and things like that.

@foogod has mapped the control registers in the open-libmain branch:
https://github.com/SuperHouse/esp-open-rtos/blob/open-libmain/core/include/esp/rtc_regs.h#L34

He's mentioned this file can probably be brought over into master separately to the rest of the branch.

Some investigation probably still required to add register bit mappings for RTC.GPIO_CONF, etc.

kanflo, 2015-10-29:
Ahaa, that explains why GPIO 16 got the special treatment. Right now I have all the GPIOs I need but if I need #16 and none else has looked into this I'll have a go at it.

projectgus, 2016-02-21:
I just opened a duplicate issue and dumped out some more thoughts about this. Repeating below:

Our `gpio.h` functions currently only support GPIOs 0-15. GPIO16 is controlled by the RTC register bank and is "special" (connected to RTC hardware, alive during sleep, has soft pulldown instead of pullup, takes more clock cycles to toggle, etc.)

However, people using the gpio_xxx wrapper functions should be able to control gpio16 same as the other GPIOs.

My suggested design would be:
* We add specific `gpio16_enable`, `gpio16_read`, etc. functions specific to gpio16. Could be defined in `gpio.h` or maybe in `rtc.h`, as relates to RTC hardware...
* We add `if(gpio == 16)` tests to all generic the `gpio.h` functions that call the gpio16_xxx version if necessary.

... but I'd be open to other designs.

kanflo, 2016-02-22:
I second that. Placing a comment  in ```rtc.h``` on why it is there and not in ```gpio.h``` would be helpful for those of us not knowing GPIO16 is special.

sheinz, 2016-11-25:
Implemented with  https://github.com/SuperHouse/esp-open-rtos/pull/211

closed 2016-11-25 by sheinz