SuperHouse/esp-open-rtos #62: How to use the smart_config which in esp-open-sdk?

hawkhsieh, 2015-11-05:


projectgus, 2015-11-05:
Hi hawkhsieh,

Unfortunately esp-open-rtos is based on Espressif RTOS SDK 0.9.9, the version before they changed from MIT license to Espressif MIT License. The version we use doesn't have the Smartconfig modes, they were added afterwards.

I'm going to convert this to an "enhancement" feature request, because it should be possible for someone to implement a "smartconfig" protocol in esp-open-rtos. It's just a matter of going into promiscuous mode, finding the correct channel, and listening for the data packets.

There's a detailed description of how the data packets work, here:
http://depletionregion.blogspot.com.au/2013/10/cc3000-smart-config-transmitting-ssid.html

Also AFAIK the protocol is not patented or otherwise restricted, except that TI asserts a trademark over the exact phrase "SmartConfig".

hawkhsieh, 2015-11-13:
Is there any open source able to port to this project and supports IOS and Android ?

mikejac, 2015-11-14:
The above description of the data packets are not correct (anymore).

I have some code here https://github.com/mikejac/Various_Stuff/blob/master/shared/com/tholusi/esp-open-rtos/wifi/wifi_smartlink.cpp that works except for one thing: I don't know what the CC3000 sends/advertise/other to signal that the "pairing" process is completed. Meaning the SimpleLink app timeout ....

hawkhsieh, 2015-11-16:
Thank you mikejac. 
Is it possible to make it work on this esp-open-rtos project?

mikejac, 2015-11-16:
It is build on esp-open-rtos, so yes :-)

hawkhsieh, 2016-08-18:
I have tried to build the wifi_smartlink.cpp into my project. The sdk_scaninfo_t in wifi.cpp is not found in the lastest version of rtos sdk and the porting is stucked. @mikejac would you help esp-open-rtos to support  the smart link enhancement, we will very very appreciate.

xlwxdl, 2018-04-20:
Hello， Is there any progress？

whlsxl, 2018-05-05:
@projectgus ESP8266_RTOS_SDK is already change License to Apache License, can we upgrade sdk to the latest version?

projectgus, 2018-05-07:
@whlsxl I'm no longer involved with esp-open-rtos. @UncleRus & @ourairquality are probably the best people to discuss this with.

(For my 2c, if ESP8266 RTOS SDK had been Apache when I was running this project then I would have kept updating with newer versions of it. But, as I said, no longer my call.)

ourairquality, 2018-05-07:
An Apache license is fine with me, and I use it myself for some projects. It is great to see this change, and to see some development of the ESP8266 RTOS SDK. They appear to have updated to FreeRTOS 10 now. Still on an older version of lwip which would be a maintenance problem. It seems under active development so perhaps we should hold back to let it settle a bit and then re-evaluate. @projectgus is an update of lwip likely any time soon?

There also appears to be an effort to harmonise the ESP8266 RTOS SDK with the ESP32 code. That would be great too if it makes it easier to maintain code that runs on either.

Guess other decisions need to be made. What C library and many other matters. Would a fork be just a different bundle of drivers etc and otherwise largely following the ESP8266 RTOS SDK? How much can the gaps be closed?? Can we expect to work together now, to follow fixes and improvements either way, to submit fixes and improvements to both projects? I think the community will move forward much faster now, a good turn of events.