SuperHouse/esp-open-rtos #63: Some functions missing from Espressif headers

foogod, 2015-11-08:
There have been a few occasions now where folks have found functions from the published Espressif SDK API which are implemented, but for some reason are simply not present in the headers Espressif provided with the RTOS 0.9.9 branch of the SDK.

We should go through and check all the current SDK APIs to make sure that if they are present in the current implementation, they are also defined in the headers in an appropriate place so people can actually use them.

(the most recent of these and the one that triggered opening this issue is that apparently `sdk_system_adc_read` is present in the libraries but missing from the header files. I know there have been others, though..)

projectgus, 2016-03-01:
We got some of these in #104. Still a good suggestion to look through the current SDK headers and see what else might be missing.

SaimenSays, 2016-09-11:
Tried to compile the I2S functionality from [https://github.com/cnlohr/esp8266ws2812i2s](https://github.com/cnlohr/esp8266ws2812i2s). But compilation fails tbecause of missing function `i2c_writeReg_Mask_rom()`. So maybe this is one of the missing functions?

projectgus, 2016-09-12:
@SaimenSays these functions are a bit of an unusual case, because they're only provided in ROM in the Espressif non-RTOS SDK (not part of the official SDK):

This is from `eagle.rom.addr.v6.ld` in the non-RTOS SDK:
```
PROVIDE ( rom_i2c_readReg = 0x40007268 );
PROVIDE ( rom_i2c_readReg_Mask = 0x4000729c );
PROVIDE ( rom_i2c_writeReg = 0x400072d8 );
PROVIDE ( rom_i2c_writeReg_Mask = 0x4000730c );
```

There are no headers for these functions in Espressif's SDK, because they're seemingly not intended for end user use. @cnlohr added #defines from the names he used (`i2c_xxx`) to the `rom_xxx` names here:
https://github.com/cnlohr/esp8266ws2812i2s/blob/master/user/ws2812_i2s.c#L66

... but he doesn't seem to have function prototypes for them either (the compiler will output a warning about a missing function prototype, but it'll use a default prototype.) You could probably figure out a proper header prototype for them, though...

Short answer: If you add the lines above to ld/rom.ld in esp-open-rtos, and disable warnings as errors, then this will probably compile & link. If you do figure out the prototypes for them, please send us a PR to add the linker entries and the lines in a rom function header file. :)

cnlohr, 2016-09-12:
You know, I didn't actually write those.  As far as I can remember, they were copied and pasted from something from espressif, I think something that had to do with I2S.  It appears the original reference may be lost to the sands of time - at least according to a recent google search.  

I wish I had taken a closer look or added a note.

Those # defines are basically useless.   What is needed is proper prototypes for use of anything like that.  I have done it for some other functions like the MD5 functions in the mask rom, but not much else that I can remember.

SaimenSays, 2016-09-12:
Thank you guys, I will give it a try in next days...

projectgus, 2016-09-12:
Ah, it's here from @Spritetm's mp3decoder example:
https://github.com/espressif/ESP8266_MP3_DECODER/blob/master/mp3/driver/i2s_freertos.c

This repo doesn't contain function prototypes either, but I believe they are as follows:

```
uint8_t rom_i2c_readReg(uint8_t block, uint8_t host_id, uint8_t reg_add);
uint8_t rom_i2c_readReg_Mask(uint8_t block, uint8_t host_id, uint8_t reg_add, uint8_t msb, uint8_t lsb);
void rom_i2c_writeReg(uint8_t block, uint8_t host_id, uint8_t reg_add, uint8_t pData);
void rom_i2c_writeReg_Mask(uint8_t block, uint8_t host_id, uint8_t reg_add, uint8_t msb, uint8_t lsb, uint8_t indata);
```

(The ABI default will be uint32_t, so if the above prototypes don't work try changing them to uint32_t. It looks like they're all correctly labelled as 8-bit arguments, though.)

The macros wrapped around these use token munging to mix in some constant values, so it's worth taking another look at them at least for that.

zheyuejiang, 2020-07-17:
i met the same bug, info: undefined reference to `sdk_rom_i2c_writeReg_Mask'. Use the RTOS developed. How to fix?