SuperHouse/esp-open-rtos #64: IROM0_LEN wrong in linker script?

DagAgren, 2015-11-12:
This does not seem quite right:

    /* Non-OTA sizes */
    #if FLASH_SIZE == 2 /* 256kB */
    #define IROM0_LEN 0x3C000
    #elif FLASH_SIZE == 4 /* 512kB */
    #define IROM0_LEN 0x7C000
    #elif FLASH_SIZE == 8 /* 1MB */
    #define IROM0_LEN 0xFC000
    #elif FLASH_SIZE == 16 /* 2MB */
    #define IROM0_LEN 0x1FC000
    #elif FLASH_SIZE == 32 /* 4MB */
    #define IROM0_LEN 0x3FC000
    #else /*FLASH_SIZE*/
    #error "Unknown flash size for linker script. Check the FLASH_SIZE variable in your Makefile."
    #endif

As the IROM starts at 0x20000, should these not be 0x1c0000, 0x5c0000, 0xdc0000, 0x1dc0000 and 0x3dc0000?

projectgus, 2015-11-12:
Yes, you're right. And (as mentioned in the other issue) they don't account for the final wifi configuration sector either.

projectgus, 2015-11-12:
Probably we can just stop doing this entirely though, as suggested - link for an 8Mbit (largest addressable) image and let the runtime flash size (picked up from the first sector) be used for any other flash size lookups. 

@foogod do you have any objection to removing all this?

foogod, 2015-11-12:
That seems fine to me.. frankly, I can't think of any reason we should really have/want to care about the flash size in the compile/link phases anyway.  If the resulting image is too large for a particular device, esptool should be smart enough to detect that and print a reasonable error when trying to flash it to that specific device.  (If it isn't, that's something that should be fixed in esptool.py IMHO)

projectgus, 2015-11-13:
OK. The only thing I can think of is not having the WiFi config sector collide with the code. But we check for that now, so we probably don't need to start doing it. And pretty soon we should be able to change that behavbiour if we want to, anyhow.

commit fbea34b0 (2015-11-13): Linker script: Fix IROM0_LEN calculation

Use linker script arithmetic (yay?) to calculate flash size in bytes
from FLASH_SIZE in megabits.

Closes #64
closed 2015-11-13 by projectgus