SuperHouse/esp-open-rtos #65: ROM functions

DagAgren, 2015-11-13:
There are a lot of functions available in the ESP8266 ROM. The Espressif RTOS SDK doesn't seem to list most of them in its linker script, however, and esp-open-rtos seems to have inherited that.

Would it be a good idea to add all of them to the linker script, and provide headers at least for the ones that are known for some reason or other? Possibly by prefixing them with rom_ or something similar.

Some of them are C library functions, too, but I don't know how well they work compared to what newlib provides. It might be worth looking into saving space by using them if they are not already.

DagAgren, 2015-11-13:
This is the list of functions provided in the non-RTOS SDK, 1.4.1:

	PROVIDE ( Cache_Read_Disable = 0x400047f0 );
	PROVIDE ( Cache_Read_Enable = 0x40004678 );
	PROVIDE ( FilePacketSendReqMsgProc = 0x400035a0 );
	PROVIDE ( FlashDwnLdParamCfgMsgProc = 0x4000368c );
	PROVIDE ( FlashDwnLdStartMsgProc = 0x40003538 );
	PROVIDE ( FlashDwnLdStopReqMsgProc = 0x40003658 );
	PROVIDE ( GetUartDevice = 0x40003f4c );
	PROVIDE ( MD5Final = 0x40009900 );
	PROVIDE ( MD5Init = 0x40009818 );
	PROVIDE ( MD5Update = 0x40009834 );
	PROVIDE ( MemDwnLdStartMsgProc = 0x400036c4 );
	PROVIDE ( MemDwnLdStopReqMsgProc = 0x4000377c );
	PROVIDE ( MemPacketSendReqMsgProc = 0x400036f0 );
	PROVIDE ( RcvMsg = 0x40003eac );
	PROVIDE ( SHA1Final = 0x4000b648 );
	PROVIDE ( SHA1Init = 0x4000b584 );
	PROVIDE ( SHA1Transform = 0x4000a364 );
	PROVIDE ( SHA1Update = 0x4000b5a8 );
	PROVIDE ( SPI_read_status = 0x400043c8 );
	PROVIDE ( SPI_write_status = 0x40004400 );
	PROVIDE ( SPI_write_enable = 0x4000443c );
	PROVIDE ( Wait_SPI_Idle = 0x4000448c );
	PROVIDE ( SPIEraseArea = 0x40004b44 );
	PROVIDE ( SPIEraseBlock = 0x400049b4 );
	PROVIDE ( SPIEraseChip = 0x40004984 );
	PROVIDE ( SPIEraseSector = 0x40004a00 );
	PROVIDE ( SPILock = 0x400048a8 );
	PROVIDE ( SPIParamCfg = 0x40004c2c );
	PROVIDE ( SPIRead = 0x40004b1c );
	PROVIDE ( SPIReadModeCnfig = 0x400048ec );
	PROVIDE ( SPIUnlock = 0x40004878 );
	PROVIDE ( SPIWrite = 0x40004a4c );
	PROVIDE ( SelectSpiFunction = 0x40003f58 );
	PROVIDE ( SendMsg = 0x40003cf4 );
	PROVIDE ( UartConnCheck = 0x40003230 );
	PROVIDE ( UartConnectProc = 0x400037a0 );
	PROVIDE ( UartDwnLdProc = 0x40003368 );
	PROVIDE ( UartGetCmdLn = 0x40003ef4 );
	PROVIDE ( UartRegReadProc = 0x4000381c );
	PROVIDE ( UartRegWriteProc = 0x400037ac );
	PROVIDE ( UartRxString = 0x40003c30 );
	PROVIDE ( Uart_Init = 0x40003a14 );
	PROVIDE ( _DebugExceptionVector = 0x40000010 );
	PROVIDE ( _DoubleExceptionVector = 0x40000070 );
	PROVIDE ( _KernelExceptionVector = 0x40000030 );
	PROVIDE ( _NMIExceptionVector = 0x40000020 );
	PROVIDE ( _ResetHandler = 0x400000a4 );
	PROVIDE ( _ResetVector = 0x40000080 );
	PROVIDE ( _UserExceptionVector = 0x40000050 );
	PROVIDE ( __adddf3 = 0x4000c538 );
	PROVIDE ( __addsf3 = 0x4000c180 );
	PROVIDE ( __divdf3 = 0x4000cb94 );
	PROVIDE ( __divdi3 = 0x4000ce60 );
	PROVIDE ( __divsi3 = 0x4000dc88 );
	PROVIDE ( __extendsfdf2 = 0x4000cdfc );
	PROVIDE ( __fixdfsi = 0x4000ccb8 );
	PROVIDE ( __fixunsdfsi = 0x4000cd00 );
	PROVIDE ( __fixunssfsi = 0x4000c4c4 );
	PROVIDE ( __floatsidf = 0x4000e2f0 );
	PROVIDE ( __floatsisf = 0x4000e2ac );
	PROVIDE ( __floatunsidf = 0x4000e2e8 );
	PROVIDE ( __floatunsisf = 0x4000e2a4 );
	PROVIDE ( __muldf3 = 0x4000c8f0 );
	PROVIDE ( __muldi3 = 0x40000650 );
	PROVIDE ( __mulsf3 = 0x4000c3dc );
	PROVIDE ( __subdf3 = 0x4000c688 );
	PROVIDE ( __subsf3 = 0x4000c268 );
	PROVIDE ( __truncdfsf2 = 0x4000cd5c );
[... 254 lines cut ...]
	PROVIDE ( uart_buff_switch = 0x400038a4 );
	PROVIDE ( uart_div_modify = 0x400039d8 );
	PROVIDE ( uart_rx_intr_handler = 0x40003bbc );
	PROVIDE ( uart_rx_one_char = 0x40003b8c );
	PROVIDE ( uart_rx_one_char_block = 0x40003b64 );
	PROVIDE ( uart_rx_readbuff = 0x40003ec8 );
	PROVIDE ( uart_tx_one_char = 0x40003b30 );
	PROVIDE ( wepkey_128 = 0x4000bc40 );
	PROVIDE ( wepkey_64 = 0x4000bb3c );
	PROVIDE ( xthal_bcopy = 0x40000688 );
	PROVIDE ( xthal_copy123 = 0x4000074c );
	PROVIDE ( xthal_get_ccompare = 0x4000dd4c );
	PROVIDE ( xthal_get_ccount = 0x4000dd38 );
	PROVIDE ( xthal_get_interrupt = 0x4000dd58 );
	PROVIDE ( xthal_get_intread = 0x4000dd58 );
	PROVIDE ( xthal_memcpy = 0x400006c4 );
	PROVIDE ( xthal_set_ccompare = 0x4000dd40 );
	PROVIDE ( xthal_set_intclear = 0x4000dd60 );
	PROVIDE ( xthal_spill_registers_into_stack_nw = 0x4000e320 );
	PROVIDE ( xthal_window_spill = 0x4000e324 );
	PROVIDE ( xthal_window_spill_nw = 0x4000e320 );

	PROVIDE ( Te0 = 0x3fffccf0 );
	PROVIDE ( UartDev = 0x3fffde10 );
	PROVIDE ( flashchip = 0x3fffc714);

DagAgren, 2015-11-13:
And this seems to be a collection of published and reverse-engineered headers for at least some of them:

https://github.com/pvvx/MinEspSDKLib/tree/master/include/bios

foogod, 2015-11-13:
This is actually something that's been on the back of my "to-do" list..  The one concern, as you mentioned, is that there are some functions in the ROM which have the same names as things in other libraries we currently build with (newlib, etc).  and we want to make sure we don't get unexpected conflicts, but I do agree that it would ultimately be good to add all these in in some form or other (and provided they're properly compatible, I don't see any reason not to use the ROM routines for the standard library functions they provide (we just need to make sure we thoroughly test that they work as expected so we don't get surprises later))..

Note that those last three on your list are actually in the RAM space, and are apparently defining certain data structures to be at fixed positions in data RAM.  We should probably try to determine exactly why they're doing that (i.e. do some of the ROM routines actually expect them to be there, or is it just something Espressif chose to do for some arbitrary reason that we don't necessarily want to copy?)..

I wasn't aware of those MinEspSDK header files.. that's a potentially very useful find (and appears to be MIT licensed, which means we can actually use it :) )

projectgus, 2015-11-13:
FWIW, the [PROVIDE directive](https://sourceware.org/binutils/docs/ld/PROVIDE.html#PROVIDE) in the ROM linker scripts mean any symbol provided in libraries should take priority over the symbol in ROM, if they're available at link time (this was something I didn't realise until yesterday when I was looking at the OTA- vs non-OTA- uses of Cache_Read_Enable). I don't know if it's possible to use a different directive to invert that priority so the ROM functions replace parts of other libraries. At least without preprocessing the libraries to strip symbols - the same kind of "magic" preprocessing we're trying to remove elsewhere!!

It'd be very interesting to understand why Espressif stripped back the ROM functions list for RTOS to begin with. My suspicion, like @foogod mentions, is that are assumptions about shared data structure sizes and offsets in static RAM (like the last 3 symbol names) shared between the IoT SDK and the ROM functions. That doesn't explain a lot of the lower level functions, though.

Personally I'm not that interested in moving functionality we have in readable source code back into the ROM functions. Apart from the performance/size gain, the functions in the boot ROM don't seem too conceptually different to me than the binary blob libraries. That said, I don't feel super strongly about it so if someone sends a PR that cleanly and safely re-integrates some ROM functions then I'm not going to oppose it on Free Software Purity grounds. :)

foogod, 2015-11-13:
Ah, I also had not realized that PROVIDE is basically a "weak" definition..  Based on the ld docs it looks like the strong equivalent is just `symbol = value;` (without the PROVIDE)..

In this case, I think it'd be fine to add them all in as PROVIDEs, and then we may want to discuss further which, if any, should be converted to non-PROVIDEs to override stdlib functions, etc..

Looking at some of the older versions of the non-RTOS SDK, they're missing a lot of the ROM symbols too, so it looks to me like Espressif was just being hesitant about what they published about the ROM in earlier versions of things, and over time have added more and more to their linker scripts..

It is possible, though, that there are some ROM routines that make assumptions about the rest of the SDK that may not be true for the RTOS version, etc, so we should probably be a bit careful about adding headers for functions without testing first to make sure that they actually work properly with esp-open-rtos..

DagAgren, 2015-11-13:
Reading too much into what Espressif does seems a bit risky, their behaviour seems quite infuriatingly arbitrary a lot of the time.

Anyway, providing headers in some subdirectory for all the functions seems safe enough even without testing, just with a "here be dragons" notice perhaps. At least if they are renamed so they never collide.