SuperHouse/esp-open-rtos #66: Is libhal actually needed?

DagAgren, 2015-11-13:
In my test build, only two libhal functions are actually used, xthal_get_ccount and xthal_set_intset. xthal_get_ccount is provided in ROM, so it can easily be eliminated. That just leaves one instance of xthal_set_intset in FreeRTOS. If that was replaced, then libhal could be dropped as a dependency.

foogod, 2015-11-13:
This was another thing I was meaning to look into at some point, but to be honest was never really that high on the priority list..

I don't see any reason not to use the xthal_get_ccount from the ROM instead.. as for xthal_set_intset, it would be good if we didn't have to make custom changes to the FreeRTOS stuff just for this..  We could potentially just put our own implementation (possibly a weak symbol) in our libs, though, so it wouldn't have to pull in libhal for just that one function..

DagAgren, 2015-11-13:
When I looked at the Espressif FreeRTOS code, it looked somewhat ghastly, so I'm not sure trying to keep their code untouched is the best option. Of course, branching off it would add another maintenance burden...

resetnow, 2016-09-03:
I assume this can get closed after #198 was merged.

projectgus, 2016-09-04:
Yes, fantastic - thanks!

closed 2016-09-04 by projectgus