SuperHouse/esp-open-rtos #67: Scanning for APs causes fatal exception

recursify, 2015-11-14:
Version: 5cd31c649a157d0635bf65e4b8551e955333bb72

I'm trying to scan for APs, and I'm getting a `fatal exception (28)`.  The call to `sdk_wifi_station_scan` returns successful, but then I am getting a fatal exception... I suspect on the invocation of the callback.  Can anyone else repro? 

```c
#include "espressif/esp_common.h"
#include "esp/uart.h"

#include <string.h>

#include "FreeRTOS.h"
#include "task.h"

#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include "lwip/netdb.h"
#include "lwip/dns.h"
#include "ssid_config.h"

static void scan_done(void *arg, sdk_scan_status_t status)

{
    printf("scan done with status %i\n", status);
}

void scan_task(void *pvParameters) {
    printf("About to start scan\n");
    if(sdk_wifi_get_opmode() == SOFTAP_MODE) {
        printf("ap mode can't scan !!!\r\n");
        return;
    }
    bool s = sdk_wifi_station_scan(NULL, &scan_done);
    printf("Scan call done, status: %i\n", s);

    while(1) {
        printf("scan task\n");
        vTaskDelay(4000 / portTICK_RATE_MS);
    }
}

void user_init(void)
{
    uart_set_baud(0, 74880);
    printf("SDK version:%s\n", sdk_system_get_sdk_version());
    sdk_wifi_set_opmode(STATION_MODE);
    xTaskCreate(&scan_task, (signed char *)"scan_task", 512, NULL, 2, NULL);
}
```

Output:
```
ESP-Open-SDK ver: 0.0.1 compiled @ Nov 14 2015 00:31:02

phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(18:fe:34:a6:04:74)
add if0
About to start scan
scandone
Scan call done, status: 1
scan task
Fatal exception (28):
epc1=0x4010689d
epc2=0x00000000
epc3=0x4022218e
excvaddr=0x00000000
depc=0x00000000
excsave1=0x40233489
```

foogod, 2015-11-14:
I don't see anything obviously wrong with the code, but I haven't ever tried to do an `sdk_wifi_station_scan` myself..  Exception cause 28 is LoadProhibitedCause, which (combined with EXCVADDR=0x00000000) means there's a null pointer access somewhere..

In the 'utils' directory of the esp-open-rtos distribution is a utility called 'addrsource'.  Can you cd into your app's directory (i.e. the one you run 'make' from) and then run `addrsource 0x4010689d`?

This should print out a bunch of info about what source line / opcode generated the exception..

recursify, 2015-11-14:
Hi Alex,

```
:> utils/addrsource 0x4010689d examples/open-rtos-app/build/app.out 0x4010689d is in memcmp (../../../../../newlib/libc/string/memcmp.c:64).
59	../../../../../newlib/libc/string/memcmp.c: No such file or directory.
Line 64 of "../../../../../newlib/libc/string/memcmp.c" starts at address 0x4010689d <memcmp+9> and ends at 0x401068aa <memcmp+22>.
Dump of assembler code for function memcmp:
   0x40106894 <+0>:	movi.n	a5, 0
   0x40106896 <+2>:	j	0x401068b0 <memcmp+28>
   0x40106899 <+5>:	add.n	a6, a2, a5
   0x4010689b <+7>:	addi.n	a5, a5, 1
   0x4010689d <+9>:	l8ui	a7, a6, 0
   0x401068a0 <+12>:	add.n	a6, a3, a5
   0x401068a2 <+14>:	addi.n	a6, a6, -1
   0x401068a4 <+16>:	l8ui	a6, a6, 0
   0x401068a7 <+19>:	beq	a7, a6, 0x401068b0 <memcmp+28>
   0x401068aa <+22>:	sub	a2, a7, a6
   0x401068ad <+25>:	ret.n
   0x401068af <+27>:	extui	a4, a0, 23, 10
   0x401068b2 <+30>:	excw
   0x401068b5 <+33>:	ret.n
End of assembler dump.
```

foogod, 2015-11-14:
Hmm.. looks like something inside the actual scanning routine (which I assume runs in the background in another task) is calling `memcmp` with a null pointer..  Unfortunately a lot of that code is still something of a black-box since Espressif won't provide the source code (and I haven't gotten to the point of reverse engineering those routines yet)..

You might try passing in an actual scan_config struct instead of NULL and see if that changes anything..  (according to the docs, it looks like NULL should be valid, but it's possible there's a bug)

recursify, 2015-11-14:
Thanks for the tips, Alex, I'll try that this weekend!

And if I get something working I'll try and create a minimal example to put in the `/examples` folder,

cheers,

recursify, 2015-11-15:
Interesting - I'm not able to repro at home.  Potentially related to the number of APs at home vs at work, or that my work router is 5ghz?

Will confirm on Monday.

foogod, 2015-11-16:
The current "Espressif MIT License" is a modified MIT license with additional restrictions on use added by Espressif, which makes it not technically open-source-compatible, and specifically, incompatible with the BSD licensing of the esp-open-rtos project.

The latest version of their code released under a compatible license was an older release (approx version 0.9.9) of their RTOS SDK, which is what esp-open-rtos is based on.  Unfortunately any later versions of the Espressif libraries are "tainted goods" and we can't use them directly.  (This also means we're a bit behind the current state of the Espressif SDKs when it comes to feature support.  The hope is that once we pull apart more of the RTOS SDK libs we'll be able to improve this situation substantially..)

chen-hsiang0020, 2017-08-17:
I am seeing the same issue. In my office space, there are 5 different APs with the same SSID and I can reproduce the same crash with the wifi_scan example code in the esp-open-rtos repo. Is anybody work on this? It can be a show stopper because it results in a stack dump....

ourairquality, 2017-08-17:
@chen-hsiang0020 Can you post a stack dump to give some clues?

recursify, 2017-08-17:
Also @chen-hsiang0020 - does the workaround listed above work for you as well?  

> Set the station_config to some other ssid before calling sdk_wifi_station_scan. The scan will succeed.

chen-hsiang0020, 2017-08-18:
@recursify - tested and the workaround doesn't work for me. I have the same execption cause 28 and EXCVADDR=0x00000000, but my epc1=0x40106269 is different from yours.

@ourairquality - The crash happen randomly, but every time with the same stack dump signature. Here is the stack dump

```
ESP-Open-SDK ver: 0.0.1 compiled @ Aug 18 2017 12:02:57
phy ver: 273, pp ver: 8.3

SDK version:0.9.9

mode : sta(18:fe:34:cb:9f:97)
add if0
Fatal exception (28):
epc1=0x40106269

0x40106269: memcmp at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/string/../../../../../newlib/libc/string/memcmp.c:64

epc2=0x00000000
epc3=0x40203101

0x40203101: prvIdleTask at /home/feng0020/ESP8266/esp-open-rtos/FreeRTOS/Source/tasks.c:4173

excvaddr=0x00000000
depc=0x00000000
excsave1=0x402182b5

0x402182b5: sdk_cnx_rc_search at ??:?

Registers:
a0 402182b5 a1 3fff19d0 a2  00000000 a3  3ffec710

0x402182b5: sdk_cnx_rc_search at ??:?

a4  00000006 a5  00000001 a6  00000000 a7  00000004
a8  3ffec724 a9  00000016 a10 ffffff60 a11 0000002d
a12 00000002 a13 3ffec710 SAR 00000010

Stack: SP=0x3fff19d0
0x3fff19d0: 40216944 3fff1a60 3ffe9260 3ffec700

0x40216944: sdk_scan_parse_beacon at ??:?

0x3fff19e0: 3ffe96d8 3fff1a60 3ffe9260 3fff1a60
0x3fff19f0: 0000009c 3fff0af8 3fff09cc 00325357
0x3fff1a00: a5a5a5a5 a5a5a5a5 a5a5a5a5 00000000
0x3fff1a10: 00000040 000c3500 2f33363c 402038cc

0x402038cc: xPortSysTickHandle at /home/feng0020/ESP8266/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:230

0x3fff1a20: 3ffec70a 3ffec710 00000029 3ffea788
0x3fff1a30: 000000fa 00000000 4021721a 00000000

0x4021721a: sdk_ieee80211_parse_wmeparams at ??:?

0x3fff1a40: 3fff5470 3ffec700 3ffec7fa 40100bb0

0x40100bb0: SV_ISR at /home/feng0020/ESP8266/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:230

Free Heap: 44300
_heap_start 0x3fff1120 brk 0x3fff5900 supervisor sp 0x3ffffb00 sp-brk 41472 bytes
arena (total_size) 18400 fordblks (free_size) 2828 uordblocks (used_size) 15572

ets Jan  8 2013,rst cause:2, boot mode:(1,6)

ets Jan  8 2013,rst cause:4, boot mode:(1,6)
```
Here is the UART prints in case of success:
```
----------------------------------------------------------------------------------
Wi-Fi networks
----------------------------------------------------------------------------------
HP-Print-0D-ENVY 4500 series (34:64:a9:84:83:0d) RSSI: -72, security: Open
InnoSquare_CWS (9c:d6:43:29:b0:e0) RSSI: -42, security: WPA/WPA2/PSK
InnoSquare_CWS2 (04:18:d6:c2:5d:af) RSSI: -59, security: WPA2/PSK
Lawsnote-2.4G (54:a0:50:59:71:9a) RSSI: -90, security: WPA2/PSK
InnoSquare_CWS2 (04:18:d6:c2:5c:5c) RSSI: -83, security: WPA2/PSK
Lily (74:d0:2b:dc:19:8c) RSSI: -89, security: WPA/WPA2/PSK
WASAI Technology (78:44:76:dd:45:a8) RSSI: -88, security: WPA/WPA2/PSK
InnoSquare_CWS (9c:d6:43:29:af:b0) RSSI: -51, security: WPA/WPA2/PSK
InnoSquare_CWS (9c:d6:43:2a:49:00) RSSI: -54, security: WPA/WPA2/PSK
InnoSquare_CWS (9c:d6:43:2a:4b:30) RSSI: -58, security: WPA/WPA2/PSK
InnoSquare_CWS (9c:d6:43:2a:4a:b0) RSSI: -61, security: WPA/WPA2/PSK
InnoSquare_CWS (9c:d6:43:2a:4b:00) RSSI: -62, security: WPA/WPA2/PSK
InnoSquare_CWS (9c:d6:43:2a:4b:60) RSSI: -80, security: WPA/WPA2/PSK
Oona (08:62:66:64:af:d7) RSSI: -83, security: WPA2/PSK
MZCAN (80:1f:02:48:c7:80) RSSI: -70, security: WPA2/PSK
NewTaipei (6c:19:8f:b0:23:51) RSSI: -81, security: Open
iTaiwan (6c:19:8f:b0:23:50) RSSI: -85, security: Open
ArienK (66:00:10:64:2c:cb) RSSI: -69, security: WPA2/PSK
AndroidAPBB18 (88:83:22:66:bb:18) RSSI: -67, security: WPA2/PSK
張學友的wifi (32:07:4d:4a:38:58) RSSI: -84, security: WPA2/PSK
InnoSquare_CWS (9c:d6:43:2a:4b:70) RSSI: -48, security: WPA/WPA2/PSK
WZ MakerSpace (c8:3a:35:3d:86:20) RSSI: -75, security: WPA/PSK
BlackBerry Mobile Hotspot 6968 (94:eb:cd:76:96:76) RSSI: -82, security: WPA2/PSK
FRNCi (60:31:97:84:33:cb) RSSI: -56, security: WPA2/PSK
```

ourairquality, 2017-08-18:
@chen-hsiang0020 Thank you. An attempt to fix this has been added to the patch set in https://github.com/SuperHouse/esp-open-rtos/pull/437 see sdk_cnx_rc_search. It think you could just cherry-pick that small patch onto the master branch with some small edits. Does that help?

chen-hsiang0020, 2017-08-18:
@ourairquality - Thank you for your patch. WIth the patch it got passed the NULL pointer error in sdk_cnx_rc_search, but it then trapped in the sdk_scan_parse_beacon with a fatal exeception 28. Error log attached.

I notice that your commit is based on other changes which are not on top of the tree. The following result is just the sdk_cnx_rc_search part of the code build on top of the tree.

Please let me know if you have any thing I can try it out. To me this bug is a show stopper that will force me to use the ESP8266_RTOS_SDK... Please help fixing this issue.

```
Fatal exception (28):
epc1=0x4021848c

0x4021848c: sdk_cnx_add_rc at ??:?

epc2=0x00000000
epc3=0x40203168

0x40203168: prvIdleTask at /home/feng0020/ESP8266/esp-open-rtos/FreeRTOS/Source/tasks.c:4173

excvaddr=0x00000078
depc=0x00000000
excsave1=0x40216ada

0x40216ada: sdk_scan_parse_beacon at ??:?

Registers:
a0 40216ada a1 3fff19d0 a2  3ffe9b40 a3  00000000

0x40216ada: sdk_scan_parse_beacon at ??:?

a4  00000000 a5  00000002 a6  00000003 a7  00000000
a8  3ffe9c7c a9  3ffe9260 a10 3ffe9c74 a11 00000000
a12 3fff1a40 a13 3ffe9260 SAR 0000001f

Stack: SP=0x3fff19d0
0x3fff19d0: 6f6e6e49 61757153 435f6572 00325357
0x3fff19e0: 000001fb 3fff0af8 3fff09cc a5a5a5a5
0x3fff19f0: 0000027c 3fff0af8 3fff09cc 00000000
0x3fff1a00: 3ffed9d6 3ffed9dc 0000000a 3ffea788
0x3fff1a10: 3ffe9b40 00000000 402172a2 00000000

0x402172a2: sdk_ieee80211_parse_wmeparams at ??:?

0x3fff1a20: 3fff5450 3ffed9cc 3ffedaa1 00000000
0x3fff1a30: a5a5a5a5 00000001 3fff1088 40100cd9

0x40100cd9: _xt_isr_handler at /home/feng0020/ESP8266/esp-open-rtos/core/esp_interrupts.c:34

0x3fff1a40: 04310b0b 00640100 0000004e 3ffed9f0

Free Heap: 45652
_heap_start 0x3fff1100 brk 0x3fff58e0 supervisor sp 0x3ffffb00 sp-brk 41504 bytes
arena (total_size) 18400 fordblks (free_size) 4148 uordb
```

ourairquality, 2017-08-18:
Looks like it crashes in sdk_cnx_add_rc and looking at this there was an overflow of the table that might explain why NULLs are being seen here. Also added code for sdk_cnx_remove_rc in case having that helps, and it does appear to compact away holes so it's not clear where they are coming from. There appear to be a few other functions referencing this table but they are static functions so not so easy to replace, but we can keep working on it. Do these help?

The other thought is that it might be a thread safety issue, a call back into these function while another thread that was removing an entry was interrupted.

chen-hsiang0020, 2017-08-19:
I don't have enough experience on reverse engineering this issue, but I am welcome to verify any patch you have for this problem.
My thought is that it is not a thread issue, otherwise it will be shown in more places and more reports should be there already.
I looked through the Espressif's SDK chage log and google searched this issue. So far I have not seen the same issue reported in other places. I will try the Espressif's SDK to see behavior.
Again I really believe this is a critical bug because scan causing  crashes in a office environment. Thank you guys for the wonderful work and please make esp-open-rtos even better.

ourairquality, 2017-08-19:
@chen-hsiang0020 Testing would help thanks, as I can not reproduce it. Could you please try the latest patches in https://github.com/SuperHouse/esp-open-rtos/pull/437 There did appear to be a problem in sdk_cnx_add_rc, it inserted new entries beyond the length sometimes. With luck that was the cause of these problems. If that works then could you please also try removing the null `ptr` guards added in cnx_rc_search sdk_cnx_add_rc as it does not appear that the table should have NULL entries and it would be better to test this assumption.

chen-hsiang0020, 2017-08-20:
It works!! Thank you @ourairquality !! I only take the following 3 commits and the scan is rock solid!!
```
sdk_cnx_rc_search: catch a null pointer dereference. 
sdk_cnx_add_rc: fix overflow of the table, when no match is found. 
sdk_cnx_remove_rc: add source for this function. 
```
I also tested with the null ptr guards removed in cnx_rc_search, as expected the behavior doesn't change. Anyway I will vote for leaving the extra check there.
Thank you @ourairquality !! Great work!!