SuperHouse/esp-open-rtos #68: I2S and SLC register headers

DagAgren, 2015-11-14:
I translated register definitions for the SLC and I2S registers from Espressif sources to try and match the esp-open-rtos style. I also tried to slightly change the naming to be a bit more readable, as this seems in line with the other headers already provided. There's a slight risk I may have guessed wrong in some places, though.

(I also left out the very last definitions for the SLC, because the I couldn't make heads or tails of the code.)

[i2s_regs.h.txt](https://github.com/SuperHouse/esp-open-rtos/files/34655/i2s_regs.h.txt)
[slc_regs.h.txt](https://github.com/SuperHouse/esp-open-rtos/files/34654/slc_regs.h.txt)

Feel free to edit these as you see fit to better match the rest. I have some code too which actually uses these and can serve as a few hints to what this stuff all means, but it's still a mess. I might post it later if anybody's interested, once it's cleaned up.

projectgus, 2015-11-15:
This looks great, thanks! Very interested in what progress you get writing code against these as well.

It looks like it would more or less be able to be merged as-is. The only thing I noticed is using bitfields for the SLCDescriptor struct. I know some people really like using bitfields on embedded, but we've chosen not use them in esp-open-rtos. We'd probably use bitmask & shift values for the first 32 bits in the struct.

Would you be able to please fork the esp-open-rtos repo and send these as a github pull request? It makes collaboration easier and less work tracking/managing changes.

DagAgren, 2015-11-15:
You can thank Espressif for those bitfields. I got rid of them, and made a pull request: https://github.com/SuperHouse/esp-open-rtos/pull/70

projectgus, 2015-11-20:
Closed as #70 is merged. Thanks!

closed 2015-11-20 by projectgus