SuperHouse/esp-open-rtos #71: Fatal exception in netconn

recursify, 2015-11-24:
After connecting to wifi, I'm running the following code in a loop with a small delay:

```
void run(void) {
    struct netconn * conn;
    struct netbuf * buffer;
    struct ip_addr remote_ip;
    uint16_t len;
    void * indata;
    printf("About to create new buffer\n");

    buffer = netbuf_new();
    if(buffer == NULL) {
        printf("buffer create failed!\n");
        return;
    }
    printf("About to alloc buffer \n");
    indata = netbuf_alloc(buffer, 256);
    if(indata == NULL) {
        printf("buffer alloc failed!\n");
        return;
    }

    printf("About to do DNS \n");
    netconn_gethostbyname("kenshi2.chefsteps.internal", &remote_ip);

    printf("About to create new tcp conn \n");
    conn = netconn_new(NETCONN_TCP);
    if(conn == NULL) {
        printf("conn new failed!\n");
        return;
    }

    printf("connecting..\n");
    netconn_connect(conn, &remote_ip, 5555);
    netconn_set_recvtimeout(conn, 20);
    printf("writing..\n");
    netconn_write(conn, (void *)data, 12, NETCONN_NOCOPY);
    printf("receiving..\n");
    netconn_recv(conn, &buffer);

    printf("Closing connection\n");
    netconn_close(conn);
    printf("delete netconn\n");
    netconn_delete(conn);
    printf("delete buffer\n");
    netbuf_delete(buffer);

}

void main_loop(void *pvParameters) {
    //... setup
    while(1) {
        run();
        vTaskDelay(100 / portTICK_RATE_MS);
    }
}
```

Which will run several dozen times before: 

```
About to create new buffer
About to alloc buffer
About to do DNS
About to create new tcp conn
connecting..
Fatal exception (3):
epc1=0x40225aff
epc2=0x00000000
epc3=0x40222379
excvaddr=0x400003e8
depc=0x00000000
excsave1=0x3fff3870
```

And here is the result of addrsource:

```
./utils/addrsource  0x40225aff examples/open-rtos-app/build/app.out
0x40225aff is in tcp_output (/home/vagrant/espressif/esp-open-rtos/lwip/lwip/src/core/tcp_out.c:994).
989	      TCPH_SET_FLAG(seg->tcphdr, TCP_ACK);
990	      pcb->flags &= ~(TF_ACK_DELAY | TF_ACK_NOW);
991	    }
992
993	    tcp_output_segment(seg, pcb);
994	    snd_nxt = ntohl(seg->tcphdr->seqno) + TCP_TCPLEN(seg);
995	    if (TCP_SEQ_LT(pcb->snd_nxt, snd_nxt)) {
996	      pcb->snd_nxt = snd_nxt;
997	    }
998	    /* put segment on unacknowledged list if length > 0 */
Line 994 of "/home/vagrant/espressif/esp-open-rtos/lwip/lwip/src/core/tcp_out.c" starts at address 0x40225aed <tcp_output+665> and ends at 0x40225b3a <tcp_output+742>.
```

recursify, 2015-11-24:
I suspect this is an issue with lwip, but let me know if you need more info.

recursify, 2015-11-25:
Sorry, this is my bad.  I misunderstood the netconn API, and was passing a reference to an already allocated buffer into netconn_recv().

Closing.

closed 2015-11-25 by recursify

projectgus, 2015-11-25:
Hi @recursify, glad you got it sorted. Sorry noone had gotten around to looking at your code. :) Let us know if you hit any more problems.