SuperHouse/esp-open-rtos #74: Linking Issues

andrewclink, 2015-11-30:
Using both eps-open-sdk and another toolchain from origins unknown, I'm having trouble building the examples in eps-open-rtos. While the examples compile fine, I'm having trouble linking.

After the first git clone --recursive and make -C examples/http_get, I was getting undefined reference errors to sdk_ prefixed symbols. I reordered the linker flags to put `$(COMPONENT_ARS)` outside the linker flags group (i.e. after -Wl,--end-group) and now simply receive an error from common.ld where pvPortMalloc is set:

    undefined symbol `malloc' referenced in expression

I'm not sure if that means the sdk_ prefixed symbols are now being found or not. -Wl,--verbose claims the following (with the failed library lookups and contents of the external linker script abridged): 

   attempt to open /usr/local/esp8266_open_rtos/bin/../xtensa-lx106-elf/sysroot/usr/lib/libhal.a succeeded
attempt to open /usr/local/esp8266_open_rtos/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/libgcc.a succeeded
attempt to open ../../libc/xtensa-lx106-elf/lib/libc.a succeeded
attempt to open ./build/sdklib/libmain.a succeeded
attempt to open /usr/local/esp8266_open_rtos/bin/../xtensa-lx106-elf/sysroot/usr/lib/libnet80211.a succeeded
attempt to open /usr/local/esp8266_open_rtos/bin/../xtensa-lx106-elf/sysroot/usr/lib/libphy.a succeeded
attempt to open /usr/local/esp8266_open_rtos/bin/../xtensa-lx106-elf/sysroot/usr/lib/libpp.a succeeded
attempt to open /usr/local/esp8266_open_rtos/bin/../xtensa-lx106-elf/sysroot/usr/lib/libwpa.a succeeded
attempt to open ./build/program.a succeeded
attempt to open ./build/freertos.a succeeded
attempt to open ./build/lwip.a succeeded
attempt to open ./build/core.a succeeded
(./build/core.a)exception_vectors.o
(./build/core.a)app_main.o
(./build/core.a)esp_interrupts.o
../../ld/common.ld:11: undefined symbol `malloc' referenced in expression

I'm looking through the makefile to try to understand the build process further. At this point I assume that the issue has to do with a problem in my toolchain, or at least the way that it is built by default on Mac OS X 10.11. I don't have a linux machine anymore to determine what's different. Either way, it would be nice if the examples built out of the box :)

projectgus, 2015-11-30:
Hi Andrew,

Thanks for getting in touch!

To work with esp-open-rtos, esp-open-sdk must build the toolchain with the `STANDALONE=n` option set. Otherwise by default esp-open-rtos combines a set of Espressif binary SDK libraries into the toolchain sysroot (you can see them getting picked up in the linker path output similar to `.../xtensa-lx106-elf/sysroot/usr/lib/libnet80211.a`, etc.)

If you rebuild the toolchain with `STANDALONE=n`, the examples should link correctly.

Please let us know if you have any other problems getting up and running.

Angus

andrewclink, 2015-11-30:
Hi Angus,

I thought for sure I had included that but I'll double check and try rebuilding to be sure. I'll update after I check.

Thanks!

projectgus, 2015-11-30:
Good luck! It is possible if you'd previously built with STANDALONE=y then the libraries that were copied to the sysroot directory the first time weren't deleted when you rebuilt with STANDALONE=n.

projectgus, 2015-11-30:
I realised we can fix this and have our linker steps work with the standalone toolchain as well. Now the libraries internal to esp-open-rtos should take priority over anything in sysroot.

closed 2015-11-30 by projectgus

andrewclink, 2015-11-30:
Embarrassingly, rebuilding the toolchain with `STANDALONE=n` did solve the issue. I am certain I built it that way the first time, but there were issues building on OS X related to Crosstool and I must have missed it the second time. Thanks so much for the push in the right direction!

projectgus, 2015-11-30:
No problem, it's an easy footnote to miss. I think it won't be necessary from now on too, so thank you for the prod to re-examine the problem and find a workaround - there are probably plenty of people who silently gave up on esp-open-rtos without getting in touch.

only1chi, 2016-09-21:
Hi, I have the same problem, except I did follow the recommendations and built the SDK standalone. 
I still have the same problem.
I'm using eclipse as an IDE to build my application and have the CDT options configured to reference the correct compiler.
I get the same error when I build with IDE, or with command line.

only1chi, 2016-09-21:
Here is a console output of my build process:

mkdir -p build/
xtensa-lx106-elf-ar cru build/core.a build/core//esp_interrupts.o build/core//app_main.o build/core//esp_gpio_interrupts.o build/core//esp_spi.o build/core//sdk_compat.o build/core//esp_gpio.o build/core//phy_info.o build/core//esp_phy.o build/core//newlib_syscalls.o build/core//esp_iomux.o build/core//esp_timer.o build/core//esp_hwrand.o build/core//debug_dumps.o build/core//exception_vectors.o build/core//spiflash-cache-enable.o build/core//cplusplus_operators.o /home/chiz/share/esp-open-rtos/core//esp_interrupts.c /home/chiz/share/esp-open-rtos/core//app_main.c /home/chiz/share/esp-open-rtos/core//esp_gpio_interrupts.c /home/chiz/share/esp-open-rtos/core//esp_spi.c /home/chiz/share/esp-open-rtos/core//sdk_compat.c /home/chiz/share/esp-open-rtos/core//esp_gpio.c /home/chiz/share/esp-open-rtos/core//phy_info.c /home/chiz/share/esp-open-rtos/core//esp_phy.c /home/chiz/share/esp-open-rtos/core//newlib_syscalls.c /home/chiz/share/esp-open-rtos/core//esp_iomux.c /home/chiz/share/esp-open-rtos/core//esp_timer.c /home/chiz/share/esp-open-rtos/core//esp_hwrand.c /home/chiz/share/esp-open-rtos/core//debug_dumps.c /home/chiz/share/esp-open-rtos/core//exception_vectors.S /home/chiz/share/esp-open-rtos/core//spiflash-cache-enable.S /home/chiz/share/esp-open-rtos/core//cplusplus_operators.cpp
grep -v "^#" /home/chiz/share/esp-open-rtos/lib/libmain.remove | cat > build/sdklib/libmain.remove
grep -v "^#" /home/chiz/share/esp-open-rtos/lib/libnet80211.remove | cat > build/sdklib/libnet80211.remove
grep -v "^#" /home/chiz/share/esp-open-rtos/lib/libphy.remove | cat > build/sdklib/libphy.remove
grep -v "^#" /home/chiz/share/esp-open-rtos/lib/libpp.remove | cat > build/sdklib/libpp.remove
grep -v "^#" /home/chiz/share/esp-open-rtos/lib/libwpa.remove | cat > build/sdklib/libwpa.remove
xtensa-lx106-elf-gcc -I./ -I./include -I/home/chiz/share/esp-open-rtos/include -I/home/chiz/share/esp-open-rtos/libc/xtensa-lx106-elf/include -I/home/chiz/share/esp-open-rtos/FreeRTOS/Source/include -I/home/chiz/share/esp-open-rtos/FreeRTOS/Source/portable/esp8266 -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include -I/home/chiz/share/esp-open-rtos/lwip/include -I/home/chiz/share/esp-open-rtos/lwip/include -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/posix -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/ipv4 -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/ipv4/lwip -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/lwip -I/home/chiz/share/esp-open-rtos/core/include -I/home/chiz/share/esp-open-rtos/open_esplibs/include  -mlongcalls -mtext-section-literals -DGITSHORTREV=\"22c480b\" -Wall -Wl,-EL -nostdlib  -ffunction-sections -fdata-sections -g -O2 -std=gnu99 -Og -ggdb -DGDBSTUB_FREERTOS=0 -DENABLE_GDB=1 -MM -MT build/open_esplibs_libmain//misc.o -MF build/open_esplibs_libmain//misc.d /mnt/share/esp-open-rtos/open_esplibs/libmain/misc.c
mkdir -p firmware/
xtensa-lx106-elf-gcc -I./ -I./include -I/home/chiz/share/esp-open-rtos/include -I/home/chiz/share/esp-open-rtos/libc/xtensa-lx106-elf/include -I/home/chiz/share/esp-open-rtos/FreeRTOS/Source/include -I/home/chiz/share/esp-open-rtos/FreeRTOS/Source/portable/esp8266 -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include -I/home/chiz/share/esp-open-rtos/lwip/include -I/home/chiz/share/esp-open-rtos/lwip/include -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/posix -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/ipv4 -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/ipv4/lwip -I/home/chiz/share/esp-open-rtos/lwip/lwip/src/include/lwip -I/home/chiz/share/esp-open-rtos/core/include -I/home/chiz/share/esp-open-rtos/open_esplibs/include  -mlongcalls -mtext-section-literals -DGITSHORTREV=\"22c480b\" -Wall -Wl,-EL -nostdlib  -ffunction-sections -fdata-sections -g -O2 -std=gnu99 -Og -ggdb -DGDBSTUB_FREERTOS=0 -DENABLE_GDB=1 -MM -MT build/open_esplibs_libmain//timers.o -MF build/open_esplibs_libmain//timers.d /mnt/share/esp-open-rtos/open_esplibs/libmain/timers.c
SDK processing stage 1: Removing unwanted objects from /home/chiz/share/esp-open-rtos/lib/libmain.a
cat /home/chiz/share/esp-open-rtos/lib/libmain.a > build/sdklib/libmain_stage1.a
xtensa-lx106-elf-ar d build/sdklib/libmain_stage1.a @build/sdklib/libmain.remove
SDK processing stage 1: Removing unwanted objects from /home/chiz/share/esp-open-rtos/lib/libnet80211.a
cat /home/chiz/share/esp-open-rtos/lib/libnet80211.a > build/sdklib/libnet80211_stage1.a
xtensa-lx106-elf-ar d build/sdklib/libnet80211_stage1.a @build/sdklib/libnet80211.remove
SDK processing stage 1: Removing unwanted objects from /home/chiz/share/esp-open-rtos/lib/libphy.a
cat /home/chiz/share/esp-open-rtos/lib/libphy.a > build/sdklib/libphy_stage1.a
xtensa-lx106-elf-ar d build/sdklib/libphy_stage1.a @build/sdklib/libphy.remove
SDK processing stage 1: Removing unwanted objects from /home/chiz/share/esp-open-rtos/lib/libpp.a
cat /home/chiz/share/esp-open-rtos/lib/libpp.a > build/sdklib/libpp_stage1.a
SDK processing stage 1: Removing unwanted objects from /home/chiz/share/esp-open-rtos/lib/libwpa.a
cat /home/chiz/share/esp-open-rtos/lib/libwpa.a > build/sdklib/libwpa_stage1.a
xtensa-lx106-elf-ar d build/sdklib/libpp_stage1.a @build/sdklib/libpp.remove
xtensa-lx106-elf-ar d build/sdklib/libwpa_stage1.a @build/sdklib/libwpa.remove
SDK processing stage 2: Renaming symbols in SDK library build/sdklib/libmain_stage1.a -> build/sdklib/libmain.a
[... cut ...] ain.a -lesp-gdbstub -lmain -lnet80211 -lphy -lpp -lwpa -Wl,--end-group -o build/Sample-blink.out
/home/chiz/share/esp-open-rtos/ld/program.ld:21: undefined symbol `malloc' referenced in expression
/home/chiz/share/esp-open-rtos/common.mk:198: recipe for target 'build/Sample-blink.out' failed
collect2: error: ld returned 1 exit status
make: *** [build/Sample-blink.out] Error 1
make: Leaving directory '/mnt/share/workspace/Sample-blink'

No where does it reference the sdk library (esp-open-sdk)...

only1chi, 2016-09-23:
I'm still having problems with building and linking esp-gdbstub properly.
I'm building esp-gdbstub and adding it as an extra component, similar to esp-openlibs-libmain.
However I'm getting a linkin error:
"undefined references to debug_saved_ctx"

as it turns out "debug_saved_ctx" is defined as a global variable in gdbstub-entry.S file. It is referenced by gdbstub.c, as an "extern uint32" variable. This looks fine to me, but the linker doesn't seem to know that the variable is a global variable defined in the .S file. 

Does anyone know what I should do next and how to get the linker to see the global variable?

resetnow, 2016-09-23:
@only1chi debug_saved_ctx is defined in core/exception_vectors.S, chances are your esp-open-rtos copy is outdated, so git fetch origin && git rebase origin/master.

only1chi, 2016-09-23:
@resetnow thanks very much for the suggestion. I didn't realize the repository changed.
I was able to compile the gdbstub module after applying your suggestions.
Many thanks.

only1chi, 2016-09-23:
Now I'm getting errors from esptool. I'm honestly don't know what the problem is.

`esptool.py elf2image --version=2 -fs 16m -fm qio -ff 40m build/BlinkTest.out -o ./firmware/BlinkTest.bin
xtensa-lx106-elf-nm: build/BlinkTest.out: no symbols
esptool.py v1.2-dev
Traceback (most recent call last):
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 1233, in <module>
    main()
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 1146, in main
    operation_func(args)
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 898, in elf2image
    image.add_segment(e.get_symbol_addr(start), data)
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 522, in get_symbol_addr
    return self.symbols[sym]
KeyError: '_text_start'
make: *** [firmware/BlinkTest.bin] Error 1
/home/chiz/share/esp-open-rtos/common.mk:209: recipe for target 'firmware/BlinkTest.bin' failed
make: Leaving directory '/mnt/share/workspace/BlinkTest'`

only1chi, 2016-09-23:
I'm also having some problems with esptool. I honestly don't know what the error is referring to. I have also checked esptool.py and I have version 1.2.

``` 
esptool.py elf2image --version=2 -fs 16m -fm qio -ff 40m build/BlinkTest.out -o ./firmware/BlinkTest.bin
xtensa-lx106-elf-nm: build/BlinkTest.out: no symbols
esptool.py v1.2-dev
Traceback (most recent call last):
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 1233, in <module>
    main()
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 1146, in main
    operation_func(args)
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 898, in elf2image
    image.add_segment(e.get_symbol_addr(start), data)
  File "/home/chiz/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 522, in get_symbol_addr
    return self.symbols[sym]
KeyError: '_text_start'
make: *** [firmware/BlinkTest.bin] Error 1
/home/chiz/share/esp-open-rtos/common.mk:209: recipe for target 'firmware/BlinkTest.bin' failed
make: Leaving directory '/mnt/share/workspace/BlinkTest
```

only1chi, 2016-09-24:
As I dug some more I narrowed this latest problem to the linker command. I am building gdbstub into my application, and also linking it as a library. I set-up gdbstub as a component and created a component.mk file for it. 
The contents are as follows:
```

INC_DIRS += $(esp-gdbstub_ROOT)include $(esp-gdbstub_ROOT)

# args for passing into compile rule generation
esp-gdbstub_SRC_DIR = $(esp-gdbstub_ROOT)
esp-gdbstub_INC_DIR = 

ALL_CPPFLAGS += -MMD -MP  $(INCLUDES)
ALL_CFLAGS +=  $(ALL_CPPFLAGS) -mlongcalls -std=c11 -Os -g
#ALL_CFLAGS +=  $(ALL_CPPFLAGS) -mlongcalls -Os -g

esp-gdbstub_CFLAGS = $(CFLAGS) $(ALL_CFLAGS)
esp-gdbstub_ARS = $(AR) -rcs
#esp-gdbstub_LDFLAGS = $(LDFLAGS) -s

$(eval $(call component_compile_rules,esp-gdbstub))
```

My application is a simple blink function. My objective is to get the debugger working. I also have my blink application configured with component.mk file. 

```
PROGRAM=BlinkTest
PROGRAM_SRC_DIR=.
ESP_OPEN_RTOS=/home/chiz/share/esp-open-rtos/
include $(ESP_OPEN_RTOS)common.mk

# Here is how to add subdirectories
PROGRAM_SRC_DIR=.

# Here is how to add extra source files
#PROGRAM_EXTRA_SRC_FILES=./subdir/file.c ./subdir/file.S

# Add include directory
INC_DIRS +=/home/chiz/share/esp-open-rtos/libc/xtensa-lx106-elf/include
EXTRA_INCDIRS +=/home/chiz/share/esp-open-rtos/esp-gdbstub/

# Local make file variables
GDBSTUB_DIR=/home/chiz/share/esp-open-rtos/esp-gdbstub/

# Compiling for Debug
export ENABLE_GDB=1

ifeq ($(ENABLE_GDB), 1)
	# Build for debug or release
	FLAVOR=debug
	LIBS += esp-gdbstub
	LDFLAGS += -L$(GDBSTUB_DIR)lib -s
else
	# Build for debug or release
	FLAVOR=release
endif
```

When I compile as follows, I get the error shown in my previous post. This is because of the line: 
	``` LDFLAGS += -L$(GDBSTUB_DIR)lib -s ```

When I do compile without '-s' option, the application builds correctly but I get the following warning:
``` 
Warning: ELF binary has undefined symbol gdbstub_exceptionStack
Warning: ELF binary has undefined symbol gdbstub_init_debug_entry
```
The symbols its looking for are defined in the gdbstub-entry.S file.

resetnow, 2016-09-24:
@only1chi `-s` strips symbols from the output file, you don't need it that's for sure.

As for warnings, it's a bug — I forgot to remove these symbols, I'll fix it. The debugging will work nonetheless, these symbols aren't referenced anywhere.

Also, feel free to open a PR to esp-gdbstub with component.mk. Unfortunately, esp-gdbstub cannot be merged in e-o-r because of a restrictive license.

only1chi, 2016-09-25:
Thanks @resetnow. Got the debugger working with gdbstub.

commit e264d63e (2015-11-30): Pass linker search paths with -L not -Wl,-L, so they take priority over sysroot

This fixes link-time errors where the toolchain used was esp-open-sdk
built with the default STANDALONE=y (meaning there are Espressif SDK
libs in the sysroot).

Closes #74