SuperHouse/esp-open-rtos #75: 7116 bytes overhead for 2048 Task stack space: maximizing heap by minimizing stacks

yesco, 2015-12-08:
I was a bit unhappy with only having about 25K free when I start my program under esp-open-rtos.

So I started experimenting some. Here is some small steps that doesn't blow up (immediately):
- #define configMINIMAL_STACK_SIZE     ( ( unsigned short )128 )
- #define configTIMER_TASK_STACK_DEPTH  ( ( unsigned short ) 0 )
- removed my user task creating in user_init ("main")<br>
  xTaskCreate(lispTask, (signed char *)"lispTask", 4096, NULL, 2, NULL);

(the defines are in esp-open-rtos/FreeRTOS/Source/include/FreeRTOSConfig.h)

This gives 37K!

It seems that there are a number of system tasks created before user_init is even called, not sure what actual memory they use

experimenting with different value for my user task

task stack | in user_init | cost of stack
----------------| ---------------- | -------------
notask | 37524  | 0
0 |   crash | ?
512 |  34504 | 3020
1024 | 32456 | 5068
2048 | 28360 | 9164
*2048 | 29464 | 8060

*note: this is setting # define configMINIMAL_STACK_SIZE        ( ( unsigned short )2048 )

9164-2048=7116 bytes OVERHEAD for a 2048 bytes stack?

There doesn't seem to be a simple way to increase the stack used by the user_init routine..., other than changing the configMINIMAL_STACK_SIZE but it gives similar overhead?

projectgus, 2015-12-10:
Interesting research! There are at least three system tasks created, at least two - one for LWIP networking and one for the MAC layer, and user_init() itself is a task - I believe it has to be so that some RTOS functions will work, otherwise the scheduler isn't running.

As you know, all of the calls to xTaskCreate() pass in a recommended stack size, and FreeRTOS will allocate the minimum of the recommended stack size and the minimum value.

If the system really is long-term stable with 128 bye stacks then we should probably change the default. We also have the option of editing the xTaskCreates() inside the SDK code in order to shrink some of those stack requested sizes, although obviously this is risky. FreeRTOS measures the stack high water mark and will crash out if a task overflows though, so with some research it should be possible to tweak them while keeping things safe.

I can't explain why varying the size of your single user task stack made such big changes in the overall stack size. If you add an xTaskDelay(100); in the user task before measuring free memory, does anything change?

yesco, 2015-12-10:
I don't need to delay measuring. the value is consistent. I start my lisp
and run some code. the free memory stayed same after.

Without the enlarged stack more more heap is available.

I haven't tested other functions, I guess the lwip task may need more
memory, but there is no way to specify this per system task. 3-4 total
tasks at startup would seem consistent with how stack min size effects heap
left over ...

I'll investigate some deeper into the task creation and stack allocation
code. maybe they used some sizeof(x) that isn't portable. There also seems
to be some hints that alignment matters. But not sure why unless it would
be an MMU thing.

projectgus, 2015-12-10:
The other possibility I just thought of is that the implementation of xPortGetFreeHeapSize() that I implemented in f0b5bc324c is wrong. The more I think about it, the more likely this seems!

You might want to look directly at the output of mallinfo(), sbrk(0) and the supervisor stack pointer (the supervisor stack is the actual main() stack.) There could be (read: probably is) a bug in how we're combining them. There may be other methods for calculating free heap size in newlib that we can use.

Let me know if you'd like a high-level summary of how I understand xPortGetFreeHeapSize() to work at the moment.

One other suggestion, you could try wrapping malloc() and free() at link time (with --wrap in LDFLAGS) and have them output debugging info for each allocation. You'll probably want to defer outputting anything until after you set a flag in your own code, because some malloc() calls happen very early on and will probably crahs otherwise. But you may be able to use that to debug exactly how much heap is allocated as a result of a single call to xTaskCreate().

Angus

bcatalin, 2016-02-09:
I am also trying to have more memory available so changing the values to 128 and 0 didn't worked for me. I have available only 9k after boot and being connected to a WiFi router, which is pretty low for waht I need.

I am open to try other changes.

Noticed that configCPU_CLOCK_HZ is set to 80Mhz right now. If I am setting the clock to 160Mhz in my user_init() so I guess that I need to configure also in esp-open-rtos/FreeRTOS/Source/include/FreeRTOSConfig.h the right value.

projectgus, 2016-02-23:
I just looked at this again and I can't believe I didn't see it to begin with.

Stack size isn't bytes, it's words. The specific type is 'postSTACK_TYPE', which is defined in `portmacro.h` to be unsigned long.

The overheads I'm seeing aren't quite as high as yours, for example if I hack up the blink.c example like this:

```
void user_init(void)
{
    uart_set_baud(0, 115200);
    int a = xPortGetFreeHeapSize();
    printf("About to create task, before heap = %d\n", a);
    xTaskCreate(blinkenTask, (signed char *)"blinkenTask", 256, NULL, 2, NULL);
    int b = xPortGetFreeHeapSize();
    printf("Created 256 word stack task, after heap = %d cost of task = %d\n", b, a - b);
    int c = xPortGetFreeHeapSize();
    printf("About to create second task, before heap = %d\n", c);
    xTaskCreate(blinkenRegisterTask, (signed char *)"blinkenRegisterTask", 2048, NULL, 2, NULL);
    int d = xPortGetFreeHeapSize();
    printf("Created 2048 word second task, after heap = %d cost of task = %d\n", d, c - d);
}
```

Then the output I get is:

```
About to create task, before heap = 41284
Created 256 word stack task, after heap = 40172 cost of task = 1112
About to create second task, before heap = 40172
Created 2048 word second task, after heap = 31892 cost of task = 8280
```

Which implies the overhead for each task (apart from stack) is 88 bytes. This is the TCB - task control block - structure.

I think any other discrepencies you were seeing were maybe due to other tasks manipulating the heap in the background - the binary SDK MAC-layer tasks seem to allocate/deallocate heap.

I think the reason why minimal task stack sizes are so high (128 words aka 512 bytes to do almost nothing!) is that exception handlers currently use the stack of the running process. This seems like it's probably a mistake, they should share a common exception stack space outside of the task - then individual tasks only need as much stack as they will use themselves.

projectgus, 2016-02-23:
I'm going to close this issue. I share your concerns about maximising free RAM, but I think apart from changing the exception handler stacks there isn't much else we can do to streamline task sizes.

However please don't hesitate to open another issue if you notice anything else weird about memory usage.

closed 2016-02-23 by projectgus