SuperHouse/esp-open-rtos #76: malloc doesn't return NULL when run out of heap

yesco, 2015-12-11:
When esp-open-rtos malloc runs out of memory malloc and pvPortMalloc both does not return NULL even if there is no more heap memory available. In conclusion I can allocate over 2M of memory before I get into problem. The pointers are indeed increasing and are indeed "writable" (no  error given), and written value retained. Not sure where the writes end up at all?

Want to test that all the indicated memory by xPortGetFreeHeapSize actually was allocable, this is in relation to the reported issue of https://github.com/SuperHouse/esp-open-rtos/issues/75

Giving back memory beyond the "free memory" seems to crash, however.

    void* a[60];
    int c = 0;

    int t = 0;
    int z = 16384;
    // TODO: keep allocated in free list and deallocate?                                                                                       
    while (z > 0) {
        //while (xPortGetFreeHeapSize() < z) z = z/2;  // enable for safe allocs!                                                                                        
        void* p = pvPortMalloc(z);
        //void* p = malloc(z);                                                                                                                 
        if (!p)
            z = z/2;
        else {
            printf("%u %d %d %d\n", (unsigned int)p, z, t, xPortGetFreeHeapSize());
            t += z;
            // test is writable                                                                                                                
            int* x = p;
            *x = 4711;
            if (*x != 4711) printf("--- Write failed! %d\n", *x);
            a[c++] = p;
            if (c == 60) break; // before it crashes!                                                                                          
        }
    }
    for (int i = 0; i < c; i++) {
        free(a[i]);
        printf("%u %d\n", (unsigned int)a[i], xPortGetFreeHeapSize());
    }

(heap)
1073717248 16384 0 7204
1073733640 16384 16384 -9188
1073750032 16384 32768 -25580
1073766424 16384 49152 -41972
1073782816 16384 65536 -58364
1073799208 16384 81920 -74756
1073815600 16384 98304 -91148
1073831992 16384 114688 -107540
1073848384 16384 131072 -123932
1073864776 16384 147456 -140324
1073881168 16384 163840 -156716
1073897560 16384 180224 -173108
...
1075766248 16384 2048000 -2041796
1075782640 16384 2064384 -2058188
1075799032 16384 2080768 -2074580
1075815424 16384 2097152 -2090972
1075831816 16384 2113536 -2107364
Fatal exception (3):
epc1=0x401065f8
epc2=0x00000000
epc3=0x40221bd6
excvaddr=0x4020240c
depc=0x00000000
excsave1=0x3fff61b0

DagAgren, 2016-01-05:
I know the implementation of sbrk in newlib's libnosys does not do any checks against heap size. It's easy enough to write a better one, for instance I've used this on STM32F4 (with my own custom linker script that provides _heap and _eheap for the start and end of the heap):

	void *_sbrk(int incr)
	{
		extern uint8_t _heap[];
		extern uint8_t _eheap[];
		static uint8_t *heapend=_heap;
	
		if(heapend+incr>_eheap)
		{
			errno=ENOMEM;
			return (void *)-1;
		}
	
		uint8_t *newheapstart=heapend; 
		heapend+=incr;
	
		return newheapstart;
	}

projectgus, 2016-01-06:
Sorry for the silence replying on this issue, @yesco. Busy time of year!

Yes, I think a smarter _sbrk is the way to go. The _sbrk we're using is here:
https://github.com/SuperHouse/esp-open-rtos/blob/master/core/newlib_syscalls.c#L16

There even was stack/heap collision code in there to detect this kind of overtflow, but it's been commented out with a TODO (well done me.) 

(The "stack" in question that we care about is the "supervisor stack" that's running in _start/main, before any RTOS tasks run. The RTOS tasks take their own stack space from the heap.

So the FreeRTOS-friendly way to detect where "stack/heap collision" happens is one of two ways - either we add heap end symbols like in @DagAgren's version above, which fixes the "supervisor stack" to a fixed size. Or we mark the "supervisor stack high water mark" at the time we start the RTOS scheduler (which is the last function call that the supervisor stack ever makes, it never returns), and then in _sbrk we consider the stack pointer to be _either_ the current stack pointer (if _sbrk was called outside an RTOS task) or the supervisor high water mark (if _sbrk was called within an RTOS task stack.)

foogod, 2016-01-06:
I haven't looked at this in detail, so I may be missing something, but since, as you say, the scheduler never returns, and if I remember, there's no way for stack-local variables to be passed into the scheduler, why do we care about the stack at all after that point?  No need to record a high water mark at all, just consider the whole supervisor stack to be free memory after we start the scheduler..

One potential issue with your second approach in cases where people are doing significant work outside of the RTOS scheduler is that it doesn't guarantee any minimum amount of stack space, so it would be possible for somebody to successfully allocate a bunch of memory, and then make a call to another function which uses some more of the stack and in the process overwrites the malloc'd memory with the (now larger than when sbrk checked) stack..

I'm thinking a hybrid of the two: reserve a fixed (compile-time or even run-time configurable) size for the supervisor stack, but once we're in the RTOS, ignore that and consider the supervisor stack to be free memory..

projectgus, 2016-01-06:
Just clobbering the supervisor stack made me deeply uncomfortable at first thought, but I think you're right that it should work fine.

The hybrid approach sounds good, I like that we can have a reasonably large default stack size but then totally overwrite it once the scheduler is running. Making it runtime modifiable should be a simple implementation - it can just be a global variable that the scheduler sets to zero as it starts.

yesco, 2016-03-01:
I consider this bug quite serious.

I have a feeling from the discussion that we're waiting for the ultimately and best solution instead of having a good enough one and improve it later. ;-)

I'm not sure why we just not enabled the check code? It was said it was commented out. Is it faulty/incorrect, dangerous? Any stacks allocated later are allocated from the heap (at least according to some docs I read), so the issue remains in knowing where the heap stops. There is a discussion regarding the supervisor thread that I don't follow completely as I'm new to rtos world, it seems indicated that the

But that contract of malloc not returning null when out of memory is not honored is quite serious and may well be the source of most instabilities observed by people trying to use esp-open-rtos. For example it causes networking to crash without any good explanation as seen in #101. Actually, any rtos app with networking enabled would easily be crashed by just doing enough connections in short time...

>> There even was stack/heap collision code in there to detect this kind of overtflow, 
>> but it's been commented out with a TODO (well done me.)

projectgus, 2016-03-01:
Hi yesco,

It's fixed in the 'feature/tests' branch already (I even have unit tests in there to prove it!) I was hoping to merge the tests after I had them a bit more fleshed out - but it's dragging on a bit so perhaps I'll merge what I have now.

yesco, 2016-03-01:
OK. I missed that. Great!

projectgus, 2016-05-27:
Sorry this dragged on for so long, the feature/tests branch is still a work in progress so I cherry-picked this one fix into master.

commit f0db2660 (2016-05-27): brk/malloc: Allow malloc to fail when out of RAM

Fixes #76.
closed 2016-05-27 by projectgus