SuperHouse/esp-open-rtos #78: Deep sleep ignores argument

stschake, 2015-12-24:
*sdk_system_deep_sleep(us)* is supposed to go into deep sleep for *us* microseconds, but instead it always sleeps 1073s ~ 18 minutes. I assume this is due to an error in the SDK, but since we're trying to get rid of that anyway, I thought I'd document my findings.

*sdk_system_deep_sleep* does not actually put the system into deep sleep mode. Instead, it first disables wifi and then arms a timer that upon expiration calls the actual deep sleep method at .irom0.text+0x734. The actual sleep method will output "deep sleep {1}s" on the serial port before continuing, with {1} being the time in seconds in deep sleep as calculated from the argument. Between setting the timer function argument and the timer function being called, the  correct argument value is lost and we get an output of "deep sleep 1073s".

To get around this, right now the easiest way is to simply call the  *actual* deep sleep function at .irom0.text+0x734 which has the same prototype and parameter semantics as system_deep_sleep.

projectgus, 2015-12-29:
Thanks for reporting this, it's kind of a weird bug!

I suspect the reason for the timer is to allow wifi MAC level (and maybe IP level) buffers to flush, so there aren't packets that were "sent" before going to deep sleep but don't get sent until after. Maybe also to allow the WiFi shutdown to complete before the CPU disables.

I'd be a little suspicious that going immediately into deep sleep might have unexpected consequences around either power consumption or potential network data loss.

@foogod has a full reverse engineered implementation of the SDK "timer" framework in the open-libmain branch, so if/when we move there we can probably fix the bug with passing arguments via the SDK timer framework (assuming that's where the bug occurs.)

foogod, 2015-12-29:
I'm not sure the timer.o stuff I RE'd will really help, as I think this process actually uses some lower level hardware timer stuff (wdev timers et al) instead.  However, I am about 75-80% done reverse-engineering user_interface.o (I got a bit sidetracked with the holidays and all) which I suspect may be useful here, as I do think I remember seeing some code that looked to be doing some of this in there.  I'll have to go back through what I have thus far and take a closer look..

I suspect the main reason for the convoluted invocation is probably to make sure that we don't shut down the network hardware in the middle of (for example) transmitting or receiving a packet, resulting in corrupted/lost network traffic..  There may also be some issues with making sure interrupts are properly disabled, etc, so that the deep-sleep process doesn't potentially get interrupted part way through at a very bad time..

pathob, 2016-05-12:
> To get around this, right now the easiest way is to simply call the actual deep sleep function at .irom0.text+0x734 which has the same prototype and parameter semantics as system_deep_sleep.

@stschake How would one technically do that?

DeuxVis, 2016-05-12:
I'd use a function pointer.

pathob, 2016-05-16:
Ok, sorry for more spamming into this issue, but...

What is .irom0.text+0x734? Does it has to do with the line I found in ld/nonota.ld?

projectgus, 2016-05-16:
The address of the function is relative to user_interface.o in libmain.a, so you have to find it relative to there. (These functions contains "symbols" at particular addresses,  but this function was declared "static" so it doesn't get a symbol for external callers). It is possible to derive the address of this function, relative to some other function in the compilation unit. It's just fiddly.

However - there's actually a new approach possible for this now. @foogod has reconstructed the internal function itself, it's here:
https://github.com/SuperHouse/esp-open-rtos/blob/open-libmain/open_esplibs/libmain/user_interface.c#L359

So I think probably the best/easiest thing you can do is copy that function into your code and call it directly. I haven't tested this, but I'd be very interested to hear how it goes.

sheinz, 2016-10-21:
Duplicate of https://github.com/SuperHouse/esp-open-rtos/issues/176
Fixed with https://github.com/SuperHouse/esp-open-rtos/pull/190

closed 2016-10-21 by sheinz