SuperHouse/esp-open-rtos #80: Update to FreeRTOS 8

DagAgren, 2016-01-05:
Has anyone looked at how difficult it would be to update to FreeRTOS 8? Personally I much prefer it, just because the various type names are not such an ugly mess any more.

I might have a look at it later if nobody has tried it yet, if I can find the time.

projectgus, 2016-01-05:
I've thought about it, but I have not looked into it yet. I'd be very interested to see how you go, and I'd be keen to merge any working update (slash help with debugging).

The main thing that was holding me back is that I think a tiny number (one or two) FreeRTOS symbols  are still in the binary libraries. But it's the kind of thing that should be easy to pull out.

DagAgren, 2016-01-06:
Remember which, or how to find them?

Right now I am trying to decide between implementing a project on ESP8266, or EMW3165. The latter has much nicer hardware than the mess on the ESP8266, but it has no equivalent to the esp-open-rtos project so you are stuck with terrible vendor libraries... Been trying to see if I can sort it out a bit but it doesn't look too promising.

Then again, I need both WiFi and some simnple audio output, and while you can abuse the I2S on the ESP to do it, it seems the pins are quite noisy and it doesn't work too well. PWM audio on the STM32F4 core in the EMW3165 is much easier and more reliable... Sigh.

projectgus, 2016-01-06:
I couldn't remember but I took a look through a build/program.map file to jog my memory. It's not as bad as I remembered, `vPortYield` in libmain.a is I can see and it's probably safe to leave alone.

The other pain point that I can see with porting is anywhere the SDK libraries call into FreeRTOS functions, if the signatures/names have changed. This is where it'll be good once we have all of the SDK libraries available at least as plain assembly files, so you can just edit the function calls. The `feature/assemble_sdk` branch has a start I made on this but it doesn't work yet.

Regarding microcontroller choice, that's a hard one. The other option, I guess, would be to use esp8266 plus a second micro/chip to do the audio over SPI or I2S. Probably still cheaper than EMW3165, but more fiddly to assemble.

DagAgren, 2016-01-07:
Supposedly FreeRTOS 8 is backwards compatible, so in theory it shouldn't be a problem. Main big visible changes are renaming types and using stdint.h, but that should leave the actual signatures the same.

Wish somebody sold an I2S codec on a convenient little module, that would be fairly easy to use from the ESP8266, since I got I2S working. Not feeling like making circuit boards though.

foogod, 2016-01-08:
Yeah.. unfortunately it looks like SparkFun used to make an I2S DAC breakout, but they discontinued it..  it's actually rather dismaying how little hobby-level support there seems to be out there for I2S in general at the moment..  (mind you, if somebody wants to make up some PCBs for a decent I2S chip, I might be interested in buying a few as well, to help offset the cost :) )

Alternately, it's not well known, but the ESP8266 does actually have a hardware PWM generator built in (which is completely unused by the Espressif SDK's fake "software PWM" routines, but is accessible via hardware registers directly).  I've never tried using it, but it might be worth looking into if you wanted a cleaner PWM audio-out from the ESP8266..

DagAgren, 2016-01-08:
I looked at it. Seems it's 8 bits and supposedly some kind of delta-sigma... That should work for simple audio output, but without DMA-driven output it might not be that useful for audio. It might be possible to do DMA to it for all I know, but the SLC modes don't seem to be documented at all, other than that mode 1 is I2S.

recursify, 2016-01-19:
Would also be interested in updating FreeRTOS, even if version 8 is a no-go for some reason.

v7.5.3 adds the uxQueueSpacesAvailable() to the Queue API, which I'm keen to use :)

recursify, 2016-02-17:
So quick update: I was able to update to FreeRTOS 8.2.3, and have it compile.  I had to add these lines to portmacro.h

```
+#define StackType_t             portSTACK_TYPE
+#define BaseType_t              portBASE_TYPE
+typedef uint32_t TickType_t;
+#define UBaseType_t             portBASE_TYPE
```

as well as change a few casts with xTaskCreate from `(signed char *)` to `(const char *)`, in both `core/app_main.c` and `lwip/sys_arch.c`

I don't have my board with me now, but I will try running it tomorrow.  If it works, it'd be great to get it in a topic branch so other's can try it out.

@projectgus - is `lwip/sys_arch.c` something you're okay modifying, or do you want to leave it untouched?

projectgus, 2016-02-17:
Awesome work, thanks @recursify! Keen to hear how it holds up.

Regarding portSTACK_TYPE & friends, if the new "correct" way is StackType_t/etc, then can we just replace all instances of those types with the newer types? (On a personal note I've always hated that portXXX naming style, anyhow.)

If there are concerns about breaking compatibility then maybe we can #define the other way (ie #define portSTACK_TYPE StackType_t) for backwards compatibility, and remove those a future major update?

`lwip/sys_arch.c` is fine to modify, it's glue code from "us" to lwip but it belongs to "us" - so go for it.

recursify, 2016-02-17:
Looks like a bit more work is needed.  Connecting to a WiFi network failed, and I started getting a bunch of these messages on the UART

```
Timer Stop Failed
Timer Start Failed
```

Anyway, don't have time to dig in right now.  But if others are interested, I can create a fork with my in-progress changes?

projectgus, 2016-02-17:
Awww, :(.

A fork would be great, thanks - save on duplication of effort if someone else wants to pick it up.

leopck, 2016-05-22:
Hi, I'm not sure how this would help, but I attempted to update to v8.2.3 as well; I'm not an expert in this but I just thought of giving this a shot. 

Continuing from @recursify work, I attempted the same steps as he did and tried to repair the compilation as I go along and adjusting some lines of codes inside the FreeRTOS/Source. I do not see the messages on the UART as he did regarding the Timer Stop Failed. In fact, I could compile and run example/blink.c and example/serial_echo.c but WiFi network is failing in mine as well.

This is my output for blink:
```
ESP-Open-SDK ver: 0.0.1 compiled @ May 22 2016 00:27:38
phy ver: 273, pp ver: 8.3

mode : sta(18:fe:34:a5:92:78)
add if0
```
usually it'll start scanning for my SSID and join my network; but in my case it just stays at add if0 but the LED continues to blink as usual.

I've also jot my step-by-step for those that wants to replicate this:
 ``` 
1. git clone esp-open-rtos //you dont want to be using your current rtos working folder
2. Download from below the FreeRTOS8.tar.gz
3.Copy then paste from the content of FreeRTOS8.tar.gz (FreeRTOS/Source/) into esp-open-rtos/FreeRTOS/Source
4. Change the SSID and git ignore the SSID file
5. Go to esp-open-rtos/FreeRTOS/Source/portable/esp8266/portmacro.h and add:

++ #define StackType_t             portSTACK_TYPE
++ #define BaseType_t              portBASE_TYPE
++ typedef uint32_t TickType_t;
++ #define UBaseType_t             portBASE_TYPE
++#define portTICK_PERIOD_MS                    ( ( portTickType ) 1000 / configTICK_RATE_HZ )

6. Go to esp-open-rtos/FreeRTOS/Source/portable/esp8266/portmacro.h and remove:
-- typedef uint32_t portTickType;
-- #define portTICK_RATE_MS                   ( ( portTickType ) 1000 / configTICK_RATE_HZ )

7. Go to /path/esp-open-rtos/lwip/sys_arch.c:517 and modify the (unsigned char ) --> (const char)

8. esp-open-rtos/core/app_main.c:389 and modify the (unsigned char ) --> (const char)

9. **For blink example**
Go to example/blink/blink.c and modify the (unsigned char ) --> (const char)
```

So far, that's all I know how to do for the porting and I do not have any ideas on how to debug the WiFi issue sadly :(

This is the FreeRTOS8 folder that contains some minor changes, just giving IRAM to the functions.

[FreeRTOS8.tar.gz](https://github.com/SuperHouse/esp-open-rtos/files/276219/FreeRTOS8.tar.gz)

sheinz, 2016-11-25:
FreeRTOS has been upgraded to v9.0.0

closed 2016-11-25 by sheinz