SuperHouse/esp-open-rtos #81: Includes not found

SaimenSays, 2016-01-07:
Is there a bug in common.mk?
I'm unable to compile because nearly all include directory is not referd. Whatever example I try to compile, there is always the message:
```
   espressif/esp_common.h: No such file or directory
    #include "espressif/esp_common.h"
```
But header file is in correct directory in respect to git checkout root directory. If I change $ROOT in common.mk to have correct include directory, the error changes. Because now compiler is unable to find specific mk files anymore. Maybe some issue with version of make?

Please help!

projectgus, 2016-01-07:
Hmm, that's odd!

Can you please run 'make V=1' (which will output the full gcc command lines), and post ithe output to pastebin.com or similar? Thanks!

projectgus, 2016-01-07:
FWIW the build systems builds all examples on every commit, and it's building right there. So it might be a make/gcc version issue. Can you also post the output of `make -v` and `xtensa-lx106-elf-gcc -v`, please?

SaimenSays, 2016-01-12:
Finally I'm able to fix it. I moved repo from my home directory to \opt and the error was gone, don't know why.
Then another error occured: It tells me C++ compiler is not available. But with the help of your command  ```xtensa-lx106-elf-gcc -v``` I found out that the wrong compiler from older Espressif sdk was used. After manually changing symbolic links in ```\usr\bin``` to new sdk, everything works now.

Thanks anyway!

closed 2016-01-12 by SaimenSays

projectgus, 2016-01-12:
Glad you got it sorted out :)