SuperHouse/esp-open-rtos #82: Support for 160MHz CPU frequency

projectgus, 2016-01-17:
As reported on IRC, there's no function prototype for sdk_system_cpu_freq(uint8_t freq) in the headers, although the function exists in the libraries.

esp-open-rtos should either implement a prototype for that function, or (better) write a new implementation that sets the appropriate register.

elmorejd, 2016-01-18:
That solved the issue. Added "void sdk_system_update_cpu_freq( uint8_t )" to "espressif/esp_system.h" which seemed to be the most accurate place to put it. I will keep you updated if anything wonky happens.

projectgus, 2016-02-28:
These functions are now part of `espressif/esp_system.h`.

Leaving this issue open as it'd be good to have a from-scratch reimplementation of CPU frequency setting and the relevant registers.

sheinz, 2016-11-25:
I'll close it as `sdk_system_update_cpu_freq`  is reimlemented [here](https://github.com/SuperHouse/esp-open-rtos/blob/master/open_esplibs/libmain/user_interface.c#L429)

closed 2016-11-25 by sheinz