SuperHouse/esp-open-rtos #86: ESP is not able to work in station mode after softAP.

hetii, 2016-02-07:
Hello.

After configure esp as AP like below code I was not able to back to station mode 
(for egg by using htttp_get example).

The only way how I restore this functionality was to burn nodemcu and then after setting connectivity to my access point I was able to run http_get example.

`    sdk_wifi_set_opmode(SOFTAP_MODE);
    //sdk_station_config
    struct sdk_softap_config apconfig;

    if(sdk_wifi_softap_get_config(&apconfig)){     
        
        strncpy((char *)apconfig.ssid, "my_nice_ssid", 32);
        strncpy((char *)apconfig.password, "some_secret_password", 32);
        
        apconfig.authmode = AUTH_WPA2_PSK;
        apconfig.ssid_hidden = 0;
        apconfig.max_connection = 4;
        apconfig.channel=7;

        if(!sdk_wifi_softap_set_config(&apconfig)){
            printf("ESP8266 not set ap config!\r\n");
        }
    }
    struct ip_info info;
    IP4_ADDR(&info.ip, 192, 168, 22, 1);
    //IP4_ADDR(&info.gw, 192, 168, 22, 1);
    IP4_ADDR(&info.netmask, 255, 255, 255, 0);`

projectgus, 2016-02-07:
Wow, that's an unexpectedly big problem!

Running `esptool.py erase_flash` should get you back to an empty ESP flash in any case, and remove any WiFi settings that were helpfully saved by the binary SDK.

However this is still a bug! I'll try to reproduce over the next couple of days.

alainmaes, 2016-05-23:
You enable the AP and that config is remembered. To avoid that you could use sdk_wifi_softap_set_config_current so the ESP does not remember this after a reboot/reflash.

Then when you enable station in another firmware based on esp-open-rtos you end up in station+softAp mode. And in that mode there is a bug that causes the AP to become the default interface causing all outgoing connections to go through there.

In core/app_main.c:
    if (sdk_g_ic.s.wifi_mode == 3) {
        sdk_wifi_station_start();
        sdk_wifi_softap_start();
        netif_set_default(sdk_g_ic.v.softap_netif_info->netif);
    }

The above should become:
    if (sdk_g_ic.s.wifi_mode == 3) {
        sdk_wifi_station_start();
        sdk_wifi_softap_start();
        netif_set_default(sdk_g_ic.v.station_netif_info->netif);
    }

I noticed connection to an MQTT server failed as soon as I set up the AP. So I started digging. For me applying the cange did the trick.