SuperHouse/esp-open-rtos #95: printf doesn't support floats

foogod, 2016-02-22:
It's my understanding that newlib has the option of supporting floats in `printf`, etc, but it doesn't appear that esp-open-rtos' `printf` can print them (`%f` just prints as `f`).  Are we disabling this in our newlib build?

The ESP8266 is powerful enough (and the core libs have the necessary functions) that we don't have to be limited to just integers for everything and floats can sometimes be very useful.  We should support this behavior, if we can.

projectgus, 2016-02-22:
I did not know this until just now, but nano mode newlib leaves out floating point support from printf/scanf unless you ask for it at link time:

https://github.com/projectgus/newlib-xtensa/blob/xtensa-2_2_0-lock/newlib/README#L375

I guess we should either document this prominently or enable it by default.

The only concern I have with enabling it by default is that printf is an IRAM function, because it gets called for all debug output from the SDK - including NMI routines and early init routines that run while flash is unmapped.

If adding float support consumes a lot more IRAM space, we might not want to do it until we have #58 and can easily rewrite the "risky" SDK routines so they don't call it directly. However it might not consume much more space, I haven't measured.

foogod, 2016-02-22:
Ah, so it sounds like all we need to fix this is to change our link flags in the makefiles?  That's easier than I expected..  My preference would probably be:

1. Make this a configurable setting
2. Make it enabled by default
3. Document that if people need more IRAM space, they might try turning off floating point for printf, and "here's how you do it"..

My feeling is (particularly with how much of the rest of the stuff we're moving into IROM) the extra IRAM usage probably won't be a problem for most people, and if it becomes a problem, people can make the choice whether this is functionality they need or not..  (of course, I haven't looked at exactly how much bigger it actually is)

If the extra IRAM usage does turn out to be a big concern, we could alternately just change all the things that can't use IROM functions to call `iprintf` instead, and have `_printf_float` be in IROM for the app to use.. this has the drawback of duplicating a bunch of the printf code between IRAM and IROM, but does not add anything to the IRAM usage over the current state..

foogod, 2016-02-22:
Ok, so I actually just tried building something with -u _printf_float as described in the newlib docs and it does not seem to work (it just gets left with an undefined symbol in the result instead of pulling in the float implementation as it's supposed to).

I suspect this may be because (according to https://github.com/SuperHouse/esp-open-rtos/wiki/libc-configuration) in addition to the nano mode, our newlib was also built with `--disable-newlib-io-float`, which means that it doesn't even include the weak routines in the result, so there's nothing in the libs to actually pull in.

Presumably with `--enable-newlib-nano-formatted-io` we should not also be using `--disable-newlib-io-float`..

foogod, 2016-02-22:
I also realized by reading the nano docs more closely, that iprintf is no longer separate from printf, so what I was suggesting about putting one in IRAM and one in IROM probably won't work if we're using the nano build..  oh well.

projectgus, 2016-02-22:
Doh, I should have remembered those additional configure args.

See how that commit works out. Turns out I was wrong about the IRAM overhead. The deal is:

* Compiling libc with weak _printf_float support increases IRAM footprint by 152 bytes. This overhead occurs regardless of whether you actually link in _printf_float or not, it's the overhead caused by checking if the symbol is non-zero and calling it. The bulk of the floating point support lives in the flash only, which is good.
* If you link in _printf_float then the irom0 section grows by 8096 bytes and the .rodata section grows by 316 bytes.

I was almost going to just hard-enable, but .rodata currently comes out of data RAM and 316 bytes is just enough to maybe care. So it's a makefile flag, enabled by default.

Once we have #11 closed (which depends on #58 for a clean implementation, which depends on #26 for regression tests) then that .rodata can all go into flash, and I think it'll be worth simply enabling it and removing the build option.

Anyhow, let me know what you think.

foogod, 2016-02-23:
That looks much better (at least it compiles/links OK now..  I don't have an ESP8266 to test with at the moment but will give it a run when I get home)..

I did make one minor comment on the commit too..  Will let you know the results once I've fully tried it out.

projectgus, 2016-02-23:
Fixed the _scanf_format thing and updated the commit. I haven't verified scanf now works with floats the same as I did with printf, but the binary size went up so something happened!

Cost of enabling floating point for both is of course higherthan what I reported - 14832 bytes flash, 448 bytes rodata.

foogod, 2016-02-23:
Hmm.. if that's the case, maybe we should separate them out to separate config options then.. I could see needing floats for `printf` being a lot more common than needing/wanting them for `scanf`..

Although if you don't even reference `scanf`, are the size increases still the same?  I guess it's not as big a deal if it only affects apps which use `scanf` to begin with (which probably won't be that many, and odds are if somebody's using it that they're doing some more sophisticated input handling (like reading floats) anyway)..

projectgus, 2016-02-23:
You get the size increase whether you use scanf or not, because passing -u _scanf_float to the linker forces it to include the symbol. Which is annoying! I can't think of a way around that apart from patching newlib to not mark the symbols as weak.

foogod, 2016-02-23:
Ok, well I've confirmed everything works in my tests.. I even verified that `scanf` will correctly read floats with the new changes as well..

(I did discover when I built with `PRINTF_SCANF_FLOAT_SUPPORT=0` that my esptool.py bombed out horribly, however..  I see you made an update to https://github.com/themadinventor/esptool earlier today to address this?  We should probably warn people somewhere that they'll need a new esptool.py to use this feature..)

Regarding the scanf stuff, I'm actually sorta torn.  I think `printf` is likely enough to be used from time to time that it warrants having it enabled by default, but I really doubt `scanf` is ever going to be used much by anybody, except possibly once or twice, and even those rare cases may not even need float support.  Given this I'm sorta inclined to say we should enable `printf` floats by default, but disable `scanf` floats by default (and let people enable it if they need it).  However, part of me also doesn't like the inconsistency of that and thinks we should default to either both enabled or both disabled..

projectgus, 2016-02-23:
Going point re:esptool, I'll update the commit with a note in the comment. If we disable any support by default then everyone will need that version of esptool, as well.

Regarding disabling scanf, I was thinking about the same conflict. The additional 7KB of flash to have scanf is probably something noone will notice unless it's pointed out (most people have at least half a megabyte of flash, most esp-open-rtos programs are in the 200-350KB range right now). An extra 132 bytes of RAM is annoying, but all of that should be moving to flash as well - sooner or later. I'm OK with leaving everything enabled by default for ease of compilation for now, with the understanding it'll get less resource-heavy in time.

foogod, 2016-02-23:
Yeah, I guess that makes the most sense.. especially since not having both enabled by default will suddenly break everyone who doesn't have the brand new esptool.py, we should just enable them both by default (at least for now)..

OK, I think I'm good with the current feature/float-io resolution if you're going to make that update to the comment; feel free to merge into master as far as I'm concerned :)

commit 6aff1e33 (2016-02-23): libc: Recompile newlib to enable float & long double i/o in printf/scanf

Enabled by default, set PRINTF_SCANF_FLOAT_SUPPORT=0 in local.mk/Makefile to disable

newlib-xtensa revision ae10be3

Configure/compile steps:

../configure --with-newlib --enable-multilib
--disable-newlib-io-c99-formats --enable-newlib-supplied-syscalls --enable-target-optspace --program-transform-name="s&^&xtensa-lx106-elf-&" --disable-option-checking --with-target-subdir=xtensa-lx106-elf
--target=xtensa-lx106-elf --prefix=/home/gus/dev/esp/rtos/open-rtos/libc/ --enable-newlib-nano-malloc --enable-newlib-nano-formatted [... cut]
commit 19b0a19a (2016-02-23): libc: Recompile newlib to enable float & long double i/o in printf/scanf

Enabled by default, set PRINTF_SCANF_FLOAT_SUPPORT=0 in local.mk/Makefile to disable

newlib-xtensa revision ae10be3

Configure/compile steps:

../configure --with-newlib --enable-multilib
--disable-newlib-io-c99-formats --enable-newlib-supplied-syscalls --enable-target-optspace --program-transform-name="s&^&xtensa-lx106-elf-&" --disable-option-checking --with-target-subdir=xtensa-lx106-elf
--target=xtensa-lx106-elf --prefix=/home/gus/dev/esp/rtos/open-rtos/libc/ --enable-newlib-nano-malloc --enable-newlib-nano-formatted [... cut]
closed 2016-02-23 by projectgus