SuperHouse/esp-open-rtos #96: Crash on boot

bcatalin, 2016-02-23:
First, thanks for the good work on this repo. Second. I have a crash in 

 xtensa-lx106-elf-addr2line -f -e build/simple.out 0x402363e7
memchr
/home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/string/../../../../../newlib/libc/string/memchr.c:133

at the boot time.  After 5 to 10 reboots is ok and start to boot correctly.

It is anything to avoid this type of crash. 

Also the esp is crashing with no heap. I've monitored all my malloc calls so I am printing every time I am allocating or deallocating something. The null heap is happening after 4 to 5 minutes after I am done with allocating memory. Basically I just stay in a loop waiting for clients.

` ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 32124, room 16 
tail 12
chksum 0xf8
ho 0 tail 12 room 4
load 0x3ffe8000, len 2196, room 12 
tail 8
chksum 0xd0
load 0x3ffe88a0, len 22468, room 0 
tail 4
chksum 0x20
csum 0x20
{gonc#ogbr{o$d`oson'xcc'oc{;g$#sc;#}3fff9bf8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 23 2016 13:51:45
phy ver: 273, pp ver: 8.3

data  : 0x3ffe8000 ~ 0x3ffe8894, len: 2196
rodata: 0x3ffe88a0 ~ 0x3ffee064, len: 22468
bss   : 0x3ffee068 ~ 0x3fff6508, len: 33952
heap  : 0x3fff6508 ~ 0x3fffc000, len: 23288
mode : softAP(5e:cf:7f:81:c7:16)
add if1
bcn 100
[simple.c] START WIFI TASK
bcn 0
del if1
usl
sul 0 0
mode : sta(5c:cf:7f:81:c7:16)
add if0
WiFi: Connecting to WiFi
scandone
add 0
aid 4
cnt 

connected with guest, channel 1
dhcp client start...
ip:94.177.7.10,mask:255.255.254.0,gw:94.177.7.254
Fatal exception (9): 
epc1=0x402363e7
epc2=0x00000000
epc3=0x4022a289
excvaddr=0x888fc20a
depc=0x00000000
excsave1=0x40235abe

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 32124, room 16 
tail 12
chksum 0xf8
ho 0 tail 12 room 4
load 0x3ffe8000, len 2196, room 12 
tail 8
chksum 0xd0
load 0x3ffe88a0, len 22468, room 0 
tail 4
chksum 0x20
csum 0x20
{on'cc'o#;sodl`osn'gxccgncs{gdcrc{c}3fff9bf8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 23 2016 13:51:45
phy ver: 273, pp ver: 8.3

data  : 0x3ffe8000 ~ 0x3ffe8894, len: 2196
rodata: 0x3ffe88a0 ~ 0x3ffee064, len: 22468
bss   : 0x3ffee068 ~ 0x3fff6508, len: 33952
heap  : 0x3fff6508 ~ 0x3fffc000, len: 23288

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 32124, room 16 
tail 12
chksum 0xf8
ho 0 tail 12 room 4
load 0x3ffe8000, len 2196, room 12 
tail 8
chksum 0xd0
load 0x3ffe88a0, len 22468, room 0 
tail 4
chksum 0x20
csum 0x20
so'gcc'oc{rnll nr'gopbcg'br{olc;c{c}3fff9bf8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 23 2016 13:51:45
phy ver: 273, pp ver: 8.3

data  : 0x3ffe8000 ~ 0x3ffe8894, len: 2196
rodata: 0x3ffe88a0 ~ 0x3ffee064, len: 22468
bss   : 0x3ffee068 ~ 0x3fff6508, len: 33952
heap  : 0x3fff6508 ~ 0x3fffc000, len: 23288
mode : softAP(5e:cf:7f:81:c7:16)
add if1
bcn 100
[simple.c] START WIFI TASK
bcn 0
del if1
usl
sul 0 0
mode : sta(5c:cf:7f:81:c7:16)
add if0
WiFi: Connecting to WiFi
scandone
add 0
aid 4
cnt 

connected with guest, channel 1
dhcp client start...
ip:94.177.7.10,mask:255.255.254.0,gw:94.177.7.254
Fatal exception (28): 
epc1=0x402363e7
epc2=0x00000000
epc3=0x4022a229
excvaddr=0x00000024
depc=0x00000000
excsave1=0x40235abe

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 32124, room 16 
tail 12
chksum 0xf8
ho 0 tail 12 room 4
load 0x3ffe8000, len 2196, room 12 
tail 8
chksum 0xd0
load 0x3ffe88a0, len 22468, room 0 
tail 4
chksum 0x20
csum 0x20
;goo#bogcs{g$d`g{oo'8c#noc{r'$bs#rb}3fff9bf8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 23 2016 13:51:45
phy ver: 273, pp ver: 8.3

data  : 0x3ffe8000 ~ 0x3ffe8894, len: 2196
rodata: 0x3ffe88a0 ~ 0x3ffee064, len: 22468
bss   : 0x3ffee068 ~ 0x3fff6508, len: 33952
heap  : 0x3fff6508 ~ 0x3fffc000, len: 23288

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 32124, room 16 
tail 12
chksum 0xf8
ho 0 tail 12 room 4
load 0x3ffe8000, len 2196, room 12 
tail 8
chksum 0xd0
load 0x3ffe88a0, len 22468, room 0 
tail 4
chksum 0x20
csum 0x20
son'cc'o#;sndl`nsn'gxbcgncs{odcrc{c}3fff9bf8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 23 2016 13:51:45
phy ver: 273, pp ver: 8.3

data  : 0x3ffe8000 ~ 0x3ffe8894, len: 2196
rodata: 0x3ffe88a0 ~ 0x3ffee064, len: 22468
bss   : 0x3ffee068 ~ 0x3fff6508, len: 33952
heap  : 0x3fff6508 ~ 0x3fffc000, len: 23288
mode : softAP(5e:cf:7f:81:c7:16)
add if1
bcn 100
[simple.c] START WIFI TASK
bcn 0
del if1
usl
sul 0 0
mode : sta(5c:cf:7f:81:c7:16)
add if0
WiFi: Connecting to WiFi
scandone
add 0
aid 4
cnt `

projectgus, 2016-02-23:
Hi,

Thanks for the detailed bug report and crash dumps.

- I think the crash at startup may be due to adding printfs to malloc/free. I had some problems with this caused by some mallocs() that happen very early from the binary SDK libraries. If you set a flag from inside user_init() and only printf from malloc/free if the flag is set, the problem may go away.

- The 'overflow the heap size' message is copied from Espressif's SDK and it's misleading, we should fix it. It actually means "task stack overflow", the task has overflowed its stack - ie the problem is stack not heap. Fixing this may be as simple as increasing the task stack size passed to `xTaskCreate`. 

Something odd is that the task name "@?D??y"@H>@" printed by the error looks like it's been corrupted, this might be indicative of more serious memory corruption (or it might just be the same stack overflow corrupted it.)

bcatalin, 2016-02-25:
Thanks for the tips. I'll try them.

sheinz, 2016-11-25:
Is there any new info on this issue? I will close it for now. Feel free to reopen if it is relevant.

closed 2016-11-25 by sheinz