SuperHouse/esp-open-rtos #101: Access point example leaks memory for each connection

pellepl, 2016-02-27:
Thanks for a great project! Stumbled into a problem with the access_point example, commit  a32d109, no modifications.

If I repeatedly connect via telnet, the free heap shrinks with ~120 bytes for each connection. Crash when reaching zero. Haven't looked into it much yet, guessing something in lwip is leaking. Or perhaps it is even expected..?

projectgus, 2016-02-27:
Thanks! Hope you find it useful.

I think I've seen this with TCP sockets being stuck in CLOSE_WAIT inside LWIP. If you leave it a little while (I think maybe 60 seconds) then you should find the heap usage goes back to normal.

However, it shouldn't crash in any case! It might be related to #76, in which case merging the `feature/tests` branch on top of current master may fix it (there's a fix for #76 in that branch.)

projectgus, 2016-02-28:
There's also come code in the LWIP stack that's supposed to prune TIME_WAIT sockets if memory gets tight:
https://github.com/SuperHouse/esp-lwip/blob/3cf8d514bd76e6ef77e6fa514d0ec6d96da7fd9a/src/core/tcp.c#L1288

Maybe that threshold value needs to be higher? Although again it shouldn't outright crash on OOM I would hope (but that's probably the malloc-never-fails issue causing it to allocate outside of accessible RAM.)

pellepl, 2016-03-01:
Confirmed - after 60 secs the memory is released. And yes, the crash is most certainly a side-effect of #76. 

However, the pruning does not seem to work. Anyhow, for me this is not critical. It might be a security aspect for some if it is this easy to crash it just by connecting to it x times within a minute. I'll try this test out with the #76 fix when I can. Closing for now - thanks for the explanation.

projectgus, 2016-03-01:
Thanks @pellepi for confirming.

I'm going to leave this open though, as I think it's a serious enough problem we should find a fix.

pellepl, 2016-05-31:
As of the latest and greatest (with #76 in), this is not reproducible any longer.
The heap repeatedly shrinks, but will stop shrinking at ~8kB - the pruning seems to work. Zombie connections are still reclaimed after 60 secs. Safe to close I'd gather.
Good work!

projectgus, 2016-06-29:
Closing this as I've awlso confirmed it's fixed since #76.

closed 2016-06-29 by projectgus