SuperHouse/esp-open-rtos #103: Fatal exception (20) at `prvIdleTask` in http_get example

MightyPork, 2016-02-28:
This is an elusive error that sometimes happens after a restart (using the RST pin).

I am using the un-modified `http_get` example on ESP-01 with 8Mb flash.

```none

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 28 2016 15:15:02
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(18:fe:34:d5:03:a4)
add if0
HTTP get task starting...
Running DNS lookup for chainxor.org...
scandone
add 0
aid 1
cnt 
Fatal exception (20): 
epc1=0x00000001
epc2=0x00000000
epc3=0x40225d90
excvaddr=0x00000000
depc=0x00000000
excsave1=0x00000001

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30928, room 16 
```

It happens at `0x40225d90`, sometimes at `0x40225df5`.

Here's the addrsource output:

```none
0x40225df5 is in prvIdleTask (/home/ondra/devel/esp/sdk/esp-open-rtos/FreeRTOS/Source/tasks.c:2133).
2128	
2129				A critical region is not required here as we are just reading from
2130				the list, and an occasional incorrect value will not matter.  If
2131				the ready list at the idle priority contains more than one task
2132				then a task other than the idle task is ready to execute. */
2133				if( listCURRENT_LIST_LENGTH( &( pxReadyTasksLists[ tskIDLE_PRIORITY ] ) ) > ( unsigned portBASE_TYPE ) 1 )
2134				{
2135					taskYIELD();
2136				}
2137			}
Line 2133 of "/home/ondra/devel/esp/sdk/esp-open-rtos/FreeRTOS/Source/tasks.c" starts at address 0x40225df5 <prvIdleTask+125> and ends at 0x40225dfd <prvIdleTask+133>.
Dump of assembler code for function prvIdleTask:
   0x40225d78 <+0>:	addi	a1, a1, -32
   0x40225d7b <+3>:	s32i.n	a12, a1, 24
   0x40225d7d <+5>:	s32i.n	a13, a1, 20
   0x40225d7f <+7>:	s32i	a14, a1, 16
   0x40225d82 <+10>:	l32r	a13, 0x40225be0
   0x40225d85 <+13>:	l32r	a12, 0x40225d18
   0x40225d88 <+16>:	l32r	a14, 0x402259dc
   0x40225d8b <+19>:	s32i.n	a0, a1, 28
   0x40225d8d <+21>:	s32i	a15, a1, 12
   0x40225d90 <+24>:	memw
   0x40225d93 <+27>:	l32i.n	a2, a13, 0
   0x40225d95 <+29>:	beqz	a2, 0x40225df5 <prvIdleTask+125>
   0x40225d98 <+32>:	memw
   0x40225d9b <+35>:	l32i.n	a2, a12, 0
   0x40225d9d <+37>:	l32i.n	a15, a14, 0
   0x40225d9f <+39>:	addi.n	a2, a2, 1
   0x40225da1 <+41>:	memw
   0x40225da4 <+44>:	s32i.n	a2, a12, 0
   0x40225da6 <+46>:	l32r	a0, 0x4022571c
   0x40225da9 <+49>:	callx0	a0
   0x40225dac <+52>:	beqz	a15, 0x40225d90 <prvIdleTask+24>
   0x40225daf <+55>:	l32r	a0, 0x40225598
   0x40225db2 <+58>:	callx0	a0
   0x40225db5 <+61>:	l32i.n	a2, a14, 12
   0x40225db7 <+63>:	l32i.n	a15, a2, 12
   0x40225db9 <+65>:	addi.n	a2, a15, 4
   0x40225dbb <+67>:	l32r	a0, 0x40225be4
   0x40225dbe <+70>:	callx0	a0
   0x40225dc1 <+73>:	l32r	a3, 0x402259c4
   0x40225dc4 <+76>:	memw
   0x40225dc7 <+79>:	l32i.n	a2, a3, 0
   0x40225dc9 <+81>:	addi.n	a2, a2, -1
   0x40225dcb <+83>:	memw
   0x40225dce <+86>:	s32i.n	a2, a3, 0
   0x40225dd0 <+88>:	memw
   0x40225dd3 <+91>:	l32i.n	a2, a13, 0
   0x40225dd5 <+93>:	addi.n	a2, a2, -1
   0x40225dd7 <+95>:	memw
   0x40225dda <+98>:	s32i.n	a2, a13, 0
   0x40225ddc <+100>:	l32r	a0, 0x402255a4
   0x40225ddf <+103>:	callx0	a0
   0x40225de2 <+106>:	l32i.n	a2, a15, 48
   0x40225de4 <+108>:	l32r	a0, 0x4022562c
   0x40225de7 <+111>:	callx0	a0
   0x40225dea <+114>:	mov.n	a2, a15
   0x40225dec <+116>:	l32r	a0, 0x4022562c
   0x40225def <+119>:	callx0	a0
   0x40225df2 <+122>:	j	0x40225d90 <prvIdleTask+24>
   0x40225df5 <+125>:	l32r	a3, 0x402259cc
   0x40225df8 <+128>:	l32i.n	a2, a3, 0
   0x40225dfa <+130>:	bltui	a2, 2, 0x40225d90 <prvIdleTask+24>
   0x40225dfd <+133>:	movi.n	a2, 1
   0x40225dff <+135>:	l32r	a0, 0x402255a0
   0x40225e02 <+138>:	callx0	a0
   0x40225e05 <+141>:	j	0x40225d90 <prvIdleTask+24>
End of assembler dump.
```

I don't really understand the assembler, but maybe you'll make some sense of it.

Sometimes the ESP also enters a weird lock-out state with a reboot loop and gets really hot, but that may as well be some electrical abuse I did to it earlier, now manifesting itself this way.

projectgus, 2016-02-28:
Hi,

Thanks for the detailed bug report! The address to look at in these dumps is actually EPC1, even though it doesn't look like a meaningful number. Exception 20 is [InstFetchProhibited](https://github.com/SuperHouse/esp-open-rtos/wiki/Crash-Dumps#exception-cause) which means the running PC address has gone somewhere wrong - in this case the processor has jumped to EPC1=0x00000001 for some reason.

(The link above explains how to decipher the various EPCx values as well.)

I have a couple of suggestions:

- Because you're on an 8mbit ESP module, you need to pass FLASH_SIZE=8 to make. You can do this as `make flash FLASH_SIZE=8`, or you can create a local.mk file in the esp-open-rtos root directory to set it permanently. The default size is 16mbit, and the Espressif binary libraries save WiFi data on the last sector of the flash (based on the flash size). So it may be trying to write off the end, and making things wonky.

- Because you have random crashes and boot loops I'm thinking you might have brownout or glitchy power problems - the symptoms are often like this. Do you have a sturdy high current 3.3V supply to your ESP-01, with a high value low ESR capacitor or two placed close to the module? Check here for some more power-related tips: https://github.com/themadinventor/esptool#insufficient-power

Hope some of that helps. :)

Angus

MightyPork, 2016-02-28:
Thanks for the reply!

ad 1), I actually changed the size in the `common.mk` file, but the local file is a better idea, I'll do that.

Good point with the power supply, that may very well be the case (and related to the previous issue). I'll try to analyse it with a scope.

I'm powering the module from USB via LD1117-3.3, and there's a 1000µF cap, but quite far. Certainly something to look into. I'll close this, since it looks like a HW problem.

closed 2016-02-28 by MightyPork

projectgus, 2016-02-28:
Feel free to reopen if you think it's not a hardware problem later on. AFAIK most people are using modules with more flash, so there might be something we haven't noticed about ESP-01s.

foogod, 2016-02-28:
Yeah, I would be very wary of that power supply.. The USB spec only allows drawing 100mA unless you're using a smart USB device that can negotiate a higher current from the USB controller.  The ESP8266 can pull up to 400mA, (especially, I've noticed, when first starting up the wifi hardware), which is far more than the USB port is supposed to allow.  I wouldn't be surprised if you're getting voltage sags as a result.

MightyPork, 2016-03-01:
I swapped the power supply for a hard one, voltage is steady, and those issues keep happening.

Weird thing is that it stops after restarting the chip several times.

```none
Fatal exception (29): 
epc1=0x4023cda0
epc2=0x40100070
epc3=0x00000000
excvaddr=0x00000000
depc=0x40101b42
excsave1=0x4023cda0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31096, room 16 
tail 8
chksum 0x90
load 0x3ffe8000, len 2168, room 0 
tail 8
chksum 0x6c
load 0x3ffe8880, len 5592, room 0 
tail 8
chksum 0x87
csum 0x87
Fatal exception (29): 
epc1=0x4023cda3
epc2=0x40100070
epc3=0x00000000
excvaddr=0x00000000
depc=0x40101b42
excsave1=0x4023cda0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31096, room 16 
tail 8
chksum 0x90
load 0x3ffe8000, len 2168, room 0 
tail 8
chksum 0x6c
load 0x3ffe8880, len 5592, room 0 
tail 8
chksum 0x87
csum 0x87
```

the address is `0x4023cda6 <sdk_meas_tone_pwr_db+86>`. 

I tried changing the flash size from 8 to 4, but the problem is still the same.

MightyPork, 2016-03-01:
Interesting new observation: I swapped the module for another one, known to work OK (with official AT firmware). And guess what: It's broken the same!

So it must be something with my circuitry. Too long wires?
It's currently on a breadboard—I'll make a little perfboard PCB for it & see if the problem solves itself.

MightyPork, 2016-03-03:
RE: previous issues, I've built a breadboard adapter with ESP-12 and it's working all right, so I guess it was really something bad with my breadboard wiring. Maybe some garbage inducting in the wires..?

projectgus, 2016-03-03:
Could be, breadboards can be finickity at the best of times - it could have been resistive drop when the power output spiked.

Glad you got a good solution now. :)