SuperHouse/esp-open-rtos #109: Bootloop with simple_cplusplus example

MightyPork, 2016-03-03:
I was curious to see if this example works, no real plans to use it.

But .. well, it doesn't. I tried some of the others (even the SSL one) and all worked fine, this one causes a bootloop.

```none
                                                 l`rr��l

                                                             r�����l
                                                                     l`
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30928, room 16 
tail 0
chksum 0xd1
load 0x3ffe8000, len 2168, room 8 
tail 0
chksum 0x21
load 0x3ffe8880, len 5696, room 8 
tail 8
chksum 0xdb
csum 0xdb
����nn�n�
            쌜���lb
                    bnn�
                          ��rrn�l�l`��n��nn�n�
                                                   쌜��b

```

projectgus, 2016-04-19:
Hi @MightyPork ,

Sorry I never got back to you on this. I think this is the same issue as #128, and there is a fix mentioned in that issue that should fix this as well.

Angus

MightyPork, 2016-04-19:
nice, thanks for info.

I'll try that when I get back to EOR, currently I've moved the project to IoT SDK + libesphttpd, need to finish it in time - will be poking around this again in the summer.

projectgus, 2016-04-24:
Sure thing!

I'm going to close this because it was same as #128. But please reopen if the issue reoccurs.

closed 2016-04-24 by projectgus