SuperHouse/esp-open-rtos #111: Add SmallJSONParser to extras

DagAgren, 2016-03-07:
There was a request to add SmallJSONParser to the extras dir, which seems quite a good idea.

I haven't had time to look at it yet, but as a first step I did create a git mirror of the repo here: https://github.com/DagAgren/SmallJSONParser

Would including it as a submodule make any sense, or should it just be copied in directly?

kanflo, 2016-03-07:
That would be a great add-on! Which way to add it could be discussed though. The most "EOR friendly" way would be to add (parts of) it to ```extras/``` but that would of course leave you (or someone) with an extra place to manage the code base. Another way would be as a sub module (see https://github.com/SuperHouse/esp-open-rtos/wiki/Third-Party-Libraries). I guess that might have limited success as the EOR make system probably would include ```SmallJSONParser/Test.c``` when building a project using the parser.

@projectgus, @foogod: what do you think? The way in which we add third party modules could need some discussion. Especially as general interest in the project grows and more third party code is added.

DagAgren, 2016-03-07:
Well, in this particular case, there are really only two files to include,
and I don't foresee a large number of updates in the future, so just
including directly is a reasonable option.

projectgus, 2016-03-07:
Thanks @DagAgren!

I think either approach is fine in this case, but if @DagAgren thinks there won't be a lot of updates then just copying two files into the tree is probably easiest.

@kanflo: Good suggestions, thanks for making them. FWIW the make system allows you to specify `component_SRC_FILES` instead of `component_SRC_DIR` if you want to list file names instead of pick up all the source files in a directory. All of the options are documented here: https://github.com/SuperHouse/esp-open-rtos/wiki/Build-Process#source-directories

Regarding `extras` in general, @foogod  and I have been discussing this offline and it's about time we took the discussion onto the mailing list. So we'll do that. But `extras` is the right place for this library, at the moment.

Will be great to have this, I'm thinking it would be neat to update the http_get_mbedtls example to parse the JSON that comes back from the howsmyssl.com API.

kanflo, 2016-08-04:
As a JSON parser [was included](https://github.com/SuperHouse/esp-open-rtos/commit/e7607ffc) I would like to close this issue if that's alright with all involved.

DagAgren, 2016-08-04:
jsmn has a number of drawbacks for highly resource-constrained systems that SmallJSONParser doesn't, though.

kanflo, 2016-08-04:
I did not know that. In that case it could be of interest to include SmallJSONParser in EOR. I will label this an enhancement and sometime in the future someone might compare the two JSON parsing frameworks.

DagAgren, 2016-08-04:
It's more a lexer than a parser. It generates a stream of tokens, without needing any memory allocation, which is the big advantage for embedded systems. It may be more cumbersome to use, though, although jsmn isn't that easy either.