SuperHouse/esp-open-rtos #112: Fatal exception when connecting to WEP AP

recursify, 2016-03-08:
I'm getting occasional crashes when trying to connect to a WEP secured AP.  Not sure if the security type is important here, but I have not seen this issue with WPA.  Seems to be coming from the `sdk_ieee80211_crypto_decap` function.

```
scandone
add 0
aid 5

connected with Farsite IIup, channel 6
dhcp client start...
cnt
Fatal exception (28):
epc1=0x402386b8
epc2=0x00000000
epc3=0x4022a76d
excvaddr=0x050c0420
depc=0x00000000
excsave1=0x050c040c
�@*rjrA(!�SK5�Pv�Qz~��Q*�KA���I*E����)�Ы�)~yh���n�*U��hI�A,PZMmr�,R�*���vAm=KA��A.-�n�*e����nA,r%��E,R�*1)�vA�9)��(!��n�*E��hInA,t[M��A,t[�pp_task_hdl : 3fff42d8, prio:14, stack:512
pm_task_hdl : 3fff4ca8, prio:1, stack:176
frc2_timer_task_hdl:3fff7760, prio:12, stack:200
```

```
:> xtensa-lx106-elf-addr2line -f -e wifi-module.out 0x402386b8
sdk_ieee80211_crypto_decap
```

Let me know if you need more info!  Using esp-open-rtos, commit hash `a32d1099fa90c8f5d4708d898e7c5cae23ff8ca7`

DongBH, 2016-07-18:
I have the same error: when connects to WEP ap， it crashes.

xlwxdl, 2018-08-17:
I have the same error: when connects to WEP ap， 
`add 0
aid 1

connected with wocao_tc_WEP, channel 6
dhcp client start...
cnt 
Fatal exception (20): 
epc1=0x00000d4c
epc2=0x00000000
epc3=0x402074bd
excvaddr=0x00000d4c
depc=0x00000000
excsave1=0x4021e15d
Registers:
a0 4021e15d a1 3fff02d0 a2  402225ec a3  3ffeac48
a4  00000018 a5  0000000e a6  3ffe9624 a7  000000ff
a8  0000003a a9  0000002a a10 00000002 a11 00000000
a12 3ffeac48 a13 3fff42b0 SAR 00000018

Stack: SP=0x3fff02d0
0x3fff02d0: 402225ec 40223095 3ffec570 3ffec57c
0x3fff02e0: 00000008 3ffec57c 00000000 00000031
0x3fff02f0: 00000018 40232522 3ffeac48 3ffec570
0x3fff0300: 000000f5 3ffec524 00000000 00000002
0x3fff0310: 0000002a 3ffeac48 3ffeac48 4020888f
0x3fff0320: 40231d7d 3feffe00 3ffebcb0 ffffffe0
0x3fff0330: 3ffebf60 00000000 00000000 00000000
0x3fff0340: 00000000 00000000 00000000 00000000

Free Heap: 51308
_heap_start 0x3fff1690 brk 0x3fff46c4 supervisor sp 0x40000000 sp-brk 47420 bytes
arena (total_size) 12340 fordblks (free_size) 3888 uordblocks (used_size) 8452

`

xlwxdl, 2018-08-21:
https://github.com/espressif/ESP8266_RTOS_SDK/issues/26
according  this issue，，，espressif has fix this bug   in   1.1.1  （sdk version ？）

xlwxdl, 2018-08-21:
OK, it seems work now.

I download libnet80211.a  from this 1.1.1 sdk

https://github.com/espressif/ESP8266_RTOS_SDK/blob/f33be1ab40a56370b68509979b0be7e017d20eec/lib/libnet80211.a

use ar commad extract libnet80211  from ESP8266_RTOS_SDK  and  esp-open-rtos.

`ar -x libnet80211.a`

replace ieee80211_crypto_wep.o   then creating  a new   libnet80211.a  by
`ar -r libnet80211.a  *.o`

Then rebuild the project.  

I have tested WPA2 ,WPA ,WEP   WIFI... It works now.
But.....  It's a workaround anyway, and I am not sure whether it will bring any other extra problems.

It would be helpful if you guys try it out as much as possible.

[lib.zip](https://github.com/SuperHouse/esp-open-rtos/files/2305368/lib.zip)