SuperHouse/esp-open-rtos #115: Fatal exception in prvTimerTask  (access_point example)

elephh, 2016-03-13:
I start exproring esp-open-rtos sdk, and when I run acces_point example unmodified code I often get that error:

`Fatal exception (9):.
epc1=0x40226a4c
epc2=0x00000000
epc3=0x40101367
excvaddr=0x00000016
depc=0x00000000
excsave1=0x40226ad4`

addrsource 0x40226a4c says:

`0x40226a4c is in prvTimerTask (/home/piotrek/Espressif/esp-open-rtos/FreeRTOS/Source/timers.c:527).
522<--->
523<---><------>while( xQueueReceive( xTimerQueue, &xMessage, tmrNO_DELAY ) != pdFAIL ) /*lint !e603 xMessage does not have to be initialised as it is passed out, not in, and it i
524<---><------>{
525<---><------><------>pxTimer = xMessage.pxTimer;
526<--->
527<---><------><------>if( listIS_CONTAINED_WITHIN( NULL, &( pxTimer->xTimerListItem ) ) == pdFALSE )
528<---><------><------>{
529<---><------><------><------>/* The timer is in a list, remove it. */
530<---><------><------><------>( void ) uxListRemove( &( pxTimer->xTimerListItem ) );
531<---><------><------>}
Line 527 of "/home/piotrek/Espressif/esp-open-rtos/FreeRTOS/Source/timers.c" starts at address 0x40226a4c <prvTimerTask+188> and ends at 0x40226a50 <prvTimerTask+192>.
Dump of assembler code for function prvTimerTask:
   0x40226990 <+0>:<--->addi<-->a1, a1, -64
   0x40226993 <+3>:<--->s32i.n<>a12, a1, 56
   0x40226995 <+5>:<--->s32i.n<>a13, a1, 52
   0x40226997 <+7>:<--->l32r<-->a12, 0x4022698c
   0x4022699a <+10>:<-->l32r<-->a13, 0x40226750
   0x4022699d <+13>:<-->s32i.n<>a0, a1, 60
   0x4022699f <+15>:<-->s32i.n<>a14, a1, 48
   0x402269a1 <+17>:<-->s32i.n<>a15, a1, 44
   0x402269a3 <+19>:<-->l32r<-->a3, 0x4022675c
   0x402269a6 <+22>:<-->movi.n<>a15, 0
   0x402269a8 <+24>:<-->l32i.n<>a2, a3, 0
   0x402269aa <+26>:<-->mov.n<->a4, a15
   0x402269ac <+28>:<-->l32i.n<>a3, a2, 0
   0x402269ae <+30>:<-->movi.n<>a5, 1
   0x402269b0 <+32>:<-->moveqz<>a4, a5, a3
   0x402269b3 <+35>:<-->extui<->a14, a4, 0, 8
   0x402269b6 <+38>:<-->bne<--->a14, a15, 0x402269bd <prvTimerTask+45>
   0x402269b9 <+41>:<-->l32i.n<>a2, a2, 12
   0x402269bb <+43>:<-->l32i.n<>a15, a2, 0
   0x402269bd <+45>:<-->l32r<-->a0, 0x40225e54
   0x402269c0 <+48>:<-->callx0<>a0
   0x402269c3 <+51>:<-->call0<->0x40226438 <xTaskGetTickCount>
   0x402269c6 <+54>:<-->mov.n<->a4, a2
   0x402269c8 <+56>:<-->l32i.n<>a2, a12, 0
   0x402269ca <+58>:<-->bltu<-->a4, a2, 0x402269d0 <prvTimerTask+64>
   0x402269cd <+61>:<-->j<----->0x40226ada <prvTimerTask+330>
   0x402269d0 <+64>:<-->s32i<-->a4, a1, 16
   0x402269d3 <+67>:<-->call0<->0x40226910 <prvSwitchTimerLists>
   0x402269d6 <+70>:<-->l32i.n<>a4, a1, 16
   0x402269d8 <+72>:<-->s32i.n<>a4, a12, 0
   0x402269da <+74>:<-->l32r<-->a0, 0x40225e64
   0x402269dd <+77>:<-->callx0<>a0
   0x402269e0 <+80>:<-->j<----->0x40226ac6 <prvTimerTask+310>
   0x402269e3 <+83>:<-->excw
   0x402269e6 <+86>:<-->l32r<-->a4, 0x401ee5ec
   0x402269e9 <+89>:<-->excw
   0x402269eb <+91>:<-->srai<-->a0, a0, 16
   0x402269ee <+94>:<-->addi.n<>a15, a15, 5
   0x402269f0 <+96>:<-->l32i.n<>a2, a3, 0
   0x402269f2 <+98>:<-->l32i.n<>a2, a2, 12
   0x402269f4 <+100>:<->l32i.n<>a14, a2, 12
   0x402269f6 <+102>:<->addi.n<>a2, a14, 4
   0x402269f8 <+104>:<->l32r<-->a0, 0x4022632c
   0x402269fb <+107>:<->callx0<>a0
   0x402269fe <+110>:<->l32i.n<>a2, a14, 28
[... 117 lines cut ...]
   0x40233245 <+21>:<-->bltui<->a0, 2, 0x4023324d <sdk_ieee80211_crypto_encap+29>
   0x40233248 <+24>:<-->addi.n<>a6, a0, -1
   0x4023324a <+26>:<-->j<----->0x4023324f <sdk_ieee80211_crypto_encap+31>
   0x4023324d <+29>:<-->mov.n<->a6, a0
   0x4023324f <+31>:<-->addx4<->a2, a0, a2
   0x40233252 <+34>:<-->l32i<-->a2, a2, 0x190
   0x40233255 <+37>:<-->slli<-->a4, a6, 6
   0x40233258 <+40>:<-->l32i<-->a0, a2, 152
   0x4023325b <+43>:<-->extui<->a4, a4, 0, 8
   0x4023325e <+46>:<-->l32i.n<>a0, a0, 16
   0x40233260 <+48>:<-->s32i.n<>a2, a1, 0
   0x40233262 <+50>:<-->callx0<>a0
   0x40233265 <+53>:<-->l32i.n<>a4, a1, 0
   0x40233267 <+55>:<-->l32i.n<>a3, a1, 4
   0x40233269 <+57>:<-->movi.n<>a0, 0
   0x4023326b <+59>:<-->addi<-->a1, a1, 16
   0x4023326e <+62>:<-->movnez<>a0, a4, a2
   0x40233271 <+65>:<-->mov.n<->a2, a0
   0x40233273 <+67>:<-->mov.n<->a0, a3
   0x40233275 <+69>:<-->ret.n
   0x40233277 <+71>:<-->movi.n<>a6, 0
   0x40233279 <+73>:<-->l8ui<-->a0, a0, 232
   0x4023327c <+76>:<-->j<----->0x4023324f <sdk_ieee80211_crypto_encap+31>
End of assembler dump.
`

projectgus, 2016-03-13:
Hi eleph,

Thanks for reporting this issue. Hopefully we can sort it out and get the AP code working properly for you.

I have a few questions to help narrow the problem down:

* Are there clients associated to the AP when the crashes happen? If so, which client(s)? Can you narrow it down a particular client, or particular timing (ie does it crash when the client first associates?)

* Do other examples (where the ESP is a WiFi client) work correctly?

* What ESP8266 module are you using? How is it connected/powered?

* Are you able to attach the full access_point.out file (in the 'build' subdirectory) that matches the crash dumps above, please?

Thanks!

Angus

mgolub2, 2016-03-14:
I am also having this issue, with an android client attached, which is the only attached client. It happens after several tcp sockets are opened, and supposedly closed...

The crash seems to be very random. I have the exact same line, 527 in timer.c, cause the exception. Here is the traceback:

```
Fatal exception (28):
epc1=0x402232d4
epc2=0x00000000
epc3=0x4010690c
excvaddr=0x0000021c
depc=0x00000000
excsave1=0x4022335c
```

I've attached the built file as well, which is from a project I'm working on. This is on a ESP01, powered by a 3.3v regulator coming off a FTDI cable - attempting now with a lab power supply just in case. 

[user_main.out.zip](https://github.com/SuperHouse/esp-open-rtos/files/171683/user_main.out.zip)

On a side note, it might be related to the line "max connection!" being spammed all the time for no apparent reason?

elephh, 2016-03-14:
Hi projectgus,

>  there clients associated to the AP when the crashes happen? If so, which client(s)? Can you narrow it down a particular client, or particular timing (ie does it crash when the client first associates?)

About Fatal exception (9). I try with few clients. One is windows pc, second is linux pc, and next is android phone. It does not matter wich client is connected, and how many clients is connected. Problem is not in particular time. It occurs at random times.

About (28) I dont have good idea when it occurs. I'll try to repeat the problem. 
I confirm what he wrote mgolub2  that it was occur when  using the android client.

> Do other examples (where the ESP is a WiFi client) work correctly?

No. This is the first I've tested.

> What  ESP8266 module are you using? How is it connected/powered?

I am  having ESP-01 module. Powered by standalone 3.3v regulator.(LF33).

> Are you able to attach the full access_point.out file (in the 'build' subdirectory) that matches the crash dumps above, please?

[access_point.out.zip](https://github.com/SuperHouse/esp-open-rtos/files/172616/access_point.out.zip)

projectgus, 2016-03-14:
Thanks @elephh . Does the crash trigger for you only if you connect multiple times via the telnet client, or does it happen just by associating?
h
> It happens after several tcp sockets are opened, and supposedly closed...

Do you see the free memory dropping down in the message printed by the TCP client? There's a bug (#101) where sockets left in TIME_WAIT (which happens when the connection is closed but there may still be data unsent, so the socket stays valid for 60 seconds) sockets can lead to out of memory.

Is it possible that's the same bug here?

elephh, 2016-03-15:
Hi @projectgus 

> Thanks @elephh . Does the crash trigger for you only if you connect multiple times via the telnet client, or does it happen just by associating?

Example crash session. There is no any tcp connection:

SDK version:0.9.9
mode : softAP(1a:fe:34:a3:93:b5)
add if1
bcn 100
add 1
aid 1
station: 6c:71:d9:6e:c4:44 join, AID = 1
max connection!
max connection!
station: 6c:71:d9:6e:c4:44 join, AID = 1
max connection!
max connection!
station: 6c:71:d9:6e:c4:44 join, AID = 1
max connection!
max connection!
add 2
aid 2
station: 1c:b7:2c:52:7d:89 join, AID = 2
max connection!
station: 1c:b7:2c:52:7d:89 join, AID = 2
max connection!
station: 6c:71:d9:6e:c4:44 join, AID = 1
max connection!
station: 6c:71:d9:6e:c4:44 join, AID = 1
max connection!
Fatal exception (9): 
epc1=0x40226a7c
epc2=0x00000000
epc3=0x4010137c
excvaddr=0x00000015
depc=0x00000000
excsave1=0x40226b04

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 31196, room 16 
tail 12
chksum 0xe2
ho 0 tail 12 room 4
load 0x3ffe8000, len 2168, room 12 
tail 12
chksum 0xca
ho 0 tail 12 room 4
load 0x3ffe8880, len 6532, room 12 
tail 8
chksum 0xe2

> Do you see the free memory dropping down in the message printed by the TCP client? 

Yes I see memmory dropping as referenced in (#101), but  in example above i don't establish any tcp connection.

projectgus, 2016-03-15:
Thanks for posting the log. There are some unusual things going on here - there are only `join` messages for the two station MAC addresses, but no messages from the DHCP server and no `leave` messages showing the client disassociating and then re-associating.

For comparison, here's a log if I connect to the access_point example from my Nexus 4:
http://pastebin.com/raw/837NSebg

The extra output is mostly the DHCP server, and you also see the client disconnecting and reconnecting after it fails to find internet access.

Whereas in your log both of the clients are associating to the AP, not requesting a DHCP lease, and never de-associating. Which seems unusual.

Are there any logs you can see on the client side showing what they think is happening from their point of view?

projectgus, 2016-03-15:
A couple of other things, here's the log from the access_point.out posted here:
http://pastebin.com/raw/eKzNDKVr

There's one different I noticed - this log dumps 10 lease slots, which implies that maybe the example was a bit modified because the default in the example is 4? Are there any other changes? Did you also increase the max_connection number in the ap_config from the default of 3?

The other thing I realised is that the flash length on ESP-01 is usually shorter than the default set by esp-open-rtos. If you run `make flash FLASH_SIZE=4` then does the example run any more reliably?

elephh, 2016-03-15:
Thanks for reply.

In log there are only the join messages, but it doesn't matter. I have simlar logs as yours with lease messages. I append this log to ilustrate that the connection and TCP client have not impact for "my exception".

Generaly I don't have problem with connection establishing, and dhcp requests and leases.

You have right, there is one difference at number of dhcp slots (10). I forgot about this.
In ap_config the max_connection number is still unchanged. 
My FLASH_SIZE was 8,  but when u changed it to 4 there is the same problem.

I added more complex log 
[log1.zip](https://github.com/SuperHouse/esp-open-rtos/files/174995/log1.zip)

projectgus, 2016-03-16:
Right, I see. This second log looks more normal (up to the crash). The one other thing is this:
> DHCP Server Error: No requested IP

Which implies client `1c:b7:2c:52:7d:89` sent a DHCP request that we weren't expecting. Looking at that code I see a bug in that code, but it's not a bug that would cause the kind of crash you're seeing.

If you just associate one of the two clients you're using, does it still crash? Is the crash associated with either client, one particular client, or with both clients being associated at once?

elephh, 2016-03-16:
Yes. This exception:

Fatal exception (9): 
epc1=0x40226a7c
epc2=0x00000000
epc3=0x4010137c
excvaddr=0x00000015
depc=0x00000000
excsave1=0x40226b04

 is not associated with particular client. It occurs in both cases: a single client or more clients connected. No matter which client is connected and how many cliets is connected. Exception not appears immediately when client is connected. Once it appears faster, sometimes slower. I don't see any time relationship betwen exception and clients connections.

projectgus, 2016-05-09:
Quite note that I think I saw this recently while investigating #140. It seems like something in SoftAP mode will eventually corrupt memory and causes a random crash.

ourairquality, 2016-09-22:
I guess this is fixed by https://github.com/SuperHouse/esp-open-rtos/pull/221too as I have not seen it again after that fix. Perhaps this could be closed and reopened if it is seen again.

closed 2016-10-26 by sheinz

trjohn, 2018-01-23:
Hi,

I get the same random crash in SoftAP mode with AUTH_WPA2_PSK and AUTH_WPA_WPA2_PSK. 

Could anyone explain what is the fix made for this issue?

Thanks in advance

ourairquality, 2018-01-23:
@trjohn The timers were badly broken, and that was fixed. There might be multiple issues. I've run in SOFTAP mode and it seems to work fine, so you might have to try to narrow it down yourself, to get some clues, and file a new issue.