SuperHouse/esp-open-rtos #116: printf doesn't support 64-bit types

foogod, 2016-03-16:
In the same vein as #95, it appears that the version of printf we're using does not support printing 64-bit integers. (for example, PRx64 is defined in inttypes.h as "llx", but using "%llx" in printf just prints "lx" in the output)

I'm not sure if this is even an option in newlib-nano, but it would be handy for some things if we can enable it somehow..

projectgus, 2016-03-16:
I believe we have to switch from -nano back to the full newlib I/O routines for 64-bit integer support.

https://answers.launchpad.net/gcc-arm-embedded/+question/219065

We could measure the size/performance impact caused by switching from nano to the full formatting routines, but I suspect it will be a significant impact (printf has to be loaded into IRAM at the moment, and IRAM is already pretty full.)

For formatting uint64_t as hex, as I'm sure you know, the quick fix is to printf %x%x and split a 64-bit integer into two 32-bit integer params.

My suggestion is we don't change libc but note this somewhere, maybe on the wiki here:
https://github.com/SuperHouse/esp-open-rtos/wiki/libc-configuration

What do you want to do?

foogod, 2016-03-16:
Ah, I was afraid it would be something like that, but hadn't had a chance to really dig into it in detail..

For hex, that's a viable (if annoying) workaround (and is what I've been doing), but it doesn't work nearly as well for decimal (particularly signed numbers), etc..  Still, until we have somebody with an actual use case for that, I guess it's not worth expending a lot of effort to try to fix what is currently only a theoretical issue..  I just figured I should bring it up in case it was as easy as the float-fix was..

It might be good for us to start maintaining a wiki page on "quirks and limitations" of our platform for things like this (and time/timezone handling, and I'm sure there's a couple of other things we've talked about which would fall in this category)..  it might be good to have something to point prospective app developers at that says, basically, "these are some things you shouldn't be surprised about when you stumble across them"..

AndreaGreco, 2016-09-01:
Me 2 found same problem
solved with:

#define PRI_UINT64_C_Val(value) ((unsigned long) (value>>32)), ((unsigned long)value)
#define PRI_UINT64 "%lx%lx"

printf("My Text" PRI_UINT64 "MyText - Text", PRI_UINT64_C_Val (tmp.Address));