SuperHouse/esp-open-rtos #120: esp-gdbstub support

aacebedo, 2016-04-03:
Hi 

I am trying to use the gdbstub from espressif (https://github.com/espressif/esp-gdbstub) and I am unable to build the blink example.

I am missing the reference to the following functions:
```
LD build/blink.out
./build/program.a(gdbstub.o): In function `gdbstub_handle_uart_int':
/tmp/esp-open-rtos/examples/blink/gdbstub.c:759: undefined reference to `user_fatal_exception_handler'
/tmp/esp-open-rtos/examples/blink/gdbstub.c:759: undefined reference to `os_install_putc1'
./build/program.a(gdbstub.o):(.text.gdbstub_init+0x1c): undefined reference to `_xt_isr_unmask'
./build/program.a(gdbstub.o): In function `gdbstub_init':
/tmp/esp-open-rtos/examples/blink/gdbstub.c:776: undefined reference to `os_install_putc1'
./build/program.a(gdbstub.o): In function `install_uart_hdlr':
/tmp/esp-open-rtos/examples/blink/gdbstub.c:763: undefined reference to `_xt_isr_unmask'
./build/program.a(gdbstub-entry.o):(.text+0x8): undefined reference to `_DebugExceptionVector'
collect2: error: ld returned 1 exit status
../../common.mk:352: recipe for target 'build/blink.out' failed
make: *** [build/blink.out] Error 1
```
Do you have an idea to where I shall find those functions ?

aacebedo, 2016-04-03:
I've found that some of the missing functions have been prefixed with sdk_ in the sdk. By adding this prefix in the gdbstub code calls I managed to remove all undefined references except one about '_DebugExceptionVector' 
The assembly source of the gdbstub declare that but for an unknown reason, the linker does not find it.

projectgus, 2016-04-03:
Hi,

A couple of people have expressed interest in porting the gdbstub, although as far as I know you're the first person to do it. Thanks for getting in touch!

You're right that esp-open-rtos prefixes all the Espressif SDK functions and internals with sdk_, so that's the first modification you'll need.

Our exception vectors are a bit different to Espressif's RTOS SDK, the DebugExceptionVector label is the same but without the underscore:
https://github.com/SuperHouse/esp-open-rtos/blob/master/core/exception_vectors.S#L57

However, I think there may be some deeper differences in the way the gdbstub hooks exceptions, so this may not be quite enough to get a working gdb environment.

Please let us know how you go. If you publish your code then I'm happy to take a look at it.

Angus

aacebedo, 2016-04-04:
Hi Angus 

I've tried the modifications but it does not work. The board does not respond to the break points when the gdb_init function has been called. I think something is messing up with the interrupt or the serial line. Do you have the assembly doc of the xtensa chip ? I do not have any jtag debugger so I am more trying to guess than to actually program this thing :(

projectgus, 2016-04-04:
The assembly reference is the Xtensa Instruction Set Reference Manual. There are copies floating around online.

If you have specific questions then I'm happy to help, or I can take a quick look at the code if you post your work in progress. :)

aacebedo, 2016-04-12:
Hi !

I've committed my modification in the branch https://github.com/aacebedo/esp-open-rtos/tree/feature/gdbstub_test .
The current status is that when the gdbstub_do_break is called, the board stops as intended, however it does not respond to gdb (timeout on serial messages exchange). I think there is something wrong with the exception vectors or the uart configuration. 
Do you have a JTAG probe to check what's actually happening ?
Regards.

resetnow, 2016-08-21:
@aacebedo I have a JTAG probe which I can't get to work right, but it does allow to see current PC and stack backtrace, so if you are willing to try, I can test your .elf file.

resetnow, 2016-08-22:
So I tried debugging gdbstub and here are some observations:

* It tries to change the debug and user exception vectors in an extremely hacky way by creating instructions at runtime
* I haven't found a way to change the debug exception vector, so I modified `core/exception_vectors.S` to add a pointer to exception handler which defaults to fatal_exception_handler.

```patch
diff --git a/core/exception_vectors.S b/core/exception_vectors.S
index 000e742..f19a7e3 100644
--- a/core/exception_vectors.S
+++ b/core/exception_vectors.S
@@ -23,6 +23,13 @@
 #define CAUSE_LOADSTORE 3
 #define CAUSE_LVL1INT 4
 
+        .section .data
+        .balign 4
+        .global debug_exception_handler
+
+debug_exception_handler:
+        .word   fatal_exception_handler
+
         .section .bss
 
 /* Stack space for NMI handler
@@ -70,8 +77,7 @@ DebugExceptionVector:
         wsr     a0, excsave2
         mov     a2, a1
         movi    a3, 0
-        call0   fatal_exception_handler
-        rfi     2
+        j       DebugExceptionHandler
 
         .org    VecBase + 0x20
 NMIExceptionVector:
@@ -362,6 +368,18 @@ LoadStoreErrorHandler:
         rsr     a1, excsave1
         rfe
 
+/*************************** Debug exception handler *************************/
+
+        .section .vecbase.text, "x"
+        .literal_position
+        .balign 4
+
+DebugExceptionHandler:
+        movi    a4, debug_exception_handler
+        l32i    a4, a4, 0
+        callx0  a4
+        rfi     2
+
 /****************************** call_user_start ******************************/
 
         .section .vecbase.text, "x"
```

I have pushed updated esp-gdbstub to https://github.com/resetnow/esp-gdbstub; as of now, the following things work:

* Breaking on gdbstub_do_break 
* Instruction stepping
* Memory and registers view.

It requires the patch above applied to esp-open-rtos. Unfortunately, after a ret instruction the processor starts spinning in the debug exception vector. I hope this goes somewhere — having no debug capability surely sucks :(.

resetnow, 2016-08-25:
I have finished porting esp-gdbstub. Line stepping works now as well, although since there is only one breakpoint, it has to be unused in order for line step, run to line and move to line commands to work. Also, fatal exception handler probably could be connected to gdbstub too, I haven't implemented it yet.

I have a couple of questions for esp-open-rtos maintainers (@projectgus?):

1. Would you be interested in merging esp-open-rtos exception handlers patch so gdbstub works with it “out of the box”?
1. gdbstub is distributed under Espressif “MIT” license, so it can't be merged as a esp-open-rtos component. Do you think it's worth a try to ask Espressif for a permission to merge the code here under the MIT license?

aacebedo, 2016-08-25:
@resetnow Wow great work !
I didn't have the time to try this out.
The gdbstub is still limited to one breakpoint ?
Regards

resetnow, 2016-08-25:
@aacebedo yes, as per Espressif source code, it's a hardware limitation (although I'm willing to try to set more — the number of breakpoints supported by lx106 core can be configured by system designer). You can place software breakpoints for code in RAM (although I haven't tried it yet), but you'll have to modify the memory map in .gdbinit.

aacebedo, 2016-08-25:
@resetnow So if I use software breakpoint I shall be able to set more than one (with set breakpoint auto-hw off) ?
Can you give me the reference of your jtag probe ? I was considering buying an olimex probe like that https://www.olimex.com/Products/ARM/JTAG/ARM-USB-OCD-H/

resetnow, 2016-08-25:
@aacebedo I haven't managed to get JTAG to work stable enough with sysprogs openocd. Sometimes it worked and I was able to pause/resume program and read registers, sometimes it didn't. I'm using a cheap FT2232 dev board.

resetnow, 2016-08-25:
Also, if you know the desired location of breakpoint before downloading the program, you can just use gdbstub_do_break() as many times as you want.

resetnow, 2016-09-03:
#205

sheinz, 2016-11-25:
As esp-gdbstub by @resetnow works, I will close this.

closed 2016-11-25 by sheinz